dariuszseweryn/RxAndroidBle #5: Possible concurrency issue

artem-zinnatullin, 2016-03-29:
Yo guys, was just looking into sources and found pretty [suspicious usage of non-thread-safe collection](https://github.com/Polidea/RxAndroidBle/blob/7e694dad75756fa1664d89409e0adc3fa3c1ecf5/rxandroidble/src/main/java/com/polidea/rxandroidble/RxBleClientImpl.java#L34).

Please consider using `ConcurrentHashMap` there because it may be accessed from different threads with `subscribeOn` and `unsubscribeOn`.

Library looks nice! (extra points for Spock!)

uKL, 2016-03-30:
Yes, this may be true, we will take a look. However ConcurrentHashMap may not be enough. We should synchronize whole read/create operation.

uKL, 2016-03-30:
Issue will be fixed by https://github.com/Polidea/RxAndroidBle/pull/6.

closed 2016-03-30 by uKL