dariuszseweryn/RxAndroidBle #7: What's the meanings of autoconnect?

tinybright, 2016-03-31:
It's the normal meaning.
Or it's same as autoConnect in BluetoothDevice

    [ * @param autoConnect Whether to directly connect to the remote device (false)
     *                    or to automatically connect as soon as the remote
     *                    device becomes available (true).](url)

mzgreen, 2016-03-31:
It's the same as `autoConnect` in BluetoothDevice. I think that it should be mentioned in the doc, do you agree @uKL ?

closed 2016-03-31 by tinybright

uKL, 2016-03-31:
True, same meaning. Without auto connect the system will throw BleGattException with detail code 133. This will happen after some timeout, various on different devices.

With auto connect it will wait until the BLE device is available.

farmazon3000, 2016-07-01:
#40 related.