dariuszseweryn/RxAndroidBle #8: Does it support onCharacteristicChanged?

tinybright, 2016-03-31:
I read doc and haven't find onCharacteristicChanged.Does it support onCharacteristicChanged?

mzgreen, 2016-03-31:
Yes it does. See https://github.com/Polidea/RxAndroidBle#change-notifications

tinybright, 2016-03-31:
I overlooked it.

closed 2016-03-31 by tinybright

xmbb, 2016-06-30:
i can't understand lambda, can you help to transform to normal

dariuszseweryn, 2016-07-11:
Lambdas are just a syntax sugar. You can read about them here: https://docs.oracle.com/javase/tutorial/java/javaOO/lambdaexpressions.html
You can find a lot of articles about lambdas on the Internet. ( https://blog.idrsolutions.com/2014/10/5-minutes-explanation-java-lambda-expression/ )

Long story short - instead of writing boilerplate in anonymous implementation of the interfaces with only one method / function you use lambda which is shorter to write.