dariuszseweryn/RxAndroidBle #10: Location permission

klemzy, 2016-04-07:
I am just curious, why the library requires to have location permissions enabled in order to work?

Thanks!

uKL, 2016-04-07:
Hey!

Starting from Android 6.0 it is required to hold **ACCESS_COARSE_LOCATION** or **ACCESS_FINE_LOCATION** in order to get scan results of device mac address. This is to strengthen user privacy. Even more, it is required that the user will have location services enabled.

You can read more [here](http://developer.android.com/reference/android/bluetooth/le/BluetoothLeScanner.html#startScan(java.util.List<android.bluetooth.le.ScanFilter>, android.bluetooth.le.ScanSettings, android.bluetooth.le.ScanCallback)).

closed 2016-04-07 by uKL

klemzy, 2016-04-07:
Hi,

I see, was not aware of that. Thanks!

-- 
Klemen
Sent with Airmail

uKL, 2016-04-07:
We handle it internally (both permission and enabled state) and BleScanException is emitted with code LOCATION_PERMISSION_MISSING or LOCATION_SERVICES_DISABLED.

mars3142, 2016-04-18:
And why don't you handle the "permission request" in your library? Do I have to handle by myself?

uKL, 2016-04-18:
Yes, you should handle them by yourself. It would be hard to do it in the library because each application has it's own design. Making an universal one might not be the best idea.

There are libraries for Runtime Permissions that might be a good choice for you.

MimgshuZhao, 2016-10-03:
I always get the hint: Location services needs to be enabled on Android 6.0
How can I enable the location services? 
Thanks

MimgshuZhao, 2016-10-03:
Oh, never mind, I just enable it (By enable GPS...)
Thanks for the source anyway.