dariuszseweryn/RxAndroidBle #9: Periodicity reading rssi

odmitriv, 2016-04-07:
Please explain the correct way of reading rssi with a certain periodicity using reactive programming. But I do not have sufficient experience in the rx. Any help would be useful to me.

uKL, 2016-04-07:
Are you talking about RSSI received in a scan or from a connected device?

odmitriv, 2016-04-07:
From a connected device

uKL, 2016-04-07:
Please have a look at the example. Simple as that.

```
bleDevice.establishConnection(this, false)
    .observeOn(AndroidSchedulers.mainThread())
    .flatMap(rxBleConnection -> // Set desired interval.
        Observable.interval(2, SECONDS).flatMap(sequence -> rxBleConnection.readRssi()))
    .subscribe(this::updateRssi, this::onConnectionFailure);
```
https://github.com/Polidea/RxAndroidBle/blob/master/sample/src/main/java/com/polidea/rxandroidble/sample/example5_rssi_periodic/RssiPeriodicExampleActivity.java

closed 2016-04-07 by uKL