dariuszseweryn/RxAndroidBle #15: Can't re-establish connection

dlykoits, 2016-04-20:
### Some times phone lost connection, and it isn't establish connection to device again
<br />
#### Preconditions
Maybe I do something wrong
<br />
#### Steps to reproduce actual result
1. connect ble device
2. turn off ble device(connection lost)
3. turn on ble device 
#### Actual result
connection establishing imidiately closed after establishing
<br />
#### Expected result
connection established again
<br />
Source code:
           RxBleDevice rxBleDevice = bleClient.getBleDevice(pairedDevice.getMac());
            compositeSubscription.add(rxBleDevice.establishConnection(context, true)
                    .subscribeOn(Schedulers.io()).observeOn(AndroidSchedulers.mainThread())
                    .subscribe(connectionSubscriber));

uKL, 2016-04-20:
The library notifies you that the connection has been lost but you need to call establish connection again to reconnect. Is that the case?

dlykoits, 2016-04-20:
yes

uKL, 2016-04-20:
This is expected behavior. Once disconnected you need to establish the connection manually again. Why do you think that it should be anyway different?

dlykoits, 2016-04-20:
Sorry, misunderstood you.
I just call this code again when connection state change(disconnected) appear.
and nothing happened.

uKL, 2016-04-20:
Could you provide code sample and preferably logs from RxAndroidBle?

dlykoits, 2016-04-20:
Please look  at sources: https://github.com/dlykoits/RxAndtoidBLESample.git
Steps to reproduce:
1) press connect -> ble device connected
2) then turn off ble device - > connection lost
3) turn on ble device
4) press connect
Actual result:
connection establishment doesn't work, "disconnected" status appear immediately after establish call 
Expected result:
connection established again

dariuszseweryn, 2016-04-20:
What I see is that you're trying to reuse a Subscriber here and that is a misuse as the Subscriber is stateful and therefore cannot be reused. It was already failed once and is probably unsubscribing automatically.
You can try either to create a new Subscriber for each subscription or to use an Observer instead - which is stateless.

dlykoits, 2016-04-21:
You right, Thanks a lot!!!
P.S.: I started learning RX week ago.

closed 2016-04-21 by dlykoits

farmazon3000, 2016-07-01:
@uKL `The library notifies you that the connection has been lost` sometimes it does not, see #40.

farmazon3000, 2016-07-01:
So this is what I have to automatically try to connect to device after disconnect happens with 5 seconds delay:
```
rxBleDevice.establishConnection(context, false) // <-- autoConnect flag
    .doOnError(throwable -> {
        Log.d(TAG, "rxBleConnection doOnError: " + throwable);
    })
    .retryWhen(new Func1<Observable<? extends Throwable>, Observable<?>>() {
        @Override
        public Observable<?> call(Observable<? extends Throwable> errorNotification) {
            return errorNotification
                    .flatMap(new Func1<Throwable, Observable<?>>() {
                        @Override
                        public Observable<?> call(Throwable throwable) {
                            Log.v(TAG, "connect retrying...");
                            return Observable.timer(5, TimeUnit.SECONDS);
                        }
                    });
        }
    })
    .subscribe(rxBleConnection -> {
        Log.d(TAG, "rxBleConnection: " + rxBleConnection);
    });
```