dariuszseweryn/RxAndroidBle #16: Support for API level 21 (package android.bluetooth.le.*)

ecdpalma, 2016-04-25:
As `BluetoothAdapter#startScan` has been deprecated, it would be interesting to support the calls from the mentioned package, which allow more control over the scanning process.

As a temporary hack, I created (copy-pasted, in fact) a scan operation class that uses the new API. It can be found [here](https://github.com/ecdpalma/RxAndroidBle/commit/3f20355a49eb2d58f45c311e4977ae24019272ac). My idea is that the scan operation receives `ScanFilter` and `ScanSettings` and emits `ScanResult`. All these classes are from the `android.bluetooth.le` package.

dariuszseweryn, 2016-04-25:
We are aware of the new scanning API. Thank you for showing your approach - we will check it.
Personally I was thinking about more universal approach in the API and just omit the settings that are unavailable on API < 21.

uKL, 2016-04-25:
Or maybe we could backport some of them... Worth considering.

mzgreen, 2016-05-09:
I agree, we could try to backport as much as we can but for now we should call new API methods on Lollipop and above and fallback to old one's on previous versions of Android.

recoverrelax, 2016-06-30:
Yes Please support api 21+. The improvements on the api are so great, it will be a waste not to use the earlier APIs. I can tell you, devices >21, little to few able problems using the new API. The older API on the other hand ..

aldoborrero, 2016-08-16:
Why not to use the [Android BLE Scanner Compat](https://github.com/NordicSemiconductor/Android-Scanner-Compat-Library) that NordicSemi have?

It allows you to abstract like the official android compat libraries does. The license is BSD clause 3 so it doesn't have to suppose a problem to use its codebase.

cpboyd, 2016-08-25:
I would at least be interested to invert the exclusivity of the service UUID filter, so that I could find different devices with specific UUIDs that are contained in the list.

e.g.:
```java
    private boolean containsDesiredServiceIds(byte[] scanRecord) {
        List<UUID> advertisedUUIDs = uuidUtil.extractUUIDs(scanRecord);

        for (UUID desiredUUID : filterServiceUUIDs) {

            if (advertisedUUIDs.contains(desiredUUID)) {
                return true;
            }
        }

        return false;
    }
```

bobroalexandr, 2016-09-23:
@uKL @dariuszseweryn I've created pull request for ble scan v21 support https://github.com/Polidea/RxAndroidBle/pull/73. Please check if that's helpful for your project

dariuszseweryn, 2017-05-22:
Just added the new scanning API to `1.3.0-SNAPSHOT`. Feel free to check it out.

closed 2017-05-22 by dariuszseweryn