dariuszseweryn/RxAndroidBle #18: RxBleRadioOperation will never FINISHED

yuanzj, 2016-04-29:
### Summary
When I turn off Bluetooth,RxBleRadioOperation will never FINISHED.
<br />
#### Preconditions
turn off Bluetooth
<br />
#### Steps to reproduce actual result
1.turn off Bluetooth
2.call 
```
connectionObservable = bleDevice
                .establishConnection(context, false)
                .takeUntil(disconnectTriggerSubject)
                .doOnUnsubscribe(BasicBle.this::clearSubscription)
                .compose(new ConnectionSharingAdapter());
```
3.log throw error
```
04-29 11:42:41.589 15014-15289/com.rokyinfo.rkbluetoothle_simple D/RxBle#Radio:   QUEUED RxBleRadioOperationConnect(237177820)
04-29 11:42:41.594 15014-15287/com.rokyinfo.rkbluetoothle_simple D/RxBle#Radio:  STARTED RxBleRadioOperationConnect(237177820)
04-29 11:42:41.618 15014-15014/com.rokyinfo.rkbluetoothle_simple D/BluetoothGatt: connect() - device: C0:27:15:09:A7:E6, auto: false
04-29 11:42:41.618 15014-15014/com.rokyinfo.rkbluetoothle_simple D/BluetoothGatt: registerApp()
04-29 11:42:41.619 15014-15014/com.rokyinfo.rkbluetoothle_simple D/BluetoothGatt: registerApp() - UUID=0a08fd3e-9dd9-4353-8a9d-597912724c34
04-29 11:42:41.621 15014-15014/com.rokyinfo.rkbluetoothle_simple E/BluetoothGatt: android.os.DeadObjectException
                                                                                      at android.os.BinderProxy.transactNative(Native Method)
                                                                                      at android.os.BinderProxy.transact(Binder.java:509)
                                                                                      at android.bluetooth.IBluetoothGatt$Stub$Proxy.registerClient(IBluetoothGatt.java:851)
                                                                                      at android.bluetooth.BluetoothGatt.registerApp(BluetoothGatt.java:752)
                                                                                      at android.bluetooth.BluetoothGatt.connect(BluetoothGatt.java:808)
                                                                                      at android.bluetooth.BluetoothDevice.connectGatt(BluetoothDevice.java:1519)
                                                                                      at android.bluetooth.BluetoothDevice.connectGatt(BluetoothDevice.java:1487)
                                                                                      at com.polidea.rxandroidble.internal.util.BleConnectionCompat.connectGattCompat(BleConnectionCompat.java:85)
                                                                                      at com.polidea.rxandroidble.internal.util.BleConnectionCompat.connectGatt(BleConnectionCompat.java:35)
                                                                                      at com.polidea.rxandroidble.internal.operations.RxBleRadioOperationConnect.connect(RxBleRadioOperationConnect.java:78)
                                                                                      at com.polidea.rxandroidble.internal.operations.RxBleRadioOperationConnect.run(RxBleRadioOperationConnect.java:67)
                                                                                      at com.polidea.rxandroidble.internal.radio.RxBleRadioImpl$$Lambda$4.call(Unknown Source)
                                                                                      at rx.Observable$27.onNext(Observable.java:7928)
                                                                                      at rx.observers.SafeSubscriber.onNext(SafeSubscriber.java:139)
                                                                                      at rx.internal.util.ScalarSynchronousObservable$ScalarSynchronousAction.call(ScalarSynchronousObservable.java:115)
                                                                                      at rx.internal.schedulers.ScheduledAction.run(ScheduledAction.java:55)
                                                                                      at android.os.Handler.handleCallback(Handler.java:815)
                                                                                      at android.os.Handler.dispatchMessage(Handler.java:104)
                                                                                      at android.os.Looper.loop(Looper.java:194)
                                                                                      at android.app.ActivityThread.main(ActivityThread.java:5779)
                                                                                      at java.lang.reflect.Method.invoke(Native Method)
                                                                                      at java.lang.reflect.Method.invoke(Method.java:372)
                                                                                      at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1004)
                                                                                      at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:799)
04-29 11:42:41.621 15014-15014/com.rokyinfo.rkbluetoothle_simple E/BluetoothGatt: Failed to register callback
```
#### Actual result
Error can not be thrown out
then queue will never release
can not step to
```
log("FINISHED", rxBleRadioOperation);
```
<br />
#### Expected result
throw the exception
<br />
Thank you very much to share, the library is great.  use Chinese I will say "大牛"

leeyc09, 2016-06-13:
Who knows how to fix this bug ?

dariuszseweryn, 2016-06-14:
Hello,
Excuse me for a long response time. This bug is on top of my priority list and it should be fixed in the snapshot release next week. I am on WWDC at the moment and I almost have no time during the evenings.
Best Regards

farmazon3000, 2016-06-29:
"15 days ago" "snapshot release next week" so where is it?

klemzy, 2016-07-04:
Is there planned snapshot release including this fix? Thanks!

dariuszseweryn, 2016-07-04:
I am a bit strained with projects at the moment. I found out that on some devices it is possible to successfully call `connectGatt()` even when the Bluetooth is off. I am checking different approaches to this situation.

dariuszseweryn, 2016-07-11:
This issue should be fixed in the current Snapshot release.

closed 2016-07-11 by dariuszseweryn