dariuszseweryn/RxAndroidBle #20: mockrxandroidble not available on gradle

fjanicki, 2016-05-12:
### Summary
Unable to install mockrxandroidble from gradle
#### Preconditions
Using jcenter repository
#### Steps to reproduce actual result
1.     add `compile 'com.polidea.rxandroidble:mockrxandroidble:1.0.1'` in build.gradle
2. Hit Sync
3. Error shows up saying it's failing to sync
#### Actual result
Error:Could not find com.polidea.rxandroidble:mockrxandroidble:1.0.1.
#### Expected result
Should install mockrxandroidble

mzgreen, 2016-05-13:
You're right, it will be pushed along with next release. Thanks for reporting.

lary-pipot, 2016-06-29:
Is mockrxandroidble available on gradle now?

uKL, 2016-06-29:
Not yet. We are currently under a heavy load in the company and there are some delays :( You can take that three classes and it should work.

https://github.com/Polidea/RxAndroidBle/tree/master/mockrxandroidble/src/main/java/com/polidea/rxandroidble/mockrxandroidble

dariuszseweryn, 2016-08-11:
Mock client is now available:

https://oss.sonatype.org/#nexus-search;quick~rxandroidble

Best Regards

closed 2016-08-11 by dariuszseweryn