dariuszseweryn/RxAndroidBle #21: Encountering MissingBackPressureException with lots of characteristic changes

pfives, 2016-05-16:
### Summary
First of all thank you for this library. You guys have done a seriously awesome job. 
<br />
Occasionally I am encountering a MissingBackPressureException when I create an observable that establishes a connection and listens for characteristic changes (`setupNotification(CHARACTERISITC)`)

In my use case there are lots of characteristic changes that we use to read data from the device but occasionally if there are lots more than normal it will crash with a MissingBackPressureException. 

Have you guys encountered this / known behavior? Is there a specific way you recommend dealing with this on the app side to avoid changes to the lib? Thanks again.

mzgreen, 2016-05-17:
This error means that your `Observable` emits items faster than you consume them. For now you can try to use one of the backpressure operators like `onBackpressureBuffer` and we'll check what we can do about it internally.

dariuszseweryn, 2016-06-06:
@pfives - as @mzgreen has proposed - you should either use onBackpressureBuffer or any other operator that suits best your needs (checkout https://github.com/ReactiveX/RxJava/wiki/Backpressure ). If we would have implemented any specific handling of the backpressure in the library someone other's usecases would not be covered.
Have I answered your question?

pfives, 2016-06-14:
@dariuszseweryn yes makes sense thank you. I have to dig into that a bit but seems pretty straightforward. Thanks for the help and the awesome lib!

closed 2016-07-04 by dariuszseweryn