dariuszseweryn/RxAndroidBle #24: Automatic disconnect after enabling notifications

Ibrahimhassan1, 2016-06-01:
### Summary
After connecting to a device then enabling the notifications, the device becomes disconnected automatically, how do I keep the device connected?
<br />
#### Preconditions
Device connected and notifications registered (I used the "CharacteristicOperationExampleActivity" provides with the samples)
<br />
#### Steps to reproduce actual result
1. Connect to a device
2. Enable notifications
3. Wait 10 seconds, then the device will be disconnected and will need to reconnect manually

#### Actual result
D/BluetoothGatt: onClientConnectionState() - status=8 clientIf=5 device=C4:xx:xx:xx:0A:9F
D/RxBle#BluetoothGatt: onConnectionStateChange newState=0 status=8
D/RxBle#Radio:   QUEUED RxBleRadioOperationDisconnect(785202367)
D/RxBle#Radio:  STARTED RxBleRadioOperationDisconnect(785202367)
D/BluetoothGatt: setCharacteristicNotification() - uuid: 713d0002-503e-4c75-ba94-3148f18d941e enable: false
D/RxBle#Radio:   QUEUED RxBleRadioOperationDescriptorWrite(433845284)
D/BluetoothManager: getConnectionState()
D/BluetoothManager: getConnectedDevices
D/RxBle#Radio: FINISHED RxBleRadioOperationDisconnect(785202367)
D/RxBle#Radio:  STARTED RxBleRadioOperationDescriptorWrite(433845284)
D/log: Notifications error: BleGattException{status=8, bleGattOperation=BleGattOperation{description='CONNECTION_STATE'}}
D/BluetoothGatt: close()
D/BluetoothGatt: unregisterApp() - mClientIf=5
D/RxBle#Radio: FINISHED RxBleRadioOperationDescriptorWrite(433845284)
<br />
#### Expected result
Connection is persisted and no need to reconnect.
<br />

dariuszseweryn, 2016-06-01:
Hello there,

Can you share which BLE device you're connecting to?
`D/BluetoothGatt: onClientConnectionState() - status=8 clientIf=5 device=C4:xx:xx:xx:0A:9F` -> status 8 is a timeout error as far as I know and it seems that your BLE device is stopping to keep the connection alive.
The error comes from the Android Bluetooth API and the RxAndroidBle cannot really do much about it.
Perhaps you're forgetting to set some configuration on your BLE device that would make it to continue the connection?

Best Regards

Ibrahimhassan1, 2016-06-01:
The device that I connect is a BlendMicro (http://redbearlab.com/blendmicro/).
Before I try RxAndroidBle I used to enable notifications using the following snippet and it didn't have any affect on the connection; so i doubt that the problem is with the Ble device:
// enable notifications
                BluetoothGattCharacteristic characteristicRX = gatt.getService(Constants.UUID_BLE_SHIELD_SERVICE).getCharacteristic(Constants.UUID_BLE_SHIELD_RX);
                setCharacteristicNotification(gatt, characteristicRX, true);

/**
     * Enables or disables notification on a give characteristic.
     *
     * @param characteristic Characteristic to act on.
     * @param enabled        If true, enable notification. False otherwise.
     */
    public void setCharacteristicNotification(BluetoothGatt gatt,
                                              BluetoothGattCharacteristic characteristic, boolean enabled) {
        if (gatt == null) {
            Log.w(LOG_TAG, "BluetoothAdapter not initialized");
            return;
        }

        if (characteristic != null) {

            final int charaProp = characteristic.getProperties();
            if ((charaProp | BluetoothGattCharacteristic.PROPERTY_READ) > 0) {
                // If there is an active notification on a characteristic, clear
                // it first so it doesn't update the data field on the user interface.
                if (gatt.equals(mGatt)) {
                    if (mNotifyCharacteristic != null) {
                        gatt.setCharacteristicNotification(
                                mNotifyCharacteristic, false);
                        mNotifyCharacteristic = null;
                    }
                }
                gatt.readCharacteristic(characteristic);
            }
            if ((charaProp | BluetoothGattCharacteristic.PROPERTY_NOTIFY) > 0) {
                if (gatt.equals(mGatt)) {
                    mNotifyCharacteristic = characteristic;
                }
                gatt.setCharacteristicNotification(characteristic, enabled);
                if (Constants.UUID_BLE_SHIELD_RX.equals(characteristic.getUuid())) {
                    BluetoothGattDescriptor descriptor = characteristic
                            .getDescriptor(UUID
                                    .fromString(Constants.CLIENT_CHARACTERISTIC_CONFIG));
                    descriptor
                            .setValue(BluetoothGattDescriptor.ENABLE_NOTIFICATION_VALUE);
                    gatt.writeDescriptor(descriptor);
                }
            }
        }
    }
As you mentioned Status 8 is a timeout error, what is the scenario that should happen after enabling the notifications?  does the remote device need to send something back to the app?
What configuration are you suggesting to check?

dariuszseweryn, 2016-06-06:
Correct me if I am wrong - you're not getting any notifications about the changes from the device even though you should. Am I right? If yes - could you also paste the code that you use for setting up the notification?

farmazon3000, 2016-07-01:
`status 8 is a timeout error as far as I know` where do you know that from?

dariuszseweryn, 2016-07-05:
I was digging through the documentation and issues that people encountered. I have double checked and found out that I was wrong. Actually status 8 is "Insufficient authorization". Probably you should have bonded device.

Here are some status codes that may be useful: https://android.googlesource.com/platform/external/bluetooth/bluedroid/+/android-4.4.4_r2.0.1/stack/include/gatt_api.h

Currently RxAndroidBle doesn't help in support of bonded devices / encrypted connections. Pull Requests are encouraged though. ;)

closed 2016-07-22 by dariuszseweryn