dariuszseweryn/RxAndroidBle #27: Sending the list of commands to device using RxAndroidBle

DKKovalev, 2016-06-03:
### Summary
<br />Hi, i'm trying to connect and send the list of commands, consisting of 16 bytes to BleDevice. How can i do this properly?
p.s. I'm pretty new to JavaRx

ardmn, 2016-06-03:
It is topical issue for me too. With Every connecting to ble device i need send "start" commnad and after that i can send next commands. In my case the ble device store some data for 7 days . To obtain the all data for one day I need send 3 commands. Therefore to get the data in the last 7 days, need to send a team of 21 (+ start command). Please can you give advice how solve this problem ? May be you know nice solution for commands queue that can easily integrated with rxAndroidBle lib? Thanks.

uKL, 2016-06-16:
I'm not sure if I get it correctly. Are you trying to write some data to a characteristic in a given order? If yes, you can just flatMap write operations and the order will be preserved.

uKL, 2016-06-20:
@DKKovalev @ardmn Could you follow up?

ardmn, 2016-06-20:
Sorry, @uKL I will try it tomorrow :)

KolinLoures, 2016-06-28:
@uKL Hi. I tried to find answer to this issue, but it still not work. May you give some helpful advice if it is not difficult for you :) 

`public void startWriteCommucation(final ArrayList<byte[]> b) {
        if (isConnected()){
                connectionObservable
                        .flatMap(new Func1<RxBleConnection, Observable<Observable<byte[]>>>() {
                            @Override
                            public Observable<Observable<byte[]>> call(RxBleConnection rxBleConnection) {
                                final List<Observable<byte[]>> list = new ArrayList<>();
                                for (byte[]  bytes: b){
                                    Log.e("MyTag", Arrays.toString(bytes));
                                    list.add(rxBleConnection.writeCharacteristic(BleDevice.characteristicWrite, bytes));
                                }
                                return Observable.from(list);
                            }
                        })
                        .flatMap(new Func1<Observable<byte[]>, Observable<byte[]>>() {
                            @Override
                            public Observable<byte[]> call(Observable<byte[]> observable) {
                                return observable;
                            }
                        })
                        .observeOn(AndroidSchedulers.mainThread())
                        .subscribe(
                                new Action1<byte[]>() {
                                    @Override
                                    public void call(byte[] bytes) {
                                        Log.e("MyTag2", Arrays.toString(bytes));
                                        view.setTextStatus("WRITE SUCCESS");
                                    }
                                },
                                new Action1<Throwable>() {
                                    @Override
                                    public void call(Throwable throwable) {
                                        view.setTextStatus("!!!FAIL WRITE!!!");
                                        throwable.printStackTrace();
                                    }
                                }
                        );
            }
    }`

This code works, and writes that it was successfull, but device does't responed.

dariuszseweryn, 2016-07-08:
I think this question should go to www.stackoverflow.com with #rxandroidble 
If I understood you correctly - you just want to perform some data exchange between the device and the app before something else will happen. In this situation you can write your own connecting function like:

```
public Observable<RxBleConnection> establishConnectionToMySpecificDevice(RxBleDevice device) {
  return rxBleDevice.establishConnection(this, false)
    .flatMap(rxBleConnection ->
        rxBleConnection.writeCharacteristic(uuid, bytes)
        .flatMap(bytes -> rxBleConnection.readCharacteristic(uuid))
        .map(bytes1 -> /*do something with read bytes*/ rxBleConnection)
    );
}
```

And use it wherever you need in the code?

KolinLoures, 2016-07-08:
@dariuszseweryn we have detail description of this issue on stackoverflow. 
[http://stackoverflow.com/questions/38250610/sending-the-list-of-commands-to-device-using-rxandroidble-rxjava](url)

closed 2016-07-22 by dariuszseweryn