dariuszseweryn/RxAndroidBle #29: Question:

zzt93, 2016-06-08:
### Some questions
<br />

-  What the best practice to unsubscribe scanning if I want to connect to many devices and communicate ?
in Activity#onDestory?
or like the following code

```
Subscription subscribe = rxBleClient.scanBleDevices().subscribe(new Action1<RxBleScanResult>() {
    @Override
    public void call(RxBleScanResult rxBleScanResult) {
        // connection made event source
        Observable<RxBleConnection> connectionSrc = rxBleScanResult
                .getBleDevice()
                .establishConnection(activity, false);

        // when found the device, stop scanning
        subscribe.unsubscribe();
   }
}
```

zzt93, 2016-06-08:
- What the best practice to _unsubscribe connection_ if I want to connect to many devices and communicate ? in Activity#onDestory? or like the following code
```
Observable<RxBleConnection> connectionSrc = rxBleScanResult
        .getBleDevice()
        .establishConnection(activity, false);

// 6/8/16  when found the device, stop scanning
subscribe.unsubscribe();

// TODO: 6/8/16 when un-subscribe this
Subscription conSubscription = connectionSrc
        .flatMap(new Func1<RxBleConnection, Observable<byte[]>>() {
            @Override
            public Observable<byte[]> call(final RxBleConnection rxBleConnection) {
                return rxBleConnection.readCharacteristic(BLE.BLE_CHAT_UUID);
            }
        })
        .subscribe(new Action1<byte[]>() {
            @Override
            public void call(byte[] bytes) {
                try {
                    MsgHistory.addReceived(Message.getFromBytes(bytes));
                } catch (UnsupportedEncodingException e) {
                    e.printStackTrace();
                }
                Log.d(CANONICAL_NAME, new String(bytes));
               
                **conSubscription.unsubscribe();**
            }
        });
```

zzt93, 2016-06-08:
- read write combined vs simple read then simple write
The following code is blog's code for read/write combined.
```
device.establishConnection(context, false)
    .flatMap(rxBleConnection -> rxBleConnection.readCharacteristic(characteristicUuid)
        .doOnNext(bytes -> {
            // Process read data.
        })
        .flatMap(bytes -> rxBleConnection.writeCharacteristic(characteristicUuid, bytesToWrite))
     ).subscribe(writeBytes -> {
        // Written data.
    });
```
What is the difference between the following code:
```
Observable<RxBleConnection> connectionSrc = device.establishConnection(activity, false);             
connectionSrc
    .flatMap(rxBleConnection -> rxBleConnection.readCharacteristic(characteristicUUID))
    .subscribe(characteristicValue -> {
        // Read characteristic value.
    });
connectionSrc
    .flatMap(rxBleConnection -> rxBleConnection.writeCharacteristic(characteristicUUID, bytesToWrite))
    .subscribe(characteristicValue -> {
        // Characteristic value confirmed.
    });
```

dariuszseweryn, 2016-06-08:
The answer is to unsubscribe from the scanning as soon as you're no longer interested in it. For instance if you're interested in connecting to a fixed amount of devices that are meeting your requirements:
```
CompositeSubscription compositeSubscription = new CompositeSubscription

Subscription scanSubscription = rxBleClient.scanBleDevices()
  .filter(scanResult -> /* filter the devices you're interested in*/)
  .take(/* number of devices you want to scan */) // this operator will automatically unsubscribe from scanSubscription once the number of scanResults will be taken
  .map(scanResult -> scanResult.getBleDevice())
  .subscribe(bleDevice -> connectToDevice(bleDevice))
compositeSubscription.add(scanSubscription)
```

where:
```
void connectToDevice(RxBleDevice device) {
  Subscription connectFlowSubscription = device.establishConnection(context, false)
    .flatMap(connection -> /* do what you need to do */)
    .subscribe()
  compositeSubscription.add(connectFlowSubscription)
}
```

and you should clear the compositeSubscription when you're no longer using the app. For instance onPause of the Activity.

closed 2016-06-09 by dariuszseweryn