dariuszseweryn/RxAndroidBle #30: Suggested way of unsubscribing connection with multiple devices

dariuszseweryn, 2016-06-08:
Originally posted by @zzt93 in http://github.com/Polidea/RxAndroidBle/issues/29#issuecomment-224528530

> - What the best practice to _unsubscribe connection_ if I want to connect to many devices and communicate ? in Activity#onDestory? or like the following code
> ```
> Observable<RxBleConnection> connectionSrc = rxBleScanResult
>         .getBleDevice()
>         .establishConnection(activity, false);
> 
> // 6/8/16  when found the device, stop scanning
> subscribe.unsubscribe();
> 
> // TODO: 6/8/16 when un-subscribe this
> Subscription conSubscription = connectionSrc
>         .flatMap(new Func1<RxBleConnection, Observable<byte[]>>() {
>             @Override
>             public Observable<byte[]> call(final RxBleConnection rxBleConnection) {
>                 return rxBleConnection.readCharacteristic(BLE.BLE_CHAT_UUID);
>             }
>         })
>         .subscribe(new Action1<byte[]>() {
>             @Override
>             public void call(byte[] bytes) {
>                 // read bytes              
>                 **conSubscription.unsubscribe();**
>             }
>         });
> ```

dariuszseweryn, 2016-06-08:
Unsubscribing from connection subscription should take place once you're no longer interested in connecting.
For instance - sometimes you're interested in showing the device state in the Activity. In this situation you should unsubscribe from the `conSubscription` in `onPause()` method. 
Sometimes the application needs to execute a specific flow when connecting to the device and then it's done - let's say we want to read a value from a characteristic and disconnect:
```
rxBleDevice.establishConnection(context, false)
  .flatMap(connection -> connection.readCharacteristic(characteristicUuid))
  .take(1)
  .subscribe(readBytes -> /* do your thing with readBytes */)
```
In this situation the `establishConnection()` subscription will be automatically unsubscribed by the `take(1)` operator. We can do it this way if we're not updating anything UI related inside an Activity which can be destroyed b the time the `readBytes` will be available,

zzt93, 2016-06-08:
The `take(1)` correspond to another device's one `writeCharacteristic()` invocation, no matter how long the characteristic is?

dariuszseweryn, 2016-06-08:
@zzt93 - Excuse me but I cannot really follow you. Could you rephrase the question?

The RxAndroidBle library is for client side only. So the client knows what to read from the server or it waits for read notifications usually. The implementation of the server may vary greatly so it is hard for me to know what you have in mind.

Basically `.take(1)` takes exactly one event from the above observable and unsubscribes from it. Unsubscribing from `.establishConnection()` closes the connection. `.readCharacteristic()` is returning an Observable that will emit only once.

Does it answer your question?

zzt93, 2016-06-09:
Say we have two devices all use RxAndroidBle to connect to each other. One device call `writeCharacteristic()` twice during connection, the other device then call  `readCharacteristic()` to read. Will the second device observe one event or two?

dariuszseweryn, 2016-06-09:
You cannot directly connect two Android devices to each other because the RxAndroidBle is a client library and BLE connection is a client-server connection. Android has introduced BLE server API though it is not covered in the library.

Either way both `readCharacteristic()` and `writeCharacteristic()` are client initiated, one-time, actions.

zzt93, 2016-06-09:
Ok, I know. Thank you.

closed 2016-06-09 by dariuszseweryn