dariuszseweryn/RxAndroidBle #31: Suggested way of using connection

dariuszseweryn, 2016-06-08:
Originally posted by @zzt93 in https://github.com/Polidea/RxAndroidBle/issues/29#issuecomment-224529920

> - read write combined vs simple read then simple write
> The following code is blog's code for read/write combined.
> ```
> device.establishConnection(context, false)
>     .flatMap(rxBleConnection -> rxBleConnection.readCharacteristic(characteristicUuid)
>         .doOnNext(bytes -> {
>             // Process read data.
>         })
>         .flatMap(bytes -> rxBleConnection.writeCharacteristic(characteristicUuid, bytesToWrite))
>      ).subscribe(writeBytes -> {
>         // Written data.
>     });
> ```
> What is the difference between the following code:
> ```
> Observable<RxBleConnection> connectionSrc = device.establishConnection(activity, false);             
> connectionSrc
>     .flatMap(rxBleConnection -> rxBleConnection.readCharacteristic(characteristicUUID))
>     .subscribe(characteristicValue -> {
>         // Read characteristic value.
>     });
> connectionSrc
>     .flatMap(rxBleConnection -> rxBleConnection.writeCharacteristic(characteristicUUID, bytesToWrite))
>     .subscribe(characteristicValue -> {
>         // Characteristic value confirmed.
>     });
> ```

dariuszseweryn, 2016-06-08:
In the first block what would happen is:
1. rxBleConnection would be established
2. once the rxBleConnection will be available the `readCharacteristic()` will trigger
3. once the `readCharacteristic()` will finish the `writeCharacteristic()` will start
4. `subscribe()` will get the `writtenBytes`

With the second block - with the current implementation (which will try to share the connection between both subscription) it is uncertain wether the `readCharacteristic()` or `writeCharacteristic()` will be triggered first. In such a simple example there is no big difference but if the communication would be stateful in some way the other subscription could change the state of the first one.

My suggestion is to use the first block as there you can execute the flow exactly as you want.

zzt93, 2016-06-08:
In the first case, you mean if I can't read some characteristic from other device, I can't write?

dariuszseweryn, 2016-06-08:
Usually if you can't read from a characteristic that means that the connection is broken and you have to establish a new one. So yes. If a `readCharacteristic()` would fail you won't be able to `writeCharacteristic`

zzt93, 2016-06-09:
But if the device in the other side doesn't write any characteristic, what will I read?

dariuszseweryn, 2016-06-09:
I think that you should learn more about the Bluetooth Low Energy. 
The characteristic is a kind of a property on the server device. What it will serve during `readCharacteristic()` is up to the server.
I am sorry but this is not really question about the library itself but more about how the BLE works. I suggest you to check www.google.com or ask a question on www.stackoverflow.com .

Best Regards

closed 2016-06-09 by dariuszseweryn

zzt93, 2016-06-09:
Thanks for your suggestion.