dariuszseweryn/RxAndroidBle #32: Needing thread hopping?

zzt93, 2016-06-09:
As far as I understand the asynchronous event, the observer's action will always be executed in another thread, right? So If I want to update ui, I have to use method like `activity.runOnUiThread()`, right?

```
.subscribe(new Action1<byte[]>() {
    @Override
    public void call(final byte[] bytes) {
        activity.runOnUiThread(
                new Runnable() {
                    @Override
                    public void run() {
                        // update ui
                    }
                }
        );
    }
});
```

dariuszseweryn, 2016-06-09:
The observing action will be executed on arbitrary thread unless you will specify `.observeOn(Scheduler)`. This is exactly in line of how rxJava works.

More information you can get here: http://reactivex.io/intro.html

Best Regards

closed 2016-06-09 by dariuszseweryn

6bangs, 2016-06-27:
> The observing action will be executed on arbitrary thread unless you will specify .observeOn(Scheduler). This is exactly in line of how rxJava works.

Is this true? As far as I can tell, an `Observer` resulting from [`RxBleRadioImple.queue(RxBleRadioOperation)`](https://github.com/Polidea/RxAndroidBle/blob/e54d8dd0e510827dcec585d6c49eb4d74cce1e42/rxandroidble/src/main/java/com/polidea/rxandroidble/internal/radio/RxBleRadioImpl.java#L52) will _not_ have `OperatorObserveOn` applied to it, and should execute on the same thread that calls `subscribe`. From the [ReactiveX docs](http://reactivex.io/documentation/scheduler.html):

> By default, an Observable and the chain of operators that you apply to it will do its work, and will notify its observers, on the same thread on which its Subscribe method is called.

6bangs, 2016-06-27:
Note that the actual BLE operation will [queue up on a new arbitrary thread](https://github.com/Polidea/RxAndroidBle/blob/e54d8dd0e510827dcec585d6c49eb4d74cce1e42/rxandroidble/src/main/java/com/polidea/rxandroidble/internal/radio/RxBleRadioImpl.java#L17) and execute on the main thread (as required by Android) - but from an Rx perspective, observing and subscribing happens on the same thread that called subscribe.