dariuszseweryn/RxAndroidBle #33: IncompatibleClassChangeError Exception when creating client

SkaveRat, 2016-06-16:
### Summary

IncompatibleClassChangeError Exception when creating client:

`rxBleClient = RxBleClient.create(getApplicationContext());`

#### Preconditions

N/A
#### Steps to reproduce actual result
1. Create an activity
2. Try Creating a client
#### Actual result
```
java.lang.IncompatibleClassChangeError: The method 'void com.polidea.rxandroidble.internal.radio.RxBleRadioImpl.com_polidea_rxandroidble_internal_radio_RxBleRadioImpl_lambda$new$73()' was expected to be of type direct but instead was found to be of type virtual (declaration of 'java.lang.reflect.ArtMethod' appears in /system/framework/core-libart.jar)
                                                       at com.polidea.rxandroidble.internal.radio.RxBleRadioImpl.access$lambda$0(Unknown)
                                                       at com.polidea.rxandroidble.internal.radio.RxBleRadioImpl$$Lambda$1.run(Unknown)
                                                       at java.lang.Thread.run(Thread.java:818)
```

#### Expected result

Getting a client

--------------

I've copied the code from the first example, pretty much 1:1, exception creation of the client. If I do it in the `onCreate` method of the activity, I get the Exception.

Is it not possible to get the client directly from inside the Activity?

uKL, 2016-06-16:
Hey, are you using proguard?

mzgreen, 2016-06-16:
@SkaveRat could you provide information about your device and Android version?

SkaveRat, 2016-06-16:
@uKL no

@mzgreen BlackPhone2 with 5.1.1, Kernel 3.10.49.

Can later try with a different phone

SkaveRat, 2016-06-19:
Other devices have the same problem.

After playing around some more, it appears that retrolambda is a required dependency.
Changed around a lot of stuff, but only after adding retrolambda it finally worked:

`# app/build.gradle`
```
apply plugin: 'me.tatarka.retrolambda'

buildscript {
    repositories {
        mavenCentral()
    }

    dependencies {
        classpath 'me.tatarka:gradle-retrolambda:3.2.5'
    }
}
```

uKL, 2016-06-20:
Hmm, what is your compile sdk, target, min sdk, build tools version and java version in gradle?

SkaveRat, 2016-06-20:
I've tried pretty much any combination.

The versions I was on while discovering the retrolambda solutions were compile/build 23, min 18 and I think java 1.7.

After it worked, I upgraded successfully to the following:
compile: 24
buildtools: 24
minSdk: 18
Java: 1.8

As soon as I remove retrolambda from these settings, the SDK tells me to enable Jack (which I also tried unsuccessfully)

Today my phone updated to 6.0.1, so that should be considered, if I should test other things

lavenj, 2016-09-20:
@uKL love this project, but it seems like it's tied to RetroLambda. Any chance/timeline for Jack compatibility?

dariuszseweryn, 2017-02-02:
Version `1.2.0-SNAPSHOT` is now compatible with Jack.

closed 2017-02-02 by dariuszseweryn