dariuszseweryn/RxAndroidBle #34: Unable to add service to advertisement payload using the mock RxBleClient/RxBleDevice

pfives, 2016-06-20:
### Summary
<br />
Using the MockRxAndroidBle class to mock bluetooth devices, unable to make an added service show in the advertisement payload.

#### Steps to reproduce actual result
1) Create a mock RxBleDevice with a name, mac address, rssi, and a service with a given UUID
2) Add that device to a mock RxBleClient
3) Scan for that device by the UUID `rxBleClient.scanDevices(UUID)
#### Actual result
no devices are found
#### Expected result
the device that was added to the mock RxBleClient is found. This works as expected if I call `rxBleClient.scanDevices()` with no UUID. So I believe the issue is the service I am adding is not being used in the advertisement payload, is there any way to accomplish this? Thanks you!

uKL, 2016-06-20:
I did quick test for this issue

```
 def "should return filtered BluetoothDevice"() {
        given:
        def serviceUUID = UUID.fromString("00001234-0000-0000-8000-000000000000")
        def rxBleClient = new RxBleClientMock.Builder()
                .addDevice(
                new RxBleClientMock.DeviceBuilder()
                        .deviceMacAddress("AA:BB:CC:DD:EE:FF")
                        .deviceName("TestDevice")
                        .scanRecord("some scan record bytes".getBytes())
                        .rssi(42)
                        .addService(serviceUUID, emptyList()
                ).build()
        ).build();
        def testSubscriber = TestSubscriber.create()

        when:
        rxBleClient.scanBleDevices(serviceUUID)
                .take(1)
                .map { scanResult -> scanResult.getBleDevice().getMacAddress() }
                .subscribe(testSubscriber)

        then:
        testSubscriber.assertValue("AA:BB:CC:DD:EE:FF")
    }
```

Am I missing something?

1. Added service does not affect advertisement payload in the mock
2. Advertisement payload is somewhat transparent for mock but required (see above)
3. Mock supports filtering.

Could you provide a code snippet with your mock?

pfives, 2016-06-20:
Sure can.

`static final String DEVICE_MAC_ADDRESS = "AA:BB:CC:DD:EE:FF";
static final UUID AD_UUID = UUID.fromString("cd336c84-0b2f-1136-b512-3e1d05defe78");
static final String DEVICE_NAME = "test1236";
static final int RSSI = 42;

RxBleDevice rxBleDevice = new RxBleClientMock.DeviceBuilder().deviceMacAddress(DEVICE_MAC_ADDRESS)
                .deviceName(DEVICE_NAME).rssi(RSSI).scanRecord(getScanRecord()).addService(AD_UUID, Collections.emptyList()).build();

RxBleClient rxBleClient = new RxBleClientMock.Builder().addDevice(rxBleDevice).build();

TestSubscriber<RxBleScanResult> testSubscriber = new TestSubscriber<>();

//test passes with this call
rxBleClient.scanBleDevices().subscribe(testSubscriber);

//test fails with this call
//rxBleClient.scanBleDevices(AD_UUID).subscribe(testSubscriber)

List<RxBleScanResult> rxBleScanResults = testSubscriber.getOnNextEvents();
Assert.assertTrue(rxBleScanResults.size() == 1);`

uKL, 2016-06-20:
Ok, fun fact. Mock was not released to central yet (strange, we will do a release soon then @dariuszseweryn gets back from WWDC).

How did you build mocking support? It looks like we had this issue in earlier version but in master it should be fixed already.

pfives, 2016-06-20:
Shoot my bad. I just grabbed the files directly from the repo because I couldn't get them from the gradle script (eventually saw this issue https://github.com/Polidea/RxAndroidBle/issues/20). I am using these 3 files located here: https://github.com/Polidea/RxAndroidBle/tree/master/mockrxandroidble/src/main/java/com/polidea/rxandroidble/mockrxandroidble

uKL, 2016-06-23:
Master is the newest one and it should work. Those three files is everything you need.

I ran your test and guess what... it's passing :) What's in your `getScanRecord()` method?

pfives, 2016-06-23:
It was a method that returned a specific `byte[]` that mimicked an actual scan record from a device. I replaced it with `"some scan record bytes".getBytes()` and still have the same issue. Passes with no UUID specified but fails when using one. 

I noticed there were some updates on the `mock-indications` that are newer than master, could those be helpful?

mzgreen, 2016-08-12:
@pfives I'm not able to reproduce this issue. I tried to run your sample and both calls:

`rxBleClient.scanBleDevices().subscribe(testSubscriber);`
and
`rxBleClient.scanBleDevices(AD_UUID).subscribe(testSubscriber);`

return correct value. Please recheck if you can reproduce the issue in the newest version of the library.

dariuszseweryn, 2016-10-25:
Since the issue was not reproducible by @mzgreen and there is no other information. I am closing this.

closed 2016-10-25 by dariuszseweryn