dariuszseweryn/RxAndroidBle #36: Scan stops briefly after stop

klemzy, 2016-06-22:
### Summary
Another issue that I have noticed is that when perfoming scan, sometimes scan just drops. I checked this by hooking with `doOnUnsubscribe` and noticed it gets called really quickly (few seconds) after starting scan.

Did I maybe miss something in implementation?

Kind Regards

uKL, 2016-06-23:
Could you provide a code snippet, Android version and device?

klemzy, 2016-06-23:
`rxBleClient.scanBleDevices()
                        .observeOn(AndroidSchedulers.mainThread())
                        .doOnUnsubscribe(() -> Timber.i("Scan unsubscribe"))
                        .subscribe((rxBleScanResult -> {                           
                            }
                        }), (throwable -> {
                            if (throwable instanceof BleScanException) {
                                String msg = getExceptionMessage((BleScanException) throwable);
                                scanCallback.scanResult(new ScanResult(msg));
                            }
                        }))`

Android N Preview 4, Nexus 6P

uKL, 2016-06-23:
Could you also turn on logging and provide ADB Logs? Best will be without package filtering because we may see system bluetooth logs as well.

SkaveRat, 2016-06-25:
I have similar problems. The device scans for half a second (sometimes even less) and stops.

If I mash the scan button multiple times I find the device I want, but it would be a lot more convinient to e possible to say "scan for X seconds"

If it's relevant: I have another BLE device connected already (my watch) 

```
06-25 21:41:03.523 17384-25926/? D/RxBle#Radio:   QUEUED RxBleRadioOperationScan(239854725)
06-25 21:41:03.524 17384-25942/? D/RxBle#Radio:  STARTED RxBleRadioOperationScan(239854725)
06-25 21:41:03.538 6124-6365/? D/BtGatt.GattService: registerClient() - UUID=0b1fbda3-6807-4c82-8c4d-fb4b95d3ffbe
06-25 21:41:03.541 6124-6224/? D/BtGatt.GattService: onClientRegistered() - UUID=0b1fbda3-6807-4c82-8c4d-fb4b95d3ffbe, clientIf=5
06-25 21:41:03.541 17384-25936/? D/BluetoothLeScanner: onClientRegistered() - status=0 clientIf=5
06-25 21:41:03.541 6124-8778/? D/BtGatt.GattService: start scan with filters
06-25 21:41:03.545 6124-6239/? D/BtGatt.ScanManager: handling starting scan
06-25 21:41:03.546 17384-25942/? D/RxBle#Radio: FINISHED RxBleRadioOperationScan(239854725)
06-25 21:41:03.563 6124-6224/? D/BtGatt.GattService: onScanFilterEnableDisabled() - clientIf=5, status=0, action=1
06-25 21:41:03.563 6124-6224/? D/BtGatt.ScanManager: callback done for clientIf - 5 status - 0
06-25 21:41:03.563 6124-6239/? D/BtGatt.ScanManager: configureFilterParamter 500 10000 1 0
06-25 21:41:03.570 6124-6224/? D/BtGatt.GattService: onScanFilterParamsConfigured() - clientIf=5, status=0, action=0, availableSpace=15
06-25 21:41:03.570 6124-6224/? D/BtGatt.ScanManager: callback done for clientIf - 5 status - 0
06-25 21:41:03.570 6124-6239/? D/BtGatt.ScanManager: configureRegularScanParams() - queue=1
06-25 21:41:03.570 6124-6239/? D/BtGatt.ScanManager: configureRegularScanParams() - ScanSetting Scan mode=2 mLastConfiguredScanSetting=-2147483648
06-25 21:41:03.570 6124-6239/? D/BtGatt.ScanManager: configureRegularScanParams - scanInterval = 8000configureRegularScanParams - scanWindow = 8000
06-25 21:41:03.571 6124-6224/? D/BtGatt.GattService: onScanParamSetupCompleted : 0
06-25 21:41:03.623 17384-25926/? I/MYAPP:DeviceScanActivity: 65:C6:F0:28:F6:F9
06-25 21:41:03.626 6124-6365/? D/BtGatt.GattService: stopScan() - queue size =1
06-25 21:41:03.626 6124-6239/? D/BtGatt.ScanManager: stop scan
06-25 21:41:03.626 6124-8778/? D/BtGatt.GattService: unregisterClient() - clientIf=5
06-25 21:41:03.639 6124-6224/? D/BtGatt.GattService: onScanFilterParamsConfigured() - clientIf=5, status=0, action=1, availableSpace=16
06-25 21:41:03.640 6124-6224/? D/BtGatt.ScanManager: callback done for clientIf - 5 status - 0
06-25 21:41:03.640 6124-6239/? D/BtGatt.ScanManager: configureRegularScanParams() - queue=0
06-25 21:41:03.640 6124-6239/? D/BtGatt.ScanManager: configureRegularScanParams() - ScanSetting Scan mode=-2147483648 mLastConfiguredScanSetting=2
06-25 21:41:03.640 6124-6239/? D/BtGatt.ScanManager: configureRegularScanParams() - queue emtpy, scan stopped
```

SkaveRat, 2016-06-25:
Okay, here's how to reproduce (and a fix, at least for me)

```
final String deviceName = rxBleScanResult.getBleDevice().getName(); // this can be null!

logger.i(deviceName); // this stops the scan without stacktrace, if the name is null

logger.i(String.format("name: %s", deviceName)); // this doesn't
```

klemzy, 2016-07-04:
Awesome. Had the same problem. This also fixed it for me.

closed 2016-07-06 by dariuszseweryn

uKL, 2016-07-06:
This is true. Name advertisement is optional.