dariuszseweryn/RxAndroidBle #40: Samsung Galaxy S5 Unhandled exception in callback

farmazon3000, 2016-07-01:
On Samsung Galaxy S5 I make:
```
rxBleDevice.establishConnection(context, false) // <-- autoConnect flag
    .doOnError(throwable -> {
        Log.d(TAG, "rxBleConnection doOnError: " + throwable);
    })
    .subscribe(rxBleConnection -> {
        Log.d(TAG, "rxBleConnection: " + rxBleConnection);
    });

rxBleDevice.observeConnectionStateChanges()
    .doOnError(throwable -> {
        Log.d(TAG, "connectionState doOnError: " + throwable);
    })
    .subscribe(connectionState -> {
        Log.d(TAG, "connectionState: " + connectionState);
    });
```
I see in logs
```
BluetoothGatt: onClientConnectionState() - status=8 clientIf=8 device=XX:XX:XX:XX:XX:XX
...
W/BluetoothGatt: Unhandled exception in callback
    rx.exceptions.OnErrorNotImplementedException
    at rx.internal.util.InternalObservableUtils$ErrorNotImplementedAction.call(InternalObservableUtils.java:386)
    at rx.internal.util.InternalObservableUtils$ErrorNotImplementedAction.call(InternalObservableUtils.java:383)
    at rx.internal.util.ActionSubscriber.onError(ActionSubscriber.java:44)
    at rx.observers.SafeSubscriber._onError(SafeSubscriber.java:157)
    at rx.observers.SafeSubscriber.onError(SafeSubscriber.java:120)
    at rx.internal.operators.OperatorDoOnEach$1.onError(OperatorDoOnEach.java:72)
    at rx.observers.Subscribers$5.onError(Subscribers.java:230)
    at rx.observers.Subscribers$5.onError(Subscribers.java:230)
    at rx.internal.operators.OperatorDoOnEach$1.onError(OperatorDoOnEach.java:72)
    at rx.observers.Subscribers$5.onError(Subscribers.java:230)
    at rx.observers.Subscribers$5.onError(Subscribers.java:230)
    at rx.observers.Subscribers$5.onError(Subscribers.java:230)
    at rx.internal.operators.OperatorMerge$MergeSubscriber.reportError(OperatorMerge.java:268)
    at rx.internal.operators.OperatorMerge$MergeSubscriber.checkTerminate(OperatorMerge.java:812)
    at rx.internal.operators.OperatorMerge$MergeSubscriber.emitLoop(OperatorMerge.java:573)
    at rx.internal.operators.OperatorMerge$MergeSubscriber.emit(OperatorMerge.java:562)
    at rx.internal.operators.OperatorMerge$InnerSubscriber.onError(OperatorMerge.java:846)
    at rx.internal.operators.OperatorMerge$MergeSubscriber.reportError(OperatorMerge.java:268)
    at rx.internal.operators.OperatorMerge$MergeSubscriber.checkTerminate(OperatorMerge.java:812)
    at rx.internal.operators.OperatorMerge$MergeSubscriber.emitLoop(OperatorMerge.java:573)
    at rx.internal.operators.OperatorMerge$MergeSubscriber.emit(OperatorMerge.java:562)
    at rx.internal.operators.OperatorMerge$MergeSubscriber.onError(OperatorMerge.java:278)
    at rx.internal.operators.OperatorMap$MapSubscriber.onError(OperatorMap.java:85)
    at rx.internal.operators.OperatorDoOnEach$1.onError(OperatorDoOnEach.java:72)
    at rx.internal.operators.OperatorFilter$FilterSubscriber.onError(OperatorFilter.java:87)
    at rx.internal.operators.OperatorMerge$MergeSubscriber.reportError(OperatorMerge.java:268)
    at rx.internal.operators.OperatorMerge$MergeSubscriber.checkTerminate(OperatorMerge.java:812)
    at rx.internal.operators.OperatorMerge$MergeSubscriber.emitLoop(OperatorMerge.java:573)
    at rx.internal.operators.OperatorMerge$MergeSubscriber.emit(OperatorMerge.java:562)
    at rx.internal.operators.OperatorMerge$InnerSubscriber.onError(OperatorMerge.java:846)
    at rx.internal.operators.NotificationLite.accept(NotificationLite.java:148)
    at rx.subjects.SubjectSubscriptionManager$SubjectObserver.emitNext(SubjectSubscriptionManager.java:255)
    at rx.subjects.BehaviorSubject.onError(BehaviorSubject.java:141)
    at com.polidea.rxandroidble.internal.connection.RxBleGattCallback.propagateStatusErrorIfGattErrorOccurred(RxBleGattCallback.java:245)
    at com.polidea.rxandroidble.internal.connection.RxBleGattCallback.access$100(RxBleGattCallback.java:26)
    at com.polidea.rxandroidble.internal.connection.RxBleGattCallback$1.onConnectionStateChange(RxBleGattCallback.java:62)
    at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:182)
    at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:70)
    at android.os.Binder.execTransact(Binder.java:446)
    Caused by: BleGattException{status=8, bleGattOperation=BleGattOperation{description='CONNECTION_STATE'}}
    at com.polidea.rxandroidble.internal.connection.RxBleGattCallback.propagateStatusErrorIfGattErrorOccurred(RxBleGattCallback.java:245) 
    at com.polidea.rxandroidble.internal.connection.RxBleGattCallback.access$100(RxBleGattCallback.java:26) 
    at com.polidea.rxandroidble.internal.connection.RxBleGattCallback$1.onConnectionStateChange(RxBleGattCallback.java:62) 
    at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:182) 
    at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:70) 
    at android.os.Binder.execTransact(Binder.java:446) 
```

farmazon3000, 2016-07-01:
I've tested it on Moto G and the problem is the same so it's not Samsung related. `Log.d(TAG, "connectionState: " + connectionState);` is still called (`onNext`) but I think library should call `onError` here to allow one of those https://github.com/ReactiveX/RxJava/wiki/Error-Handling-Operators.

dariuszseweryn, 2016-07-01:
I see that in your code you do not handle `.subscribe()` `onError()`. From the Javadoc you can read that `.establishConnection()` may emit errors and thus in `.subscribe()` you should handle the error.

Part of the `.establishConnection()` Javadoc.
```
     * @throws BleDisconnectedException emitted when the BLE link has been disconnected either when the connection was already established
     *                                  or was in pending connection state. This occurs when the connection was released as a
     *                                  part of expected behavior (with {@link android.bluetooth.BluetoothGatt#GATT_SUCCESS} state).
     * @throws BleGattException         emitted when the BLE link has been interrupted as a result of an error. The exception contains
     *                                  detailed explanation of the error source (type of operation) and the code proxied from
     *                                  the Android system.
```

You should use:
```
.subscribe(
  result -> {},
  throwable -> { /*handle throwable*/ }
)
```

farmazon3000, 2016-07-01:
I think you are wrong, I DO handle `onError` in my snippet:
```
.doOnError(throwable -> {
        Log.d(TAG, "connectionState doOnError: " + throwable);
    })
```

uKL, 2016-07-01:
This does not consume errors, they are passed to subscriber anyway. You should handle them in subscribe method or use one of onErrorReturn/onErrorResume

uKL, 2016-07-02:
@farmazon3000 for further reference go to:
http://stackoverflow.com/questions/30250467/rxjava-what-is-difference-between-callbacks-in-doonerrorcallback-and-subscr

farmazon3000, 2016-07-04:
Thank you guys for pointing it out.

closed 2016-07-04 by farmazon3000