dariuszseweryn/RxAndroidBle #41: Performance

ecdpalma, 2016-07-02:
Has anyone noticed performance degradation by using the library? Or is it a common issue? I've been achieving, on average, 1.5kbits/s when exchanging data with another Android device.

dariuszseweryn, 2016-07-11:
No detailed tests were performed. 
Usually BLE is not about sending a bulk of data quickly but adding additional layers of abstraction and using RxJava obviously adds some computation work for the processor and memory allocation.
If I would have enough time I would perform some comparison test between specialised implementation of Android BLE API vs RxAndroidBle in firmware update scenario.

maoueh, 2016-12-14:
We are currently using RxAndroidBle to flash the firmware of a device over the air. The firmware is about 300kb. Nice thing is that I also have a proof of concept app for the same flashing process using pure Android API.

These are not real experiments but simple rough estimate. Using RxAndroidBle, it takes up to 3 minutes to transfer everything as for native API, it's about 40s.

The big bottleneck in the library right now for heavy data transfer is that all write operations (in fact simply all operations) are done on the UI thread. This make the transfer process really longer than expected as it's fighting for the UI thread resources.

We noted a minimal 2x increase when just shutting down the screen of the device for example.

I read in the code that you're doing this because Samsung 4.3 fails when `connectGatt` operation is not performed on the UI thread. Do you think it would be possible to make a conditional to only run the operations on the UI thread for a subset of all devices?

uKL, 2017-04-06:
#165

uKL, 2017-04-06:
As a result of this issue, there will be a report of write operation benchmark (16kB) with nRF based mock device:

1. Write with native Android APIs (write with **BluetoothGatt** and direct use of **BluetoothGattCallback**)
1. Write with **LongWriteOperationBuilder**
1. Looped writeCharacteristic operation of this library
1. Custom operation with write with **BluetoothGatt** and response with **RxBleGattCallback**.
1. Custom operation with write with **BluetoothGatt** and direct use of **BluetoothGattCallback**.

dariuszseweryn, 2017-10-02:
Now I finally made some performance tests for sending out as much data as possible using various approaches.

## Specification
Peripheral: `nRF51822` with `SoftDevice S110 8.0.0`
Connection Interval: `11.25 ms`
MTU: `23` (default—nRF51 does not support different MTUs)

## Test Algorithm
1. The central (phone) is sending 19 packets with 20 bytes each to the peripheral. The first byte contains the index of the packet.
2. After every 19 packets the peripheral sends a notification to the central with indexes of the packets it received. The first byte contains the index of the response. (i.e. the first response packet would contain [0, 0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18], second [1, 19, 20, 21, ...])
3. The central receives the notification and repeats from the first step.

## Speed calculation
Every response notification (2. step) is timestamped. All responses are buffered for 60 seconds. We take the last notification timestamp and the first notification timestamp and calculate the `time difference`. The resulting speed is `((number of received notifications - 1) * number of packets for each notification (19) * number of bytes in each packet (20)) / time difference`

## Code:
```
Observable.concat(
        perform(device, this::unoptimizedSendUuid).map(aFloat -> "Unoptimized UUID: " + String.valueOf(aFloat) + " Bps"),
        Observable.<String>empty().delay(1, TimeUnit.SECONDS),
        perform(device, this::unoptimizedSendCharacteristic).map(aFloat -> "Unoptimized Char: " + String.valueOf(aFloat) + " Bps"),
        Observable.<String>empty().delay(1, TimeUnit.SECONDS),
        perform(device, this::longWriteSend).map(aFloat -> "Long Write: " + String.valueOf(aFloat) + " Bps"),
        Observable.<String>empty().delay(1, TimeUnit.SECONDS),
        perform(device, this::optimizedSend).map(aFloat -> "Optimized: " + String.valueOf(aFloat) + " Bps")
)
        .observeOn(AndroidSchedulers.mainThread())
        .doOnUnsubscribe(this::clearSubscription)
        .subscribe(
                result -> Log.e("RESULT", result),
                e -> Log.e("ERROR", "Whoops!", e)
        );

private Observable<Float> perform(RxBleDevice device, TestSetup testSetup) {
    return device.establishConnection(false)
            .flatMap(RxBleConnection::discoverServices,
                    (connection, services) -> services.getCharacteristic(genericCommunicationCharacteristicUuid)
                            .flatMap(
                                    connection::setupNotification,
                                    (characteristic, responseObs) -> {
                                        characteristic.setWriteType(BluetoothGattCharacteristic.WRITE_TYPE_NO_RESPONSE);
                                        final Completable readyForTransmission =
                                                responseObs.filter(bytes -> bytes[0] == 0x01).first().toCompletable();
                                        final Observable<byte[]> speedTest = testSetup
                                                .create(new TestConnection(connection, responseObs, characteristic, 19));
                                        return readyForTransmission.andThen(speedTest);
                                    }
                            )
                            .flatMap(observable -> observable)
            )
            .flatMap(observable -> observable)
            .timestamp()
            .buffer(1, TimeUnit.MINUTES)
            .take(1)
            .map(timestampeds -> {
                final int bytesSent = (timestampeds.size() - 1) * 20 * 19;
                final float timeSeconds = (timestampeds.get(timestampeds.size() - 1).getTimestampMillis() - timestampeds.get(0).getTimestampMillis()) * 0.001f;
                return bytesSent / timeSeconds;
            });
}

interface TestSetup {

    Observable<byte[]> create(TestConnection testConnection);
}

static class TestConnection {

    final RxBleConnection connection;

[... 149 lines cut ...]
| Micromax Canvas A107 (5.0) | 2463.4 | 2544.2 | 3479.5 | 4604.4 |
| Nexus 5 (6.0.1) | 2555.1 | 2593.2 | 2657.8 | 2759.1 |
| Samsung Galaxy S6 SM-G920F (7.0) | 1631.9 | 1789.4 | 2290.3 | 3867.7 |
| Motorola Droid XT1030 (4.4.4) | 1737.2 | 1960.8 | 2707.6 | 3010.1 |
| Asus Zenfone 5 T00J (4.4.2) | 1589.2 | 1758.0 | 2450.2 | 4010.2 |

## Results for `Connection Interval = 100 ms`

| central \ algorithm | `unoptimizedSendUuid` | `unoptimizedSendCharacteristic` | `longWriteSend` | `optimizedSend` |
| --- | --- | --- | --- | --- |
| Micromax Canvas A107 (5.0) | 576.0 | 571.9 | 567.8 | 570.8 |
| Nexus 5 (6.0.1) | 605.0 | 612.0 | 623.9 | 624.5 |
| Samsung Galaxy S6 SM-G920F (7.0) | 629.0 | 619.9 | 623.2 | 632.3 |
| Motorola Droid XT1030 (4.4.4) | 342.3 | 342.9 | 343.0 | 344.8 |
| Asus Zenfone 5 T00J (4.4.2) | 546.7 | 542.9 | 560.7 | 545.7 |
| Samsung Galaxy S3 GT-I9300 (4.3) | 579.2 | 585.6 | 577.3 | 568.7 |

## Conclusion
* When interacting with a low `Connection Interval` peripheral it may be worth to use native implementation or custom operation to mitigate the downturn that `RxJava` causes.
* When the `Connection Interval` grows the benefit from optimisations are less visible. It could need a detailed research but it seems that for `Connection Interval > 50 ms` every implementation should perform similarly.
* The potential throughput of different Android handsets varies a lot (even in the same environment) because of the raw speed of the device (the route time of going through all the system layers from `BluetoothGatt.writeCharacteristic()` to `BluetoothGattCallback.onCharacteristicWrite()`) and number of buffers that it's Bluetooth Chip has.

I hope this quick research helps you.

Best Regards

closed 2017-10-02 by dariuszseweryn