dariuszseweryn/RxAndroidBle #42: Make RxBleConnectionState compile time constant, e.g. an enum. Otherwise I get "constant expression required".

farmazon3000, 2016-07-04:
RxBleConnectionState can't be used in the switch-case statement because they are not compile time constant, see http://stackoverflow.com/questions/3827393/java-switch-statement-constant-expression-required-but-it-is-constant. Maybe make RxBleConnectionState an enum?

For example this is illegal:
```
import static com.polidea.rxandroidble.RxBleConnection.RxBleConnectionState.CONNECTED;

switch(connectionState){
    case CONNECTED:
        break;
    default:
        break;
}
```
because Android Studio complaints `Constant expression required`.

dariuszseweryn, 2016-07-11:
Hello,
Basically using enums in Android development is discouraged because `#perfmatters` -> https://www.youtube.com/watch?v=Hzs6OBcvNQE
I was thinking if using `@IntDef` would solve the problem you have and it would kind of do but passing ints without any context is not very informative and cumbersome to get a human-readable meaningful value.

farmazon3000, 2016-07-11:
OK, enum was just an idea, you can also go Google style like https://developer.android.com/reference/android/bluetooth/BluetoothGatt.html#GATT_SUCCESS.

dariuszseweryn, 2017-05-16:
Finally done.

closed 2017-05-16 by dariuszseweryn

RobLewis, 2017-05-16:
I've used `switch( connectionState.toString() )`. Not the safest, but it works this week, anyway :-)