dariuszseweryn/RxAndroidBle #45: Discovery service fails after bluetooth turn off/on

klemzy, 2016-07-13:
### Summary
My current situation is that I have connected to device with autoconnect flag set. Which means I am waiting for the device to appear in order to connect. When I turn off/on bluetooth I establish again the connection. When the device becomes visible it connects but it hangs on discover services. I have done some debugging with [nrf Connect](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en) and discover services returns true. Also I debugged RxAndroidBle and `RxBleRadioOperationServicesDiscover` operation returns `true` from `bluetoothGatt.discoverServices()`. However `onServicesDiscovered` is never called inside `RxBleGattCallback`. If I clear the app and open again then it immediatelly connects. I am still looking what might be wrong but no solution so far. 
#### Preconditions
Connect with auto connect flag needs to be established.
#### Steps to reproduce actual result
1. Connect with auto connect flag
2. Turn airplane mode on
3. Turn airplane mode off
4. Connect with auto connect flag (make sure device is advertising)

#### Actual result
Discover services hangs
#### Expected result
Connection proceeds with `onServicesDiscovered` callback fired

Do you maybe have any ideas what I might be doing wrong?

Thanks!

dariuszseweryn, 2016-07-13:
Hello,
Thank you for your report. 
I will check tomorrow this scenario. I think that it may be a bug in the library because it is listening for `BluetoothAdapter`'s state changes and maybe when the connection is not yet established it's misbehaving.

klemzy, 2016-07-14:
I see. Also some more information. The weird thing is that in `RxBleGattCallback` `onConnectionStateChange` gets called but `onServicesDiscovered`. Also another thing. When I debugged with breakpoints on `bluetoothGatt.discoverServices()` and `onComplete` inside `RxBleRadioOperation` `getSubscriber then it worked.

dariuszseweryn, 2016-07-14:
I have tried to reproduce the issue today on Motorola XT1030 but apparently switching off the BluetoothAdapter while `establishConnection(_, true)` is making the connection never being established without any exceptions. I have waited for around 10 minutes.

Could you paste the code you are using and which mobile phone model?

Apparently it would be best to make the library fail any connections that are in the process of establishing if the BluetoothAdapter will go off.

klemzy, 2016-07-15:
I see, but did you try to connect again after turn off/on with device being discoverable? Because my problem is that when Bluetooth turns on I call `establishConnection`

I am using Nexus 6P, Android N Preview 4

`device.establishConnection(context, true)              
                .flatMap(rxBleConnection ->
                        rxBleConnection.getCharacteristic(CHARACTERISTIC)
                                .flatMap(characteristic -> rxBleConnection.writeCharacteristic(setValue(characteristic))
                                        .cache()
                                        .flatMap(characteristic1 -> readData(rxBleConnection))));`

This is the code.

dariuszseweryn, 2016-07-15:
Yes, I have tried that but I assume you're using the 1.0.2 version of the library whereas I am using the latest SNAPSHOT. Currently you cannot establish the connection twice with the same device if the previous connection did not finished. (I don't see if you keep the subscription and unsubscribe from it in any way)

For a temporary workaround I would suggest to monitor the BluetoothAdapter state changes and unsubscribe from the connection as soon as you see that it state changes.

It would be great to see the logs from the library (use `RxBleLog.setLogLevel(RxBleLog.VERBOSE)`)

klemzy, 2016-07-15:
I am calling unsubscribe but I did not include the whole snippet. I am listening to the adapter on/off and unsubscribing appropriately. Also I unsubscribe when I finish reading from the device.
[logs.txt](https://github.com/Polidea/RxAndroidBle/files/366581/logs.txt)

I am using latest code commited in this repository. 

When I was debugging (I included library source into my project) the code worked?!?!?, services got discovered.

I also found a way to simulate this by not turning on/off the adapter:
`.subscribe(
                        bluetoothGatt -> {
                            try {
                                Method method = bluetoothGatt.getClass().getMethod("refresh");
                                if (method != null) {
                                    method.setAccessible(true);
                                    boolean result = (boolean) method.invoke(bluetoothGatt);
                                    RxBleLog.d("Refresh result " + result);
                                }
                            } catch (NoSuchMethodException | InvocationTargetException | IllegalAccessException e) {
                                e.printStackTrace();
                            }
                            bluetoothGatt.close();
                        },
                        throwable -> onError(throwable),
                        () -> onCompleted()`

Basically inside `RxBleRadioOperationDisconnect` I have added refresh function. This clear discovered services cache from the device (which also happens on adapter off). At least what the Google and internet says.

So at the moment I managed to fix this with the folling code inside `RxBleRadioOperationServicesDiscover`.
`  Handler handler = new Handler(Looper.myLooper());
                    handler.postDelayed(() -> {
                        final boolean success = bluetoothGatt.discoverServices();
                        RxBleLog.d("Discover services " + success);
                        if (!success) {
                            subscription.unsubscribe();
                            subscriber.onError(new BleGattCannotStartException(BleGattOperationType.SERVICE_DISCOVERY));
                        }
                    }, 5000);`

Maybe there is a better solution to this problem.

I have attached logs.

dariuszseweryn, 2016-07-20:
What do you refer to when saying `I am using latest code committed in this repository.`? The latest SNAPSHOT or the latest release? Maybe your SNAPSHOT is not up to date. Unfortunately I couldn't reproduce your problem and refreshing the BluetoothGatt is not the best idea to reproduce the behaviour as I think it may not be the cause of the original problem.

dariuszseweryn, 2016-07-22:
Actually last SNAPSHOT was from May since there was a styling issue which prohibited pushing to the maven repo. There is a new SNAPSHOT available since yesterday now.

closed 2016-08-11 by dariuszseweryn