dariuszseweryn/RxAndroidBle #47: Read/write secured (with pin) characteristics

mars3142, 2016-07-20:
How is it possible to read/write secured characteristics with this library? I have a device, which is secured with a pin.

dariuszseweryn, 2016-07-20:
Hello,
Currently the library doesn't help with binding the devices and you have to bind it yourself.

mars3142, 2016-07-20:
I don't want to bind the device. Just connecting, reading and writing.

uKL, 2016-07-21:
How does your device get the pin then? Standard Bluetooth implementation requires to pair the device in a PIN mode (binding process), which is currently unsupported. You can always bind the device yourself using BluetoothDevice object and continue with the library then. Although pull requests are kindly welcome!

mars3142, 2016-07-21:
I added a notification to a characteristic and in the StatusBar a "binding notification" pops up. This is really bad, because I expected a PIN dialog popping up.

I'll check your advice with manual binding, but this way I had to know which characteristics are protected or is this a one time thing? How about PIN changes? Will I need to bind again?

dariuszseweryn, 2016-07-27:
I am not an expert in binding the device just yet. The behaviour for binding may be different from vendor to vendor.
When binding devices - with the "Just Works" solution it's a one time thing. Other types you need to check out by yourself. If you will do that - please share the info here. The library may benefit from it.

Best Regards

ipodib, 2017-05-22:
Any updates on this item? It would be nice have binding devices feature built into library.

dariuszseweryn, 2017-05-22:
There is a chance that soon the connection will not be closed automatically when there is an encryption error. It will enable users to workaround this issue.

ipodib, 2017-05-22:
And what about functionality, that will automatically propose to pair device using pin number if it's required by device during connection (when invoking `establishConnection`)? For example how it's done from the Android's settings app. It's handled automatically there.

dariuszseweryn, 2017-05-22:
I do not have much experience with devices that need bonding so I cannot really tell. I have read that usually Android shows a dialog / notification by itself.

If you will have a moment for investigation - feel free to look into it and report your findings or submit a PR :)

ipodib, 2017-05-22:
If I have a chance I will :) Since it demand feature for me right now. 
Thanks for update.

ipodib, 2017-05-28:
I figured out how we can bond device by prompting entering ping. Since `RxBleDevice` has a reference to `BluetoothDevice` we can simply call `createBond()` and Android will do everything else for us. 

Reference:
https://developer.android.com/reference/android/bluetooth/BluetoothDevice.html#createBond()

dariuszseweryn, 2017-05-29:
Unfortunately when testing I could not get stable results with `.createBond()`. Once bonded — when canceled the bond — I could not make the device to bond again. Is that an issue for you?

ipodib, 2017-05-30:
I can't reproduce this issue. Everything works good for me. Small note, that I've checking for already bounded devices, and don't bond it when it's already bonded.

JohnnyJem, 2017-05-30:
According to Apple guidelines(which most devices should aim to fulfill):
"as explained in Apple’s Bluetooth Accessory Design Guidelines. Instead, you should make sure to set the permissions as you want on your characteristics (see the usage of BLE_GAP_CONN_SEC_MODE_SET_ENC_NO_MITM() above), so that an error will be returned to the Central when it tries to do something that requires encryption. This error will then trigger most Central devices to automatically start bonding."

Basically if you attempt to read a characteristic that requires "authorization" the device should return a "ERROR_INSUFFICIENT_AUTHEN (0x41)" error code that will prompt the android system to automatically begin bonding the device. As for the pin requirement, i believe that you should set up your device to require MITM when pairing/bonding. This should cause a "enter passkey" prompt to appear when the bonding procedure is kicked off by the Android system in response to receiving an  "ERROR_INSUFFICIENT_AUTHEN (0x41)"error code.

See 
https://stackoverflow.com/questions/29204634/ios-bluetooth-ble-security-and-just-works-association-model
for a link to a table which explains the types of device configurations needed in order to have a pairing response needing "passkey entry" .

dariuszseweryn, 2017-10-02:
As it is now possible to workaround this by either using `BluetoothDevice RxBleDevice.getBluetoothDevice()` and calling `BluetoothDevice.createBond()` or by retrying `RxBleConnection.readCharacteristic()`/`RxBleConnection.writeCharacteristic()`/`RxBleConnection.setupNotification()`—closing.

There is already [an issue](#35) for adding helpers.

closed 2017-10-02 by dariuszseweryn