dariuszseweryn/RxAndroidBle #48: Indicate not working

fracturedpsyche, 2016-07-20:
I am attempting to establish a connection to a device that is already bonded to my tablet. 

If I just attempt to connect the bonded device I get an error with status 133 after 30 seconds. 

If the device is in pairing mode and I can connect fine, but once the device stops advertising the connection is terminated and any further indications are lost.

If I set the device up to be attempting to broadcast data and then connect, the system errors with a status 19 right after discover Services, and the indications are never sent.

Thoughts on this? This does not happen with my own home grown BLE framework. I am evaluating yours as a replacement to mine but cannot get this blood pressure cuff to work.

Thanks!

dariuszseweryn, 2016-07-22:
Hello,

status 133 is a `GATT_ERROR`
status 19 if received from `onConnectionStateChanged` is `GATT_CONN_TERMINATE_PEER_USER`
That is according to: https://android.googlesource.com/platform/external/bluetooth/bluedroid/+/android-5.1.0_r1/stack/include/gatt_api.h

Since I cannot recreate your problem it is quite hard to judge - it seems that the device you're connecting to drops the connection. Maybe it is possible to compare your framework and the RxAndroidBle to check out what is different in the framework and how you do use it?

Best Regards

dariuszseweryn, 2016-08-25:
Hello, any news?

fracturedpsyche, 2016-08-25:
No news. I have moved on from what I was doing with Android on to other things. Should I get back to evaluating RxAndroidBle I will dig more into it and provide an update. Thanks!

closed 2016-08-25 by fracturedpsyche

tymoxx, 2021-07-14:
@fracturedpsyche Did you eventually find out what was causing the disconnection with the error status 19?

Yazazzello, 2024-04-03:
For future references, I share my experience, because I didn't find a better thread related to this error.
GATT_CONN_TERMINATE_PEER_USER(19) also happens with bonded device if ... **device _forgot_ bonding**, but Android OS doesn't know that.

> Happened to me, costed few good hours to troubleshoot.
> Long story short: I'm working on an app which communicates with custom hardware via BLE. I received new firmware and uploaded it to previously bonded device. App connects for a second and then immediately receives  GATT_CONN_TERMINATE_PEER_USER(19). Rolling back to previous FW didn't help either. Searching in issue trackers wasn't fruitful. Somehow, I started to suspect that firmware manipulation was the root cause. Obviously, it's a bug - FW update shouldn't clear bonding status. 
> 

Solution: You have to '**unpair**' device in bluetooth settings.