dariuszseweryn/RxAndroidBle #49: setupNotification freezes

farmazon3000, 2016-08-04:
I try to set up notifications just like you propose:
```
RxBleDevice rxBleDevice = rxBleClient.getBleDevice(address);
rxBleDevice.establishConnection(this, false)
        .flatMap(rxBleConnection -> rxBleConnection.setupNotification(BATTERY_LEVEL_UUID))
        .doOnNext(notificationObservable -> {
            // Notification has been set up
        })
        .flatMap(notificationObservable -> notificationObservable) // <-- Notification has been set up, now observe value changes.
        .subscribe(bytes -> {
            // Given characteristic has been changes, here is the value.
            Log.d(TAG, "bytes" + bytes.toString());
        });
```

And it does not work - I see in logcat:
```
D/RxBle#Radio:   QUEUED RxBleRadioOperationConnect(19575723)
D/RxBle#Radio:  STARTED RxBleRadioOperationConnect(19575723)
D/RxBle#BluetoothGatt: onConnectionStateChange newState=2 status=0
D/RxBle#Radio:   QUEUED RxBleRadioOperationServicesDiscover(175585553)
D/RxBle#Radio: FINISHED RxBleRadioOperationConnect(19575723)
D/RxBle#Radio:  STARTED RxBleRadioOperationServicesDiscover(175585553)
```
`STARTED` is the last log, then nothing is logged.

farmazon3000, 2016-08-04:
Even when I try to disconnect later it's queued but not proceeded (i.e. when I unsubscribe from connection subscription I see in logs `RxBle#Radio:   QUEUED RxBleRadioOperationDisconnect(237327482)` and then silence - it's not proceeded).

farmazon3000, 2016-08-05:
It (of course :)) seems to be some kind of problem with BT on Android itself as it does not occur always. And sometimes it may be fixed but switching off/on BT, and sometimes it may be fixed but switching off/on whole device.

dariuszseweryn, 2016-08-09:
Could you share more info about this issue? Do you connect to a device that is changing it's services or something?

farmazon3000, 2016-08-09:
`Do you connect to a device that is changing it's services or something?` No, the device I am connecting to definitly does not change its services. It seems that I was so long messing around with BLE on my Moto G that it was the time when nothing works and you have to restart BT or restart Moto G. After 2 years working with BLE in Android I can tell that it happens every once in a while (especially when you make some stress tests and connect/disconnect to multiple devices very frequently). But it would be nice to have at least some way to try to solve this issue.

aldoborrero, 2016-08-16:
Or at least to warn the programmer like SweetBlue does with its [UhOhListener](https://github.com/iDevicesInc/SweetBlue/wiki/Android-BLE-Issues):

> The native stack's bluetooth service can sometimes get into states where it can't be turned on, or even can't be turned off. SweetBlue cannot fix this, but is does detect and report this issue through UhOhListener. Usually this means a phone restart is required.

Just an idea (that unfortunately can't be translated directly).

dariuszseweryn, 2016-11-15:
Long time no see. 
The hanging discovery issue should be fixed (+ a workaround) #86 

UhOh is a totally different topic.

Best Regards

closed 2016-11-15 by dariuszseweryn