dariuszseweryn/RxAndroidBle #51: On some device, gatt.close() may cause problem: can't find specified device later when scanning

nziyouren, 2016-08-07:
### Summary
<br /> On some devices, such as Huawei hornor. After user directly switched to glucose device power off which was connected and written data before, The device (Huawei) can't find previous glucose device again when scanning. But I found that ,if I remove the gatt.close() method after disconnected, the Huawei device can find glucose. So Can you help me how to process this special situation in an elegant way when using your library. Temporarily , I just remove gatt.close()  in RxBleRadioOperationDisconnect class. But I know, it's not the best choice.
#### Preconditions
<br />Successfully connected to glucose device and written data
#### Steps to reproduce actual result
1. Directly power off glucose device
2. Power on glucose device again.
#### Actual result
<br />Can't find previous glucose device when scanning, except restart the whole BT(turn off BT and then turn on in Android device settings).
#### Expected result
<br />Normal scan and find previous glucose device.

@uKL

dariuszseweryn, 2016-08-08:
Hello @nziyouren !
Thanks for the info. Could you show me the code you're using and the device log when the issue happens?
Best Regards

nziyouren, 2016-08-14:
Hi @dariuszseweryn , sorry for the delay. I just use your sample code, nothing special. Device logs are below. I just remove some duplicate scan result logs

08-14 21:23:25.810 6843-6843/? D/dalvikvm: Late-enabling CheckJNI
08-14 21:23:25.870 6843-6843/com.polidea.rxandroidble.sample I/dalvikvm: Could not find method android.bluetooth.BluetoothDevice.connectGatt, referenced from method com.polidea.rxandroidble.internal.util.BleConnectionCompat.connectGattCompat
08-14 21:23:25.870 6843-6843/com.polidea.rxandroidble.sample W/dalvikvm: VFY: unable to resolve virtual method 263: Landroid/bluetooth/BluetoothDevice;.connectGatt (Landroid/content/Context;ZLandroid/bluetooth/BluetoothGattCallback;I)Landroid/bluetooth/BluetoothGatt;
08-14 21:23:25.870 6843-6843/com.polidea.rxandroidble.sample D/dalvikvm: VFY: replacing opcode 0x6e at 0x0011
08-14 21:23:25.910 6843-6843/com.polidea.rxandroidble.sample W/dalvikvm: VFY: unable to find class referenced in signature (Landroid/view/SearchEvent;)
08-14 21:23:25.910 6843-6843/com.polidea.rxandroidble.sample I/dalvikvm: Could not find method android.view.Window$Callback.onSearchRequested, referenced from method android.support.v7.view.WindowCallbackWrapper.onSearchRequested
08-14 21:23:25.910 6843-6843/com.polidea.rxandroidble.sample W/dalvikvm: VFY: unable to resolve interface method 18925: Landroid/view/Window$Callback;.onSearchRequested (Landroid/view/SearchEvent;)Z
08-14 21:23:25.910 6843-6843/com.polidea.rxandroidble.sample D/dalvikvm: VFY: replacing opcode 0x72 at 0x0002
08-14 21:23:25.910 6843-6843/com.polidea.rxandroidble.sample I/dalvikvm: Could not find method android.view.Window$Callback.onWindowStartingActionMode, referenced from method android.support.v7.view.WindowCallbackWrapper.onWindowStartingActionMode
08-14 21:23:25.910 6843-6843/com.polidea.rxandroidble.sample W/dalvikvm: VFY: unable to resolve interface method 18929: Landroid/view/Window$Callback;.onWindowStartingActionMode (Landroid/view/ActionMode$Callback;I)Landroid/view/ActionMode;
08-14 21:23:25.910 6843-6843/com.polidea.rxandroidble.sample D/dalvikvm: VFY: replacing opcode 0x72 at 0x0002
08-14 21:23:25.950 6843-6843/com.polidea.rxandroidble.sample I/dalvikvm: Could not find method android.content.res.TypedArray.getChangingConfigurations, referenced from method android.support.v7.widget.TintTypedArray.getChangingConfigurations
08-14 21:23:25.950 6843-6843/com.polidea.rxandroidble.sample W/dalvikvm: VFY: unable to resolve virtual method 497: Landroid/content/res/TypedArray;.getChangingConfigurations ()I
08-14 21:23:25.950 6843-6843/com.polidea.rxandroidble.sample D/dalvikvm: VFY: replacing opcode 0x6e at 0x0002
08-14 21:23:25.950 6843-6843/com.polidea.rxandroidble.sample I/dalvikvm: Could not find method android.content.res.TypedArray.getType, referenced from method android.support.v7.widget.TintTypedArray.getType
08-14 21:23:25.950 6843-6843/com.polidea.rxandroidble.sample W/dalvikvm: VFY: unable to resolve virtual method 519: Landroid/content/res/TypedArray;.getType (I)I
08-14 21:23:25.950 6843-6843/com.polidea.rxandroidble.sample D/dalvikvm: VFY: replacing opcode 0x6e at 0x0002
08-14 21:23:26.050 6843-6843/com.polidea.rxandroidble.sample D/OpenGLRenderer: Enabling debug mode 0
08-14 21:23:41.590 6843-6843/com.polidea.rxandroidble.sample E/BleCenterManager: >>>>>>actual do scan>>>>>>
08-14 21:23:41.610 6843-6843/com.polidea.rxandroidble.sample D/RxBle#Radio:   QUEUED RxBleRadioOperationScan(1101803368)
08-14 21:23:41.610 6843-6884/com.polidea.rxandroidble.sample D/RxBle#Radio:  STARTED RxBleRadioOperationScan(1101803368)
08-14 21:23:41.610 6843-6843/com.polidea.rxandroidble.sample D/BluetoothAdapter: startLeScan(): null
08-14 21:23:41.640 6843-6858/com.polidea.rxandroidble.sample D/BluetoothAdapter: onClientRegistered() - status=0 clientIf=4
08-14 21:23:41.640 6843-6884/com.polidea.rxandroidble.sample D/RxBle#Radio: FINISHED RxBleRadioOperationScan(1101803368)
08-14 21:24:02.320 6843-6857/com.polidea.rxandroidble.sample D/BluetoothAdapter: onScanResult() - Device=DE:5F:D1:B2:CB:5C RSSI=-70
08-14 21:24:02.350 6843-6892/com.polidea.rxandroidble.sample D/BluetoothAdapter: onScanResult() - Device=DE:5F:D1:B2:CB:5C RSSI=-70
08-14 21:24:02.380 6843-6858/com.polidea.rxandroidble.sample D/BluetoothAdapter: onScanResult() - Device=DE:5F:D1:B2:CB:5C RSSI=-71
08-14 21:24:02.420 6843-6857/com.polidea.rxandroidble.sample D/BluetoothAdapter: onScanResult() - Device=DE:5F:D1:B2:CB:5C RSSI=-70
08-14 21:24:02.450 6843-6892/com.polidea.rxandroidble.sample D/BluetoothAdapter: onScanResult() - Device=DE:5F:D1:B2:CB:5C RSSI=-71
08-14 21:24:02.480 6843-6858/com.polidea.rxandroidble.sample D/BluetoothAdapter: onScanResult() - Device=DE:5F:D1:B2:CB:5C RSSI=-75
08-14 21:24:02.510 6843-6857/com.polidea.rxandroidble.sample D/BluetoothAdapter: onScanResult() - Device=DE:5F:D1:B2:CB:5C RSSI=-75
08-14 21:24:02.530 6843-6892/com.polidea.rxandroidble.sample D/BluetoothAdapter: onScanResult() - Device=DE:5F:D1:B2:CB:5C RSSI=-70
08-14 21:24:02.570 6843-6858/com.polidea.rxandroidble.sample D/BluetoothAdapter: onScanResult() - Device=DE:5F:D1:B2:CB:5C RSSI=-70
08-14 21:24:02.600 6843-6857/com.polidea.rxandroidble.sample D/BluetoothAdapter: onScanResult() - Device=DE:5F:D1:B2:CB:5C RSSI=-69
08-14 21:24:02.630 6843-6892/com.polidea.rxandroidble.sample D/BluetoothAdapter: onScanResult() - Device=DE:5F:D1:B2:CB:5C RSSI=-70
[... cut ...]
08-14 21:25:37.940 6843-6843/com.polidea.rxandroidble.sample D/RxBle#Radio:   QUEUED RxBleRadioOperationScan(1102461472)
08-14 21:25:37.950 6843-6884/com.polidea.rxandroidble.sample D/RxBle#Radio:  STARTED RxBleRadioOperationScan(1102461472)
08-14 21:25:37.950 6843-6843/com.polidea.rxandroidble.sample D/BluetoothAdapter: startLeScan(): null
08-14 21:25:37.950 6843-6892/com.polidea.rxandroidble.sample D/BluetoothAdapter: onClientRegistered() - status=0 clientIf=4
08-14 21:25:37.960 6843-6884/com.polidea.rxandroidble.sample D/RxBle#Radio: FINISHED RxBleRadioOperationScan(1102461472)

dariuszseweryn, 2016-08-16:
It seems that the connection is somehow closed externally as first it is disconnected and after that the application is queueing disconnect operation. Maybe the glucose meter is turning off after some time of inactivity and that is why it doesn't advertise?

nziyouren, 2016-08-16:
@dariuszseweryn no, after turn on glucose meter again, it advertises. But my device can not find it. While another phone can find glucose meter in scan result. Maybe it is a bug of my phone, but I want to solve it in an elegant way while using your library.

kexuejin, 2016-08-16:
I have same problem,i found it's have problem when turn off the device,then try to connect it,when we start scan ble, it's will scan noting.If exit app,the gatt not close immediately.When gatt closed,the scan will worked.

dariuszseweryn, 2017-02-02:
The issue with `BluetoothGatt` not being closed promptly was fixed some time ago. About the device not being scannable after closing the connection - I have no idea what could be done. Closing for now.

closed 2017-02-02 by dariuszseweryn