dariuszseweryn/RxAndroidBle #54: Writing to a characteristic and getting a response on a Notification Characteristic

fjanicki, 2016-08-11:
### Summary
First of all, thank you for this awesome library. I am currently working on a BLE project where to do read requests I need to write certain bytes on a write-only characteristic, the hardware module processes the request then answers back on a notification-only characteristic. How would you suggest that I proceed to Write -> Get response in these conditions using RxAndroidBle? I want to avoid writing multiple times then getting de-synced responses since I am working with 2 different characteristics.

I need to wait for response because my next write will usually depend on the type of response I get back.

dariuszseweryn, 2016-08-12:
The answer will be posted here: http://stackoverflow.com/questions/38902913/rxandroidble-keeping-a-persistant-connection-write-notification-handling

Best Regards

closed 2016-08-12 by dariuszseweryn

Maple-Ro, 2018-06-22:
is the answer use the rxjava not rxjava2? I am confused in using the answer code(many method can not resolve) in rxjava2 & Rxandroidble2 project, thank you!

dariuszseweryn, 2018-06-22:
Hello 

As you can see the answer was added almost two years ago — at the time the library supported `RxJava 1` only. You are now trying to use the same code for `RxJava 2` version of the library — this will not work without some minor changes.

I will try to find some time to update the answer in the future but I cannot commit to any particular time frame.

You can adjust the code by yourself as the general API of the library has not changed — it was just updated to `RxJava 2`. You should be able to find some [tutorials on the topic](https://www.google.pl/search?q=rxjava+1+to+rxjava2).

Best Regards

Maple-Ro, 2018-06-23:
Thank you very much!
with your guides, I have get the answer, more detail see this[https://stackoverflow.com/questions/49921894/write-notification-handling-with-rxandroidble?noredirect=1&lq=1]