dariuszseweryn/RxAndroidBle #56: BluetoothGatt can't close when device is not connect.

kexuejin, 2016-08-16:
### Summary
<br />Normal if device is connected,when exit activity or fragment,the bluetooth will close,but if the device is try to connect(the device is offline),the bluetooth not close right now,after maybe one or two minute later,the bluetooth close.
#### Preconditions
<br />
#### Steps to reproduce actual result
1.
2.
3.
#### Actual result
<br />
#### Expected result
<br />

kexuejin, 2016-08-17:
By the way,if device is try to connect, 
at the same time  to scan device,the scan will return nothing until device gatt close,please help me.

dariuszseweryn, 2016-08-17:
Hello @kexuejin 
I am currently working on the fix to close the connection asap. For the second issue it is by design that only one operation is being performed at the time and establishing the connection is an atomic operation if `autoConnect == false`. Once I will finish the fix scan should happen earlier.
Best Regards

kexuejin, 2016-08-17:
Thank you very much!

dariuszseweryn, 2016-08-18:
Fix is ready. It's waiting for our internal code review but unfortunately it is our holiday season and it may take up to two weeks. If you need it quicker you can always checkout the latest `master` branch and exchange `RxBleRadioOperationConnect.java` file code to:
```
package com.polidea.rxandroidble.internal.operations;

import android.bluetooth.BluetoothDevice;
import android.bluetooth.BluetoothGatt;
import android.support.annotation.NonNull;

import com.polidea.rxandroidble.RxBleConnection;
import com.polidea.rxandroidble.internal.RxBleLog;
import com.polidea.rxandroidble.internal.RxBleRadioOperation;
import com.polidea.rxandroidble.internal.connection.RxBleGattCallback;
import com.polidea.rxandroidble.internal.util.BleConnectionCompat;

import rx.Observable;
import rx.subjects.BehaviorSubject;

import static com.polidea.rxandroidble.RxBleConnection.RxBleConnectionState.CONNECTED;

public class RxBleRadioOperationConnect extends RxBleRadioOperation<BluetoothGatt> {

    private final BluetoothDevice bluetoothDevice;
    private final RxBleGattCallback rxBleGattCallback;
    private final BleConnectionCompat connectionCompat;
    private final boolean autoConnect;
    private BehaviorSubject<BluetoothGatt> bluetoothGattBehaviorSubject = BehaviorSubject.create();
    @SuppressWarnings("Convert2MethodRef")
    private final Runnable releaseRadioRunnable = () -> releaseRadio();
    private final Runnable emptyRunnable = () -> {
    };
    private final BehaviorSubject<Boolean> isSubscribed = BehaviorSubject.create(false);
    private final Observable<BluetoothGatt> asObservable = super.asObservable()
            .doOnSubscribe(() -> isSubscribed.onNext(true))
            .doOnUnsubscribe(() -> isSubscribed.onNext(false))
            .share();

    public RxBleRadioOperationConnect(BluetoothDevice bluetoothDevice, RxBleGattCallback rxBleGattCallback,
                                      BleConnectionCompat connectionCompat, boolean autoConnect) {
        this.bluetoothDevice = bluetoothDevice;
        this.rxBleGattCallback = rxBleGattCallback;
        this.connectionCompat = connectionCompat;
        this.autoConnect = autoConnect;
    }

    @Override
    public Observable<BluetoothGatt> asObservable() {
        return asObservable;
    }

    @Override
    protected void protectedRun() {
        final Runnable onConnectionEstablishedRunnable = autoConnect ? emptyRunnable : releaseRadioRunnable;
        final Runnable onConnectCalledRunnable = autoConnect ? releaseRadioRunnable : emptyRunnable;

        getConnectedBluetoothGatt()
                // when there are no subscribers there is no point of continuing work -> next will be disconnect operation
                .takeUntil(hasNoSubscribers().doOnNext(noSubscribers -> RxBleLog.d("No subscribers, finishing operation")))
                .doOnCompleted(onConnectionEstablishedRunnable::run)
                .subscribe(getSubscriber());
        onConnectCalledRunnable.run();
    }

    @NonNull
    private Observable<Boolean> hasNoSubscribers() {
        return isSubscribed.filter(aBoolean -> !aBoolean);
    }

    /**
     * Emits BluetoothGatt and completes after connection is established.
     *
     * @return BluetoothGatt after connection reaches {@link RxBleConnection.RxBleConnectionState#CONNECTED} state.
     * @throws com.polidea.rxandroidble.exceptions.BleDisconnectedException if connection was disconnected/failed before it was established.
     */
    @NonNull
    private Observable<BluetoothGatt> getConnectedBluetoothGatt() {
        // start connecting the BluetoothGatt
        // note: Due to different Android BLE stack implementations it is not certain whether `connectGatt()` or `BluetoothGattCallback`
        // will emit BluetoothGatt first
        return Observable.fromCallable(() ->
                connectionCompat.connectGatt(bluetoothDevice, autoConnect, rxBleGattCallback.getBluetoothGattCallback())
        )
                .mergeWith(rxBleGattCallback.getBluetoothGatt())
                // relay BluetoothGatt instance updates
                .doOnNext(bluetoothGattBehaviorSubject::onNext)
                // capture BluetoothGatt when connected
                .sample(rxBleGattCallback
                        .getOnConnectionStateChange()
                        .filter(rxBleConnectionState -> rxBleConnectionState == CONNECTED))
                // disconnect may happen even if the connection was not established yet
                .mergeWith(rxBleGattCallback.observeDisconnect())
                .take(1)
                // finish relaying if there won't be more updates
                .doOnTerminate(bluetoothGattBehaviorSubject::onCompleted);
    }

    /**
     * Obtain observable emitting most recent {@link BluetoothGatt instance}.
     * NOTE: Connection may be released and/or GATT may be closed in any point of time.
     *
     * @return Observable with BluetoothGatt. Most recent GATT will be emitted instantly after subscription if it is available.
     */
    public Observable<BluetoothGatt> getBluetoothGatt() {
        return bluetoothGattBehaviorSubject;
    }
}
```

Best Regards

s0nerik, 2016-09-05:
Hey guys, I suddenly found that reverting the RxJava version to 1.1.7 (previously was 1.1.9) seems to help with this problem. I'm gonna assume there were some changes in the way Subjects work in the later versions that made the library behave strangely.

kexuejin, 2016-09-06:
Maybe, but i don't want to reverting to rxjava version...

dariuszseweryn, 2016-09-08:
Fixed with https://github.com/Polidea/RxAndroidBle/commit/604853c4f39c5e8a19e02415c50b547b0befd0e7 
Should be available in the SNAPSHOT shortly.

closed 2016-09-08 by dariuszseweryn