dariuszseweryn/RxAndroidBle #59: Usage of BleConnectionCompat.connectUsingReflection() causes connection being never established on Sony Experia Z (C6603)

lkjh654, 2016-08-20:
### Summary
A connection cannot be established using `BleConnectionCompat.connectUsingReflection()` (`autoConnect=true` )  on Sony Experia Z (C6603) Android 5.1.1. Subscribe action is never called.
When I modified the library source code so that `connectGattCompat(...)` is always used the connection is established successfully.
<br />
#### Preconditions
`autoConnect=true`
 Sony Experia Z (C6603) Android 5.1.1
<br />
#### Steps to reproduce actual result
```
RxBleDevice device = rxBleClient.getBleDevice(macAddress);
Subscription subscription = device.establishConnection(context, true) // <-- autoConnect flag
                .subscribe(rxBleConnection -> {
                    Log.d(TAG, "Connection established");
                });
```
#### Actual result
`Log.d(TAG, "Connection established");` is never called

Logs:
```
08-20 14:59:18.333 27298-27298/bletest D/RxBle#Radio:   QUEUED RxBleRadioOperationConnect(185299140)
08-20 14:59:18.334 27298-27888/bletest D/RxBle#Radio:  STARTED RxBleRadioOperationConnect(185299140)
08-20 14:59:18.347 27298-27298/bletest V/RxBle#RadioOperationConnect: Trying to connectGatt using reflection.
08-20 14:59:18.348 27298-27298/bletest V/RxBle#BleConnectionCompat: Found constructor with args count = 4
08-20 14:59:18.349 27298-27298/bletest V/RxBle#BleConnectionCompat: Connecting using reflection
08-20 14:59:18.349 27298-27298/bletest D/BluetoothGatt: connect() - device: 6F:14:AB:E7:6D:01, auto: true
08-20 14:59:18.349 27298-27298/bletest D/BluetoothGatt: registerApp()
08-20 14:59:18.349 27298-27298/bletest D/BluetoothGatt: registerApp() - UUID=ac648d82-77cf-4068-9fcf-3753edb45e28
08-20 14:59:18.350 27298-27315/bletest D/BluetoothGatt: onClientRegistered() - status=0 clientIf=6
08-20 14:59:18.351 27298-27888/bletest D/RxBle#Radio: FINISHED RxBleRadioOperationConnect(185299140)
```
<br />
#### Expected result
`Log.d(TAG, "Connection established");` should be called. It is called when `autoConnect=false` or the library source code is modified so only `connectGattCompat(...)` is used. 
<br />

Logs:
```
08-20 14:57:46.612 23982-23982/bletest D/RxBle#Radio:   QUEUED RxBleRadioOperationConnect(805970639)
08-20 14:57:46.613 23982-24778/bletest D/RxBle#Radio:  STARTED RxBleRadioOperationConnect(805970639)
08-20 14:57:46.624 23982-23982/bletest V/RxBle#BleConnectionCompat: Connecting without reflection
08-20 14:57:46.625 23982-23982/bletest D/BluetoothGatt: connect() - device: 6F:14:AB:E7:6D:01, auto: true
08-20 14:57:46.625 23982-23982/bletest D/BluetoothGatt: registerApp()
08-20 14:57:46.625 23982-23982/bletest D/BluetoothGatt: registerApp() - UUID=e14d9e6e-dd10-4487-be70-5a51f0629b4d
08-20 14:57:46.628 23982-24020/bletest D/BluetoothGatt: onClientRegistered() - status=0 clientIf=5
08-20 14:57:46.630 23982-24778/bletest D/RxBle#Radio: FINISHED RxBleRadioOperationConnect(805970639)
08-20 14:57:46.846 23982-24020/bletest D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=5 device=6F:14:AB:E7:6D:01
08-20 14:57:46.846 23982-24020/bletest D/RxBle#BluetoothGatt: onConnectionStateChange newState=2 status=0
08-20 14:57:46.850 23982-24789/bletest D/MainActivity: Connection established
```

#### Moreover
Could you please elaborate more on the used reflection workaround? Could you provide a list of affected firmwares / Android versions. Maybe the workaround should be disabled by default and used on some devices instead of enabling it by default for all the devices?

dariuszseweryn, 2016-08-22:
Hello,
When connecting with `autoConnect` flag the `RxBleRadioOperationConnect` finishes after registering the client. The reason is that with `autoConnect` flag set to `true` the connection usually is established quite a lot later even if the device you're trying to connect is near by and advertising. I have seen times even above 10 minutes for connecting where connection with `autoConnect = false` took 5 seconds.
How long have you waited for the connection to be established?
From the logs I see that calling the `connectGatt` using reflection finished successfully - there are no errors with that.
Best Regards

lkjh654, 2016-08-27:
Hi,
thanks for the prompt response. I've been trying to connect for dozens of minutes without success.

The thing is that `autoConnect` flag works perfectly fine on the same device when:
- using android-BluetoothLeGatt sample from Google
- reflection based method in the library is replaced with the regular API

I know this issue is weird especially because:
- it happens only on one device among many I have used
- the reflection based implementation seems to be identical to the inner Android implementation

Anyways I've changed my app implementation so that `autoConnect=false` but still it would be cool if you could address the last part of my comment regarding why the reflection based workaround is used in the first place and which Android versions / devices are affected.

thanks.

wickedpygmy, 2016-10-13:
Reflection workaround is used to fix this bug: 
https://code.google.com/p/android/issues/detail?id=69834
Which affects the majority of Android devices, as it has only been fixed very recently.

dariuszseweryn, 2017-10-16:
Since I do not own nor I do have access to the said model I do not think this issue will get tested/fixed. Closing.

closed 2017-10-16 by dariuszseweryn