dariuszseweryn/RxAndroidBle #61: Is possible to subscribe to multiple characteristic notifications at once?

aldoborrero, 2016-08-24:
Is there any possibility where I can subscribe to those characteristics that allow notifications in one row?

In the demos you only listen to one characteristic at a time but would be great to have in `RxBleDeviceServices`:

```Java
public Observable<List<BluetoothGattCharacteristic>> getCharacteristics(@NonNull UUID serviceUuid) {
    return getService(serviceUuid).map(BluetoothGattService::getCharacteristics)
        .filter(bluetoothGattCharacteristics -> (bluetoothGattCharacteristics != null && bluetoothGattCharacteristics.size() > 0))
        .switchIfEmpty(Observable.error(new BleCharacteristicNotFoundException(serviceUuid)));
  }
```

As I'm quite new to RxJava (and your codebase) I don't know if is a good idea to do this (or even the approach) but I would gladly code it if I'm given more explanations on how to achieve it (or maybe I'm thinking a bad assumption and is already implemented).

The use case is that I have one object that acts as a model with different parameters and those parameters are filled with the notifications that each characteristic sends with the updated value. As I have 10 characteristics sending notifications, I have to setup every notification by hand, then listen to the change and then update the object.

Below is an example of what to achieve (don't look too much to the code as I know that is faulty)

```Java
connectionObservable.flatMap(RxBleConnection::discoverServices)
        .flatMap(rxBleDeviceServices -> rxBleDeviceServices.getCharacteristics(ALDeviceUuids.SERVICE_ID))
        .flatMapIterable(bluetoothGattCharacteristics -> bluetoothGattCharacteristics)
        .filter(characteristic -> BleUtils.hasNotificationProperty(characteristic.getProperties()))
        .doOnNext(rxBleConnection::setupNotification)
        .observeOn(AndroidSchedulers.mainThread())
        //.flatMap(characteristic -> characteristic)
        .subscribe(this::onNotificationWritten);
```

The same could apply if I would like to read all the values at once by filtering those characteristics that allow reading. 

Thanks :)

dariuszseweryn, 2016-08-24:
Hello there.

Question: Is possible to subscribe to multiple characteristic notifications at once?
Answer: It is possible to subscribe to notifications from multiple characteristics but not in one line.
Explanation: There are almost infinite use cases for the BLE applications and it is impossible to cover them all. I feel that with the current state of the code it is quite easy to implement such a feature but baking it into the library won't benefit many people.

From what I understand you would like to map characteristics values into a POJO model in this scenario either to manually assign a characteristic value to a property or have a model that is very generic (a `Map<UUID, byte[]> perhaps). In the first situation you would need to either set notifications by hand or match the resulting `byte[]` by hand. In the second situation you will end up with something like:

        final UUID serviceUuid = // your service UUID
        final Map<UUID, byte[]> genericModel = new HashMap<>();
        final Observable<RxBleConnection> connectionObservable = // your connectionObservable
        
        connectionObservable
                .flatMap(connection -> 
                        connection.discoverServices()
                                .flatMap(services -> services.getService(serviceUuid).map(BluetoothGattService::getCharacteristics)) // get characteristics you're interested in
                                .flatMap(Observable::from) // deal with every characteristic separately
                                .flatMap(characteristic -> connection
                                                .setupNotification(characteristic) // setup notification for each
                                                .flatMap(observable -> observable), // to get the raw bytes from notification
                                        Pair::new) // merge characteristic with byte[] to keep track from which characteristic the bytes came
                )
                .subscribe(
                        pair -> genericModel.put(pair.first.getUuid(), pair.second),
                        throwable -> { /* handle errors */}
                ); 

I hope this answers your question.

Best Regards

aldoborrero, 2016-08-24:
Thanks! It helps! :)

My main concern is that as I'm learning RxJava I missed that.

dariuszseweryn, 2016-08-24:
Just have in mind that Android allows to setup only a finite number of notifications.

aldoborrero, 2016-08-24:
Yeah I believe that are 4 for version below 4.4 and the incremented the value to 7 upwards! But thanks for the remainder :)

aldoborrero, 2016-08-25:
Hi @dariuszseweryn 

I have another related question with what you have shared yesterday I can subscribe perfectly to the notifications and I can map those to a pair to latter transform the bytes[] values to meaning data.

The same applies if I want to read those characteristics that allows reading with the following snippet of code:

```
    connectionObservable.flatMap(connection ->
        connection.discoverServices()
            .flatMap(services -> services.getService(ALDeviceUuids.SERVICE_ID).map(BluetoothGattService::getCharacteristics))
            .flatMap(Observable::from)
            .filter(characteristic -> BleUtils.hasReadProperty(characteristic.getProperties()))
            .flatMap(connection::readCharacteristic, Pair::new)
            .compose(valueAdapterTransformer)
    )
        .subscribe(pair -> logger.d(TAG, "Pair: " + pair));
```

I would like to upon the first connection: read an amount of characteristics that has the reading property and **only after that** subscribe to those notifications that have the notification property pretty much as you explained me yesterday.

The thing is that when I do the first part the reading observable won't emit a finish a `doOnComplete` call as is waiting for more, so I can't kickstart or compose the next operation which is subscribing and listening to changes. I know for sure the amount of characteristic that has the read property but I would like to do it in a generic fashion (i.e it doesn't matter if I have 7 or 15 characteristics that I want read, I only want to read them all, write the pojo values, and after that start listening the notifications).

(Sorry to bother you again with more questions about RxJava than the library itself but I've been searching for examples and I can't find nothing valuable). 

Maybe the option is to use an observable that counts the emission of the first one and then after that chain through that to start listening notifications. 

How is the best option to achieve that?

BTW I think that this answers would be good to create a Wiki and explain advanced usage of the library (providing more examples to the sample app). I would gladly do it.

Thanks

dariuszseweryn, 2016-08-25:
Actually this kind of questions should go to stackoverflow.com with the tag `rxandroidble` as these are not questions about the library itself but use-cases.

aldoborrero, 2016-08-25:
Yeah, you're right :)

closed 2016-08-25 by aldoborrero