dariuszseweryn/RxAndroidBle #63: BleGattException CHARACTERISTIC_READ: Device disconnects but BluetoothGatt is not closed properly

WIStudent, 2016-08-27:
I am currently using Bluez 4.51 and its D-Bus API to turn a Raspberry Pi 3 into a BLE peripheral. It provides a single service with a few characteristics to read from and write to. 

The app uses the following code to connect to the periheral and to read some characteristics:
```java
connectionObservable = bleDevice.establishConnection(this, false)
    .compose(bindUntilEvent(ActivityEvent.PAUSE))
    .doOnUnsubscribe(this::clearConnectionObservable)
    .unsubscribeOn(AndroidSchedulers.mainThread())
    .compose(new ConnectionSharingAdapter());

connectionObservable
    .observeOn(AndroidSchedulers.mainThread())
    .subscribe(this::onDeviceConnected, this::onConnectionFailure);

// In onDeviceConnected
connection
    .readCharacteristic(UUID.fromString(row_uuid))
    .observeOn(AndroidSchedulers.mainThread())
    .subscribe(bytes -> setButtonColors(temp_row, bytes), this::onConnectionFailure);
```
As long as nothing unexpected happens, this works as intended. The app searches for the peripheral, connects to it, reads some characteristics and disconnects from it on the pause event.
```
08-27 20:27:28.389 9464-9464/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio:   QUEUED RxBleRadioOperationScan(208722037)
08-27 20:27:28.390 9464-9507/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio:  STARTED RxBleRadioOperationScan(208722037)
08-27 20:27:28.396 9464-9512/de.tobiastrumm.bluetoothledmatrix D/OpenGLRenderer: Use EGL_SWAP_BEHAVIOR_PRESERVED: true
08-27 20:27:28.402 9464-9464/de.tobiastrumm.bluetoothledmatrix D/BluetoothAdapter: startLeScan(): null
08-27 20:27:28.404 9464-9464/de.tobiastrumm.bluetoothledmatrix D/BluetoothAdapter: STATE_ON
08-27 20:27:28.409 9464-9482/de.tobiastrumm.bluetoothledmatrix D/BluetoothLeScanner: onClientRegistered() - status=0 clientIf=5
08-27 20:27:28.412 9464-9507/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio: FINISHED RxBleRadioOperationScan(208722037)
08-27 20:27:28.483 9464-9512/de.tobiastrumm.bluetoothledmatrix I/Adreno: QUALCOMM build                   : 63c06b2, I8366cd0437
                                                                         Build Date                       : 12/06/15
                                                                         OpenGL ES Shader Compiler Version: XE031.05.13.02
                                                                         Local Branch                     : mybranch17112971
                                                                         Remote Branch                    : quic/LA.BF64.1.2.9_v2
                                                                         Remote Branch                    : NONE
                                                                         Reconstruct Branch               : NOTHING
08-27 20:27:28.489 9464-9512/de.tobiastrumm.bluetoothledmatrix I/OpenGLRenderer: Initialized EGL, version 1.4
08-27 20:27:32.754 9464-9464/de.tobiastrumm.bluetoothledmatrix D/BluetoothAdapter: stopLeScan()
08-27 20:27:32.757 9464-9464/de.tobiastrumm.bluetoothledmatrix D/BluetoothAdapter: STATE_ON
08-27 20:27:32.837 9464-9464/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio:   QUEUED RxBleRadioOperationConnect(33739039)
08-27 20:27:32.838 9464-9507/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio:  STARTED RxBleRadioOperationConnect(33739039)
08-27 20:27:32.861 9464-9464/de.tobiastrumm.bluetoothledmatrix D/BluetoothGatt: connect() - device: B8:27:EB:A7:1B:9D, auto: false
08-27 20:27:32.862 9464-9464/de.tobiastrumm.bluetoothledmatrix D/BluetoothGatt: registerApp()
08-27 20:27:32.862 9464-9464/de.tobiastrumm.bluetoothledmatrix D/BluetoothGatt: registerApp() - UUID=48690a14-deaa-4d00-9907-bb50ddeef349
08-27 20:27:32.866 9464-9482/de.tobiastrumm.bluetoothledmatrix D/BluetoothGatt: onClientRegistered() - status=0 clientIf=5
08-27 20:27:34.204 9464-9482/de.tobiastrumm.bluetoothledmatrix D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=5 device=B8:27:EB:A7:1B:9D
08-27 20:27:34.225 9464-9482/de.tobiastrumm.bluetoothledmatrix D/RxBle#BluetoothGatt: onConnectionStateChange newState=2 status=0
08-27 20:27:34.254 9464-9507/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio: FINISHED RxBleRadioOperationConnect(33739039)
08-27 20:27:34.280 9464-9464/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio:   QUEUED RxBleRadioOperationServicesDiscover(105210533)
08-27 20:27:34.282 9464-9507/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio:  STARTED RxBleRadioOperationServicesDiscover(105210533)
08-27 20:27:34.295 9464-9464/de.tobiastrumm.bluetoothledmatrix D/BluetoothGatt: discoverServices() - device: B8:27:EB:A7:1B:9D
08-27 20:27:34.301 9464-9482/de.tobiastrumm.bluetoothledmatrix D/BluetoothGatt: onSearchComplete() = Device=B8:27:EB:A7:1B:9D Status=0
08-27 20:27:34.302 9464-9482/de.tobiastrumm.bluetoothledmatrix D/RxBle#BluetoothGatt: onServicesDiscovered status=0
08-27 20:27:34.329 9464-9664/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(7225693)
08-27 20:27:34.332 9464-9664/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(123841184)
08-27 20:27:34.334 9464-9664/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(142559999)
08-27 20:27:34.336 9464-9664/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(43676202)
08-27 20:27:34.338 9464-9664/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(143636369)
08-27 20:27:34.341 9464-9664/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(20737380)
08-27 20:27:34.343 9464-9664/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(15134611)
08-27 20:27:34.345 9464-9664/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(218267342)
08-27 20:27:34.347 9464-9507/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio: FINISHED RxBleRadioOperationServicesDiscover(105210533)
08-27 20:27:34.348 9464-9507/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio:  STARTED RxBleRadioOperationCharacteristicRead(7225693)
08-27 20:27:34.420 9464-9482/de.tobiastrumm.bluetoothledmatrix D/RxBle#BluetoothGatt: onCharacteristicRead characteristic=12345678-1234-5678-1234-56789abc0000 status=0
08-27 20:27:34.426 9464-9507/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio: FINISHED RxBleRadioOperationCharacteristicRead(7225693)
08-27 20:27:34.427 9464-9507/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio:  STARTED RxBleRadioOperationCharacteristicRead(123841184)
08-27 20:27:34.512 9464-9482/de.tobiastrumm.bluetoothledmatrix D/RxBle#BluetoothGatt: onCharacteristicRead characteristic=12345678-1234-5678-1234-56789abc0001 status=0
08-27 20:27:34.523 9464-9507/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio: FINISHED RxBleRadioOperationCharacteristicRead(123841184)
08-27 20:27:34.525 9464-9507/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio:  STARTED RxBleRadioOperationCharacteristicRead(142559999)
08-27 20:27:34.602 9464-9482/de.tobiastrumm.bluetoothledmatrix D/RxBle#BluetoothGatt: onCharacteristicRead characteristic=12345678-1234-5678-1234-56789abc0002 status=0
[... 220 lines cut ...]
                                                                                           at com.polidea.rxandroidble.internal.connection.RxBleGattCallback.access$100(RxBleGattCallback.java:26)
                                                                                           at com.polidea.rxandroidble.internal.connection.RxBleGattCallback$1.onCharacteristicRead(RxBleGattCallback.java:94)
                                                                                           at android.bluetooth.BluetoothGatt$1.onCharacteristicRead(BluetoothGatt.java:356)
                                                                                           at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:249)
                                                                                           at android.os.Binder.execTransact(Binder.java:453)
08-27 20:30:30.952 12770-12785/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio: FINISHED RxBleRadioOperationCharacteristicRead(218267342)
08-27 20:30:30.953 12770-12785/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio:  STARTED RxBleRadioOperationDisconnect(223066586)
[... cut ...]
08-27 20:30:31.005 12770-12781/de.tobiastrumm.bluetoothledmatrix D/RxBle#BluetoothGatt: onConnectionStateChange newState=0 status=0
```
The reason for that could be that Bluez does not assign the same handles to services and characteristics if they are removed and added again. But because BluetoothGatt was not closed, Android still expects the characteristics to be reachable under the hold handles, I guess.

The only way I could get the app running again was to disable and then re-enable bluetooth on the Android device.

WIStudent, 2016-08-27:
Some additional infos: I am using a Nexus 5X with Android 6.0.1

dariuszseweryn, 2016-09-08:
Hello there,
I have been thinking how to address this issue. I currently have no possibility of test this scenario.
I can give you the change I think would fix the problem - would you be so kind to test it?

The thing is to change the `isDisconnected()` function implementation in `RxBleRadioOperationDisconnect` class. It would go like this:

    private boolean isDisconnected(BluetoothGatt bluetoothGatt) {
        return bluetoothManager.getConnectionState(bluetoothGatt.getDevice(), BluetoothProfile.GATT) == BluetoothProfile.STATE_DISCONNECTED
                && rxBleGattCallback.getOnConnectionStateChange().toBlocking().first() == RxBleConnection.RxBleConnectionState.DISCONNECTED;
    }

WIStudent, 2016-09-08:
I am not sure if I applied your changes correctly, but it seams that BluetoothGatt is still not closed
```
09-08 19:48:18.962 26825-27052/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(25496177)
09-08 19:48:18.966 26825-27052/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(116164036)
09-08 19:48:18.970 26825-27052/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(254716787)
09-08 19:48:18.972 26825-27052/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(1233966)
09-08 19:48:18.974 26825-27052/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(43775077)
09-08 19:48:18.976 26825-27052/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(202579784)
09-08 19:48:18.978 26825-27052/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(27928007)
09-08 19:48:18.981 26825-27052/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(171573650)
09-08 19:48:18.982 26825-27049/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio: FINISHED RxBleRadioOperationServicesDiscover(266080148)
09-08 19:48:18.984 26825-27049/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio:  STARTED RxBleRadioOperationCharacteristicRead(25496177)
09-08 19:48:19.039 26825-26869/de.tobiastrumm.bluetoothledmatrix D/RxBle#BluetoothGatt: onCharacteristicRead characteristic=12345678-1234-5678-1234-56789abc0000 status=1
09-08 19:48:19.049 26825-26869/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio:   QUEUED RxBleRadioOperationDisconnect(176898444)
09-08 19:48:19.051 26825-27049/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio: FINISHED RxBleRadioOperationCharacteristicRead(25496177)
09-08 19:48:19.053 26825-27049/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio:  STARTED RxBleRadioOperationCharacteristicRead(116164036)
09-08 19:48:19.068 26825-26825/de.tobiastrumm.bluetoothledmatrix E/LedControlActivity: BleGattException{status=1, bleGattOperation=BleGattOperation{description='CHARACTERISTIC_READ'}}
                                                                                           at com.polidea.rxandroidble.internal.connection.RxBleGattCallback.propagateStatusErrorIfGattErrorOccurred(RxBleGattCallback.java:248)
                                                                                           at com.polidea.rxandroidble.internal.connection.RxBleGattCallback.access$100(RxBleGattCallback.java:27)
                                                                                           at com.polidea.rxandroidble.internal.connection.RxBleGattCallback$1.onCharacteristicRead(RxBleGattCallback.java:97)
                                                                                           at android.bluetooth.BluetoothGatt$1.onCharacteristicRead(BluetoothGatt.java:356)
                                                                                           at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:249)
                                                                                           at android.os.Binder.execTransact(Binder.java:453)
09-08 19:48:19.082 26825-26825/de.tobiastrumm.bluetoothledmatrix E/LedControlActivity: BleGattException{status=1, bleGattOperation=BleGattOperation{description='CHARACTERISTIC_READ'}}
                                                                                           at com.polidea.rxandroidble.internal.connection.RxBleGattCallback.propagateStatusErrorIfGattErrorOccurred(RxBleGattCallback.java:248)
                                                                                           at com.polidea.rxandroidble.internal.connection.RxBleGattCallback.access$100(RxBleGattCallback.java:27)
                                                                                           at com.polidea.rxandroidble.internal.connection.RxBleGattCallback$1.onCharacteristicRead(RxBleGattCallback.java:97)
                                                                                           at android.bluetooth.BluetoothGatt$1.onCharacteristicRead(BluetoothGatt.java:356)
                                                                                           at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:249)
                                                                                           at android.os.Binder.execTransact(Binder.java:453)
09-08 19:48:19.085 26825-27049/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio: FINISHED RxBleRadioOperationCharacteristicRead(116164036)
09-08 19:48:19.086 26825-27049/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio:  STARTED RxBleRadioOperationCharacteristicRead(254716787)
09-08 19:48:19.163 26825-26825/de.tobiastrumm.bluetoothledmatrix E/LedControlActivity: BleGattException{status=1, bleGattOperation=BleGattOperation{description='CHARACTERISTIC_READ'}}
                                                                                           at com.polidea.rxandroidble.internal.connection.RxBleGattCallback.propagateStatusErrorIfGattErrorOccurred(RxBleGattCallback.java:248)
[... cut ...]
For me personally this isn't a big issue because all I wanted to do was to experiment with BLE and to control some LEDs with it. I just have to make sure that the handles don't change while I am using it, or I have to restart bluetooth on my Android device.

But if you want to look further into the issue of changing handles, I put the code that uses BlueZ to create the peripheral and the code for the app on Github a few days ago.
https://github.com/WIStudent/Bluetooth-Low-Energy-LED-Matrix
https://github.com/WIStudent/Bluetooth-LED-Matrix-App

dariuszseweryn, 2016-09-08:
Thanks for sharing the code - I will try to recreate the issue locally.
There are two issues I want to address:
1. Connection is not being properly closed because somehow the final callback comes after the `RxBleOperationDisconnect` finishes like you can see here:

```
09-08 19:48:19.361 26825-27049/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio:  STARTED RxBleRadioOperationDisconnect(176898444)
09-08 19:48:19.375 26825-26825/de.tobiastrumm.bluetoothledmatrix E/LedControlActivity: BleGattException{status=1, bleGattOperation=BleGattOperation{description='CHARACTERISTIC_READ'}}
                                                                                           at com.polidea.rxandroidble.internal.connection.RxBleGattCallback.propagateStatusErrorIfGattErrorOccurred(RxBleGattCallback.java:248)
                                                                                           at com.polidea.rxandroidble.internal.connection.RxBleGattCallback.access$100(RxBleGattCallback.java:27)
                                                                                           at com.polidea.rxandroidble.internal.connection.RxBleGattCallback$1.onCharacteristicRead(RxBleGattCallback.java:97)
                                                                                           at android.bluetooth.BluetoothGatt$1.onCharacteristicRead(BluetoothGatt.java:356)
                                                                                           at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:249)
                                                                                           at android.os.Binder.execTransact(Binder.java:453)
09-08 19:48:19.376 26825-26825/de.tobiastrumm.bluetoothledmatrix D/BluetoothManager: getConnectionState()
09-08 19:48:19.377 26825-26825/de.tobiastrumm.bluetoothledmatrix D/BluetoothManager: getConnectedDevices
09-08 19:48:19.379 26825-26825/de.tobiastrumm.bluetoothledmatrix D/BluetoothGatt: cancelOpen() - device: B8:27:EB:A7:1B:9D
09-08 19:48:19.382 26825-27049/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio: FINISHED RxBleRadioOperationDisconnect(176898444)
09-08 19:48:19.385 26825-26837/de.tobiastrumm.bluetoothledmatrix D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=5 device=B8:27:EB:A7:1B:9D
09-08 19:48:19.386 26825-26837/de.tobiastrumm.bluetoothledmatrix D/RxBle#BluetoothGatt: onConnectionStateChange newState=0 status=0
```

This situation prevent's closing the `BluetoothGatt` even when calling `BluetoothGatt.close()`.

The second thing is to allow using `BluetoothGatt.refresh()` method via reflection. Calling this method should allow proper functioning of BLE devices that change their services / characteristics without putting the `BluetoothAdapter` through on/off cycle.

lavenj, 2016-09-26:
Hi @DariuszAniszewski -- I'm seeing the same output in my project. It seems like the disconnect callbacks are being fired before the connection actually closes. Is this a correct assessment? Any possibility for a bugfix?

dariuszseweryn, 2016-10-25:
@lavenj @WIStudent Sorry for such a long wait. I have a possible bugfix for the situation where the `BluetoothGatt.close()` method is being called before the Android actually disconnects the `BluetoothGatt`.
I could either push the possible fix to a branch for you to test or I will try to setup the test procedure myself.
Best Regards

WIStudent, 2016-10-25:
I moved on to a different project so I can't test it at the moment.

lavenj, 2016-11-02:
@dariuszseweryn thanks for looking at this! We have a reasonable workaround implemented so are OK for now. This library continues to be a huge help. 

Off-topic but discovered RxBleAdapterStateObservable while digging through the source. So clean and easy! You should throw it in the docs if it's not already in there 😄

dariuszseweryn, 2016-11-03:
@lavenj can you share more info about the workaround?
About the `RxBleAdapterStateObservable` I will think about mentioning it the docs. ;)

@WIStudent I am trying to use your Raspberry Pi 3 script to recreate the issue. I have installed a fresh Raspbian on the RPi3 and commented out all interactions with the display from the source code. I have enabled the bluetooth by calling `sudo systemctl enable bluetooth` but still the script returns:
```
GattManager1 interface not found
LEAdvertisingManager1 interface not found
Traceback (most recent call last):
  File "led_matrix.py", line 216, in <module>
    main()
  File "led_matrix.py", line 194, in main
    service_manager.RegisterApplication(app.get_path(), {},
AttributeError: 'NoneType' object has no attribute 'RegisterApplication'
```
I am not very experienced with linux and Raspberry Pi configuration so maybe you could point me in the right direction. Possible causes I can think off:
- I should enable something in the system prior to running the script
- I must update the BlueZ 5.23 -> 5.41 (rather a cumbersome process from what I have googled so far)
- I lack of some dependencies with the DBus but the imports didn't failed so I don't think so
Any hints you could give me?

WIStudent, 2016-11-03:
I highly recommend updating Bluez to the latest version (which is currently 5.43) . Some of its BLE features are quite new and might not be impemented in an older version. I used [this tutorial](https://learn.adafruit.com/install-bluez-on-the-raspberry-pi/installation) to compile and install Bluez from source on a Raspberry Pi. It should only take a few minutes. 

If you don't want to update, you need to set at least the `--experimental` flag in the bluez service configuration in order to access the missing dbus interfaces. The process is described at the end of the tutorial. 

If you have still issues accessing the dbus interfaces after that, make sure that you have the packages `python-gi`, `python3-gi` and `python-dbus` installed. 

`sudo apt-get install python-gi python3-gi python-dbus` 

The first two will install the python 2 and 3 bindings for GObject, the third one will install the python-dbus library.

If you downloaded the bluez source directory, you can also try to run the `example-gatt-server` and the `example-advertisement` python script from the `bluez-5.xx/test` folder. The first one creates a heart rate GATT service and the second one creates the corresponding advertisement. They use the same libraries I used in my script.

dariuszseweryn, 2016-11-04:
@WIStudent 
Thank you for the instructions. I have updated the BlueZ according to the tutorial flawlessly. I have checked your code against current `master` branch using code:
```
rxBleClient.getBleDevice(macAddress)
                .establishConnection(this, false)
                .flatMap(rxBleConnection -> Observable
                        .interval(1, TimeUnit.SECONDS)
                        .flatMap(i -> rxBleConnection.readCharacteristic(UUID.fromString("00002a38-0000-1000-8000-00805f9b34fb")))
                )
                .subscribe(
                        read -> Log.d("read", "read"),
                        error -> Log.e("XXX", "Error", error)
                );
```
which connects to the RPi3 and every second it tries to read a characteristic. Then at some point I closed the gatt server script and got a proper closing of the `BluetoothGatt`.
```
11-04 14:42:31.026 25628-26420/com.polidea.rxandroidble.sample D/read: read
11-04 14:42:31.028 25628-25641/com.polidea.rxandroidble.sample D/RxBle#Radio: FINISHED RxBleRadioOperationCharacteristicRead(203958134)
11-04 14:42:31.924 25628-30697/com.polidea.rxandroidble.sample D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(21262938)
11-04 14:42:31.924 25628-25641/com.polidea.rxandroidble.sample D/RxBle#Radio:  STARTED RxBleRadioOperationCharacteristicRead(21262938)
11-04 14:42:32.035 25628-25639/com.polidea.rxandroidble.sample D/RxBle#BluetoothGatt: onCharacteristicRead characteristic=00002a38-0000-1000-8000-00805f9b34fb status=0
11-04 14:42:32.036 25628-26420/com.polidea.rxandroidble.sample D/read: read
11-04 14:42:32.036 25628-25641/com.polidea.rxandroidble.sample D/RxBle#Radio: FINISHED RxBleRadioOperationCharacteristicRead(21262938)
11-04 14:42:32.924 25628-30697/com.polidea.rxandroidble.sample D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(86345726)
11-04 14:42:32.925 25628-25641/com.polidea.rxandroidble.sample D/RxBle#Radio:  STARTED RxBleRadioOperationCharacteristicRead(86345726)
11-04 14:42:32.995 25628-25658/com.polidea.rxandroidble.sample D/RxBle#BluetoothGatt: onCharacteristicRead characteristic=00002a38-0000-1000-8000-00805f9b34fb status=0
11-04 14:42:32.996 25628-26420/com.polidea.rxandroidble.sample D/read: read
11-04 14:42:32.996 25628-25641/com.polidea.rxandroidble.sample D/RxBle#Radio: FINISHED RxBleRadioOperationCharacteristicRead(86345726)
11-04 14:42:33.929 25628-30697/com.polidea.rxandroidble.sample D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(116747362)
11-04 14:42:33.943 25628-25641/com.polidea.rxandroidble.sample D/RxBle#Radio:  STARTED RxBleRadioOperationCharacteristicRead(116747362)
11-04 14:42:34.076 25628-25640/com.polidea.rxandroidble.sample D/RxBle#BluetoothGatt: onCharacteristicRead characteristic=00002a38-0000-1000-8000-00805f9b34fb status=1
11-04 14:42:34.076 25628-25640/com.polidea.rxandroidble.sample I/RxBle#RadioOperation: onError
                                                                                       BleGattException{status=1, bleGattOperation=BleGattOperation{description='CHARACTERISTIC_READ'}}
                                                                                           at com.polidea.rxandroidble.internal.connection.RxBleGattCallback.propagateStatusErrorIfGattErrorOccurred(RxBleGattCallback.java:248)
                                                                                           at com.polidea.rxandroidble.internal.connection.RxBleGattCallback.access$100(RxBleGattCallback.java:25)
                                                                                           at com.polidea.rxandroidble.internal.connection.RxBleGattCallback$1.onCharacteristicRead(RxBleGattCallback.java:97)
                                                                                           at android.bluetooth.BluetoothGatt$1.onCharacteristicRead(BluetoothGatt.java:356)
                                                                                           at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:249)
                                                                                           at android.os.Binder.execTransact(Binder.java:453)
11-04 14:42:34.077 25628-25640/com.polidea.rxandroidble.sample E/XXX: Error
                                                                      BleGattException{status=1, bleGattOperation=BleGattOperation{description='CHARACTERISTIC_READ'}}
                                                                          at com.polidea.rxandroidble.internal.connection.RxBleGattCallback.propagateStatusErrorIfGattErrorOccurred(RxBleGattCallback.java:248)
                                                                          at com.polidea.rxandroidble.internal.connection.RxBleGattCallback.access$100(RxBleGattCallback.java:25)
                                                                          at com.polidea.rxandroidble.internal.connection.RxBleGattCallback$1.onCharacteristicRead(RxBleGattCallback.java:97)
                                                                          at android.bluetooth.BluetoothGatt$1.onCharacteristicRead(BluetoothGatt.java:356)
                                                                          at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:249)
                                                                          at android.os.Binder.execTransact(Binder.java:453)
[... cut ...]
So perhaps some recent fix has unintentionally fixed this issue as well. I was checking with Nexus 5 and Nexus 5X both with Android 6.0.1. Fun fact: I couldn't get any device with Android < 6.0 to connect to the Raspberry Pi 3 - always getting status=133 (GATT_ERROR) when connecting.

It would be awesome if you could re-check this issue using the SNAPSHOT release or the newest version if it will be available before.

Yet - I still need to add a `refresh()` function for clearing the `BluetoothGatt` state.

WIStudent, 2016-11-06:
I tested it again with the 1.1.0-SNAPSHOT. I used bluez-5.41 again on the Pi to rule out any changes on that side. But Unfortunately I have upgraded my Nexus 5x to Android 7.0 in the meantime so I can't tell if the problem is solved in Android 6.

Interestingly the behavior changed in Android 7 again. Reading from and writing to an existing characteristic works as expected.
```
11-06 22:01:59.785 28082-28082/de.tobiastrumm.bluetoothledmatrix D/LedControlActivity: Pressed button [0][0]
11-06 22:01:59.794 28082-28082/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(90263684)
11-06 22:01:59.795 28082-28082/de.tobiastrumm.bluetoothledmatrix D/LedMatrix: Old: Col 0 Color 1: 00000000 00000000
11-06 22:01:59.796 28082-28105/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio:  STARTED RxBleRadioOperationCharacteristicRead(90263684)
11-06 22:01:59.796 28082-28082/de.tobiastrumm.bluetoothledmatrix D/LedMatrix: New: Col 0 Color 1: 01000000 00000000
11-06 22:01:59.805 28082-28082/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicWrite(58975849)
11-06 22:01:59.960 28082-28098/de.tobiastrumm.bluetoothledmatrix W/BluetoothGatt: onCharacteristicRead() - Device=B8:27:EB:A7:1B:9D handle=48 Status=0
11-06 22:01:59.962 28082-28098/de.tobiastrumm.bluetoothledmatrix D/RxBle#BluetoothGatt: onCharacteristicRead characteristic=12345678-1234-5678-1234-56789abc0000 status=0
11-06 22:01:59.965 28082-28214/de.tobiastrumm.bluetoothledmatrix D/LedControlActivity: read
11-06 22:01:59.968 28082-28105/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio: FINISHED RxBleRadioOperationCharacteristicRead(90263684)
11-06 22:01:59.968 28082-28105/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio:  STARTED RxBleRadioOperationCharacteristicWrite(58975849)
11-06 22:02:00.079 28082-28094/de.tobiastrumm.bluetoothledmatrix D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=12345678-1234-5678-1234-56789abc0000 status=0
11-06 22:02:00.081 28082-28214/de.tobiastrumm.bluetoothledmatrix D/LedControlActivity: write
11-06 22:02:00.081 28082-28105/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio: FINISHED RxBleRadioOperationCharacteristicWrite(58975849)
```
Bluez on the Pi received and sent the following packets (Use `sudo btmon` see the packets in real time)
```
> ACL Data RX: Handle 64 flags 0x02 dlen 7                   [hci0] 1357.182785
      ATT: Read Request (0x0a) len 2
        Handle: 0x0030
< ACL Data TX: Handle 64 flags 0x00 dlen 7                   [hci0] 1357.185679
      ATT: Read Response (0x0b) len 2
        Value: 0000
> ACL Data RX: Handle 64 flags 0x02 dlen 9                   [hci0] 1357.302857
      ATT: Write Request (0x12) len 4
        Handle: 0x0030
          Data: 4000
< ACL Data TX: Handle 64 flags 0x00 dlen 5                   [hci0] 1357.312835
      ATT: Write Response (0x13) len 0
```
The next log shows the attempt to read and write the same characteristic after it was removed:
```
11-06 22:04:49.168 28082-28082/de.tobiastrumm.bluetoothledmatrix D/LedControlActivity: Pressed button [0][0]
11-06 22:04:49.171 28082-28082/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(229704388)
11-06 22:04:49.172 28082-28082/de.tobiastrumm.bluetoothledmatrix D/LedMatrix: Old: Col 0 Color 2: 01010000 00000000
11-06 22:04:49.172 28082-28105/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio:  STARTED RxBleRadioOperationCharacteristicRead(229704388)
11-06 22:04:49.172 28082-28082/de.tobiastrumm.bluetoothledmatrix D/LedMatrix: New: Col 0 Color 2: 10010000 00000000
11-06 22:04:49.175 28082-28082/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicWrite(48455593)
11-06 22:04:49.281 28082-28094/de.tobiastrumm.bluetoothledmatrix W/BluetoothGatt: onCharacteristicRead() - Device=B8:27:EB:A7:1B:9D handle=48 Status=1
11-06 22:04:49.283 28082-28094/de.tobiastrumm.bluetoothledmatrix D/RxBle#BluetoothGatt: onCharacteristicRead characteristic=12345678-1234-5678-1234-56789abc0000 status=1
11-06 22:04:49.286 28082-28105/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio: FINISHED RxBleRadioOperationCharacteristicRead(229704388)
11-06 22:04:49.286 28082-28094/de.tobiastrumm.bluetoothledmatrix E/LedControlActivity: Read: Error
                                                                                       BleGattException{status=1, bleGattOperation=BleGattOperation{description='CHARACTERISTIC_READ'}}
                                                                                           at com.polidea.rxandroidble.internal.connection.RxBleGattCallback.propagateStatusErrorIfGattErrorOccurred(RxBleGattCallback.java:248)
                                                                                           at com.polidea.rxandroidble.internal.connection.RxBleGattCallback.access$100(RxBleGattCallback.java:25)
                                                                                           at com.polidea.rxandroidble.internal.connection.RxBleGattCallback$1.onCharacteristicRead(RxBleGattCallback.java:97)
                                                                                           at android.bluetooth.BluetoothGatt$1.onCharacteristicRead(BluetoothGatt.java:287)
                                                                                           at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:117)
[... cut ...]
        Error: Invalid Handle (0x01)
> ACL Data RX: Handle 64 flags 0x02 dlen 9                   [hci0] 1526.625134
      ATT: Write Request (0x12) len 4
        Handle: 0x0030
          Data: 9000
< ACL Data TX: Handle 64 flags 0x00 dlen 9                   [hci0] 1526.625376
      ATT: Error Response (0x01) len 4
        Write Request (0x12)
        Handle: 0x0030
        Error: Invalid Handle (0x01)
```
The BleGattExceptions are still thrown but instead of closing the connection automatically it is now kept alive.

dariuszseweryn, 2016-11-07:
I wish I had written tests for `RxBleGattCallback` earlier. No disconnection when receiving status error is a behaviour planned for release 2.0.0 but it is a bug in 1.x.x.

dariuszseweryn, 2016-11-10:
@WIStudent Excuse me for asking again but I have fixed the behaviour to be inline with 1.x.x and it should be working fine now. (I hope... - it at least works as intended for me)
Thanks! Cheers!

WIStudent, 2016-11-18:
I switched to the 1.2.0-SNAPSHOT and tested it again. Now I see the same behavior as you do, 
```
11-18 21:01:59.212 23273-23273/de.tobiastrumm.bluetoothledmatrix D/LedControlActivity: Pressed button [2][3]
11-18 21:01:59.217 23273-23273/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(75725609)
11-18 21:01:59.219 23273-23273/de.tobiastrumm.bluetoothledmatrix D/LedMatrix: Old: Col 3 Color 2: 00000001 00000000
11-18 21:01:59.220 23273-23273/de.tobiastrumm.bluetoothledmatrix D/LedMatrix: New: Col 3 Color 2: 00000010 00000000
11-18 21:01:59.220 23273-23627/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio:  STARTED RxBleRadioOperationCharacteristicRead(75725609)
11-18 21:01:59.226 23273-23273/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicWrite(57374650)
11-18 21:01:59.355 23273-23285/de.tobiastrumm.bluetoothledmatrix W/BluetoothGatt: onCharacteristicRead() - Device=B8:27:EB:A7:1B:9D handle=35 Status=1
11-18 21:01:59.357 23273-23285/de.tobiastrumm.bluetoothledmatrix D/RxBle#BluetoothGatt: onCharacteristicRead characteristic=12345678-1234-5678-1234-56789abc0002 status=1
11-18 21:01:59.363 23273-23285/de.tobiastrumm.bluetoothledmatrix E/LedControlActivity: Read: Error
                                                                                       BleGattException{status=1, bleGattOperation=BleGattOperation{description='CHARACTERISTIC_READ'}}
                                                                                           at com.polidea.rxandroidble.internal.connection.RxBleGattCallback.propagateStatusErrorIfGattErrorOccurred(RxBleGattCallback.java:248)
                                                                                           at com.polidea.rxandroidble.internal.connection.RxBleGattCallback.access$100(RxBleGattCallback.java:25)
                                                                                           at com.polidea.rxandroidble.internal.connection.RxBleGattCallback$1.onCharacteristicRead(RxBleGattCallback.java:97)
                                                                                           at android.bluetooth.BluetoothGatt$1.onCharacteristicRead(BluetoothGatt.java:287)
                                                                                           at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:117)
                                                                                           at android.os.Binder.execTransact(Binder.java:565)
11-18 21:01:59.363 23273-23285/de.tobiastrumm.bluetoothledmatrix E/LedControlActivity: Write: Error
                                                                                       BleGattException{status=1, bleGattOperation=BleGattOperation{description='CHARACTERISTIC_READ'}}
                                                                                           at com.polidea.rxandroidble.internal.connection.RxBleGattCallback.propagateStatusErrorIfGattErrorOccurred(RxBleGattCallback.java:248)
                                                                                           at com.polidea.rxandroidble.internal.connection.RxBleGattCallback.access$100(RxBleGattCallback.java:25)
                                                                                           at com.polidea.rxandroidble.internal.connection.RxBleGattCallback$1.onCharacteristicRead(RxBleGattCallback.java:97)
                                                                                           at android.bluetooth.BluetoothGatt$1.onCharacteristicRead(BluetoothGatt.java:287)
                                                                                           at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:117)
                                                                                           at android.os.Binder.execTransact(Binder.java:565)
11-18 21:01:59.364 23273-23285/de.tobiastrumm.bluetoothledmatrix E/LedControlActivity: Read: Error
                                                                                       BleGattException{status=1, bleGattOperation=BleGattOperation{description='CHARACTERISTIC_READ'}}
                                                                                           at com.polidea.rxandroidble.internal.connection.RxBleGattCallback.propagateStatusErrorIfGattErrorOccurred(RxBleGattCallback.java:248)
                                                                                           at com.polidea.rxandroidble.internal.connection.RxBleGattCallback.access$100(RxBleGattCallback.java:25)
                                                                                           at com.polidea.rxandroidble.internal.connection.RxBleGattCallback$1.onCharacteristicRead(RxBleGattCallback.java:97)
                                                                                           at android.bluetooth.BluetoothGatt$1.onCharacteristicRead(BluetoothGatt.java:287)
                                                                                           at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:117)
                                                                                           at android.os.Binder.execTransact(Binder.java:565)
11-18 21:01:59.364 23273-23285/de.tobiastrumm.bluetoothledmatrix E/LedControlActivity: Write: Error
[... cut ...]
11-18 21:01:59.495 23273-23286/de.tobiastrumm.bluetoothledmatrix D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=5 device=B8:27:EB:A7:1B:9D
11-18 21:01:59.496 23273-23286/de.tobiastrumm.bluetoothledmatrix D/RxBle#BluetoothGatt: onConnectionStateChange newState=0 status=0
11-18 21:01:59.497 23273-23627/de.tobiastrumm.bluetoothledmatrix D/RxBle#Radio: FINISHED RxBleRadioOperationDisconnect(221169268)
11-18 21:01:59.539 23273-23273/de.tobiastrumm.bluetoothledmatrix D/BluetoothGatt: close()
11-18 21:01:59.539 23273-23273/de.tobiastrumm.bluetoothledmatrix D/BluetoothGatt: unregisterApp() - mClientIf=5
```

dariuszseweryn, 2016-11-18:
So it seems that cause of this issue (that is in the topic) is resolved.
The error occurs because the handles had changed but the connection is disconnected and closed properly. Thank you for reporting and feedback. Cheers!

closed 2016-11-18 by dariuszseweryn