dariuszseweryn/RxAndroidBle #64: DESCRIPTOR_WRITE error with setupNotifications()

cpboyd, 2016-08-29:
### Summary
I'm getting a "DESCRIPTOR_WRITE" error when attempting to setup notifications for a specific characteristic UUID.

The device only has a single service UUID (BLE_TOS_SERVICE) with the specified characteristic UUID (BLE_TOS_CHARACTERISTIC).
<br />
#### Preconditions
This code works to subscribe to notifications using Android's BluetoothGatt.
```Java
        BluetoothAdapter.getDefaultAdapter().startDiscovery();
        BluetoothAdapter.getDefaultAdapter().cancelDiscovery();
        BluetoothAdapter.getDefaultAdapter().getRemoteDevice(mDeviceAddress).connectGatt(getApplicationContext(), false, new BluetoothGattCallback() {
            @Override
            public void onConnectionStateChange(BluetoothGatt gatt, int status, int newState) {
                Log.d(TAG, "State: " + newState);
                if (newState == BluetoothProfile.STATE_CONNECTED) {
                    gatt.discoverServices();
                }
            }
            @Override
            public void onServicesDiscovered(BluetoothGatt gatt, int status) {
                Log.d(TAG, "Services: " + gatt.getServices().toString());
                for(BluetoothGattService service : gatt.getServices()) {
                    Log.d(TAG, "Service " + service.getUuid().toString());
                    for(BluetoothGattCharacteristic charact : service.getCharacteristics()) {
                        Log.d(TAG, "  Characteristic " + charact.getUuid().toString());
                    }
                }
                BluetoothGattCharacteristic characteristic = gatt.getService(BLE_TOS_SERVICE).getCharacteristic(BLE_TOS_CHARACTERISTIC);
                gatt.setCharacteristicNotification(characteristic, true);
            }
        });
```
<br />
#### Steps to reproduce actual result
This is code based on the example.
```Java
        rxBleClient = RxBleClient.create(getApplicationContext());
        device = rxBleClient.getBleDevice(mDeviceAddress);
        observable = device.establishConnection(getApplicationContext(), true);

        subscription = observable
                .flatMap(rxBleConnection -> rxBleConnection.setupNotification(BLE_TOS_CHARACTERISTIC))
                .doOnNext(notificationObservable -> {
                    // Notification has been set up
                    Log.d(TAG, "BLE Notifications!");
                })
                .flatMap(notificationObservable -> notificationObservable) // <-- Notification has been set up, now observe value changes.
                .subscribe(bytes -> {
                    // Given characteristic has been changes, here is the value.
                    Log.d(TAG, "Bytes: " + bytes.length);
                }, ex -> Log.e(TAG, "Notif error", ex));
```
#### Actual result
When I try to use the above code for setting up notifications, I get an error:
```
BleGattException{status=3, bleGattOperation=BleGattOperation{description='DESCRIPTOR_WRITE'}}
   at com.polidea.rxandroidble.internal.connection.RxBleGattCallback.propagateStatusErrorIfGattErrorOccurred(RxBleGattCallback.java:245)
   at com.polidea.rxandroidble.internal.connection.RxBleGattCallback.access$100(RxBleGattCallback.java:26)
   at com.polidea.rxandroidble.internal.connection.RxBleGattCallback$1.onDescriptorWrite(RxBleGattCallback.java:153)
   at android.bluetooth.BluetoothGatt$1.onDescriptorWrite(BluetoothGatt.java:548)
   at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:369)
   at android.os.Binder.execTransact(Binder.java:453)
```
<br />

I don't get any error when setting up notifications using the BluetoothGatt commands, and it seems like RxAndroidBle should use those underneath it all, so I'm not really sure what's going on.

dariuszseweryn, 2016-08-30:
Hello @cpboyd 

From what I see in the code you have pasted - your code only does `.setCharacteristicNotification(characteristic, true)` where to get the notifications callbacks you need also enable the notifications by setting the so-called `Client Characteristic Config` which has `UUID.fromString("00002902-0000-1000-8000-00805f9b34fb")` with value of `BluetoothGattDescriptor.ENABLE_NOTIFICATION_VALUE`. 
From what I see there is an error on writing the value to the descriptor with `status=3` which corresponds to `GATT_WRITE_NOT_PERMIT` (check https://android.googlesource.com/platform/external/bluetooth/bluedroid/+/android-5.1.0_r1/stack/include/gatt_api.h ). It seems that your device is misconfigured or you're trying to set a notification on a characteristic that doesn't support it.
You can check it by yourself if alongside of `.setCharacteristicNotification(characteristic, true)` you can write `BluetoothGattDescriptor.ENABLE_NOTIFICATION_VALUE` to the `Client Characteristic Config`.

Best Regards

cpboyd, 2016-08-30:
I've updated the code in the "Preconditions" section above.
I tried writing to the specified descriptor and it seems to work there.

This is the output I get:
```
BleDeviceActivity: Service 00001800-0000-1000-8000-00805f9b34fb
BleDeviceActivity:   Characteristic 00002a00-0000-1000-8000-00805f9b34fb
BleDeviceActivity:   Characteristic 00002a01-0000-1000-8000-00805f9b34fb
BleDeviceActivity: Service 0000180a-0000-1000-8000-00805f9b34fb
BleDeviceActivity:   Characteristic 00002a29-0000-1000-8000-00805f9b34fb
BleDeviceActivity:   Characteristic 00002a24-0000-1000-8000-00805f9b34fb
BleDeviceActivity:   Characteristic 00002a23-0000-1000-8000-00805f9b34fb
BleDeviceActivity: Service e079c6a0-aa8b-11e3-a903-0002a5d5c51b
BleDeviceActivity:   Characteristic b38312c0-aa89-11e3-9cef-0002a5d5c51b
BleDeviceActivity:     Descriptor 00002902-0000-1000-8000-00805f9b34fb
BluetoothGatt: setCharacteristicNotification() - uuid: b38312c0-aa89-11e3-9cef-0002a5d5c51b enable: true
BleDeviceActivity: Test: true | true | true
```

As you can see, the Toshiba SPP-over-LE characteristic appears to have the appropriate descriptor.

Additionally, I've been able to successfully subscribe to notifications in iOS as well and seem to be able to receive the notifications properly.

From perusing the RxAndroidBle code, it seems like it shouldn't have any difficulty finding the appropriate characteristic.

cpboyd, 2016-08-30:
It turns out that the device allows indications, but not notifications.

Thanks for the help!

closed 2016-08-30 by cpboyd

dariuszseweryn, 2016-08-30:
Glad I could help. If you could confirm if in this situation `setupIndications()` will work - I would be grateful.

Best Regards

cpboyd, 2016-08-30:
`setupIndications()` does, in fact, get the correct status `0` (`GATT_SUCCESS`).

Thanks again!