dariuszseweryn/RxAndroidBle #65: While auto reconnecting the library will try to send again previous successfully written characteristics

aldoborrero, 2016-08-30:
Hi guys!

(Feel free to categorise this issue as question and I will forward them to StackOverflow if you believe that is more a question than an issue but I wanted to stay sure).

For now what I observed is when I establish a connection, pretty much like this:

```java
connectionObservable = rxBleDevice.establishConnection(context, false)
          .takeUntil(disconnectionTrigger)
          .doOnUnsubscribe(this::onUnsubscribeConnection)
          .compose(new ConnectionSharingAdapter())
          .retryWhen(this::onShouldRetryOperation);

      connectionObservable.subscribe(this::onConnectionSuccess, this::onConnectionFailure);
```

And later on when I write a characteristic (triggered by the user at some point of time after establishing the connection, just like clicking on a button, I don't care about the result of the operation):

```java
connectionObservable
          .observeOn(Schedulers.computation())
          .flatMap(
              connection -> connection.writeCharacteristic(command.getCommandUuid(), command.getData())
          )
          .doOnNext(bytes ->Log.d(TAG, "Command written successfully! Command: " + command))
          .subscribe();
```

The behaviour that I'm facing is that if for whatever reason I change the BT adapter state (from on to off and the again to on), those previous commands that I sent successfully in a response to a user behaviour will again be fired (i.e if I sent 10 write operations, will again send those 10 operations).

Is this behaviour expected?

dariuszseweryn, 2016-08-30:
Hello,

This behaviour is expected. You have posted an approximated version of the code. Depending on how have you implemented the `onConnectionSuccess()` and reacting to the button clicks the previously sent operations will be sent again or not.

For instance if you have used a `ReplaySubject` it will reemit all `onNext()` on the next subscription that may have happened after `retryWhen()`. Too less info for me to tell. Either way it's connected strictly to how you use rxJava, not the library as every operation is executed only one time.

Best Regards

aldoborrero, 2016-08-30:
Hi @DariuszAniszewski thanks for your quick answer as always!

I'm going to create then a question in StackOverflow (I'll update this comment with the link when is posted, for helping other newbies in RxJava).

Thanks a lot!!! :)

closed 2016-08-30 by aldoborrero

DariuszAniszewski, 2016-08-30:
No problem, but real credits should go to @dariuszseweryn  :)