dariuszseweryn/RxAndroidBle #66: if getDescriptor()=null ,can't setCharacteristicNotification

qqnp1100, 2016-09-01:
my device notification characteristic descriptor is null,so I can use rxandroidble to set notification. 
int rxandroidble code : 
```
    private Observable<BluetoothGattDescriptor> getClientCharacteristicConfig(BluetoothGattCharacteristic characteristic) {
        return Observable.fromCallable(() -> {
            final BluetoothGattDescriptor descriptor = characteristic.getDescriptor(CLIENT_CHARACTERISTIC_CONFIG_UUID);
            if (descriptor == null) {
                throw new BleCannotSetCharacteristicNotificationException(characteristic);
            } else {
                return descriptor;
            }
        });
    }
```
so if descriptor is null,can't setCharacteristicNotification.
the  onCharacteristicChanged() not callback.

if i just run      bluetoothGatt.setCharacteristicNotification(characteristic, true);
the onCharacteristicChanged() callback success.
Is there any way to solve it? thanks

dariuszseweryn, 2016-09-06:
Hello,
Thanks for the info - I haven't encountered a situation where there is a characteristic that supports notifications but doesn't have a Client Characteristic Config descriptor. I will try to find some more information about it.
Can you tell which device is having this issue?

qqnp1100, 2016-09-07:
The device is still in development,In my previous app，it can connect the devices and setCharacteristicNotification。set notification code:
```
    public boolean setCharacteristicNotification(BluetoothGattCharacteristic characteristic, boolean enabled) {
        boolean ok = false;
        if (mBluetoothGatt.setCharacteristicNotification(characteristic, enabled)) {
            BluetoothGattDescriptor clientConfig = characteristic.getDescriptor(BleUUID.CLIENT_CHARACTERISTIC_CONFIG);//in my device clientConfig is null
            if (clientConfig != null) {
                if (enabled) {
                    ok = clientConfig.setValue(BluetoothGattDescriptor.ENABLE_NOTIFICATION_VALUE);
                } else {
                    ok = clientConfig.setValue(BluetoothGattDescriptor.DISABLE_NOTIFICATION_VALUE);
                }

                if (ok) {
                    ok = mBluetoothGatt.writeDescriptor(clientConfig);
                }
            }
        }
        return ok;
    }
```
Although descriptor is null and void return false , but it has been run `mBluetoothGatt.setCharacteristicNotification(characteristic, enabled)`
so the onCharacteristicChanged() callback success.

but use rxandroidble,it not run anything when descriptor is null.
I want change the rxandroidble code make it not throw and continue run when descriptor is null. 
but it was too complicated so I gave up.

At the end,I got on my knees and begged My Hardware Engineer to add descriptor in this device.
now,descriptor is not null,it's work _(:3 」∠)_...

dariuszseweryn, 2016-09-07:
The standard configuration for BLE devices is that if a characteristic is supporting notifications / indications it is having a `Client Characteristic Configuration` descriptor which is the place where the client may turn on / off notifications.
A bit more info is here: http://stackoverflow.com/questions/22817005/why-does-setcharactersticnotification-not-actually-enable-notifications
And even more is in the Bluetooth Core specification: 
- `Vol 3, Part G, 3.3.3`
- `Vol 3, Part G, 3.3.3.3`
- `Vol 3, Part G, 4.10`
My personal opinion is that if there are specifications - everyone should stick to it. ;)
Best Regards

closed 2016-09-07 by dariuszseweryn

Neogene, 2017-12-02:
Having same problem with keychain bluetooth.