dariuszseweryn/RxAndroidBle #67: Strange fehavioral of  writeCharacteristic(BluetoothGattCharacteristic bluetoothGattCharacteristic, byte[] data)  method;

vadstep, 2016-09-05:
### Summary
<br />Method writeCharacteristic(BluetoothGattCharacteristic bluetoothGattCharacteristic, byte[] data); always trows an exception.(Write error: java.lang.NullPointerException: Attempt to get length of null array)
#### Preconditions
<br />My problem is very strange i can use method writeCharacteristic(UUID characteristicUuid,  byte[] data); without any problems but if i swith to different constructor that receives in a first argument BluetoothGattCharacteristic  instead UUID  i got an exception.
#### Steps to reproduce actual result
A).  works perfect -  `   connectionObservable
                    .flatMap(rxBleConnection -> rxBleConnection.writeCharacteristic(bluetoothGattCharacteristic.getUuid(), b))
                    .observeOn(AndroidSchedulers.mainThread())
                    .subscribe(bytes -> {
                        onWriteSuccess(bytes);
                    }, this::onWriteFailure);`
B). calls onWriteFailure with (Write error: java.lang.NullPointerException: Attempt to get length of null array)- `  connectionObservable
                    .flatMap(rxBleConnection -> rxBleConnection.writeCharacteristic(bluetoothGattCharacteristic, b))
                    .observeOn(AndroidSchedulers.mainThread())
                    .subscribe(bytes -> {
                        onWriteSuccess(bytes);
                    }, this::onWriteFailure);`

C). bluetoothGattCharacteristic is the same object.

dariuszseweryn, 2016-09-06:
Hello,
Can you confirm that `bluetoothGattCharacteristic` that you use is obtained during the connection on which you are trying to use it?

vadstep, 2016-09-06:
Hi. Yes sure. I am obtaining it from:

`      bleDevice.establishConnection(this, false)
                .flatMap(RxBleConnection::discoverServices)
                .first() 
                .compose(bindUntilEvent(PAUSE))
                .observeOn(AndroidSchedulers.mainThread())
                .doOnUnsubscribe(this::updateUI)
                .subscribe(adapter::swapScanResult, this::onConnectionFailure);  `

dariuszseweryn, 2016-09-06:
Tell me what will happen if you do:
```
bleDevice.establishConnection(this, false)
    .flatMap(connection -> {
        connection.discoverServices()
            .flatMap(services -> /* get your bluetoothGattCharacteristic */)
            .flatMap(bluetoothGattCharacteristic -> connection.writeCharacteristic(bluetoothGattCharacteristic, bytes))
    })
    .observeOn(AndroidSchedulers.mainThread())
    .subscribe(
        this::onWriteSuccess,
        this::onWriteFailure
    )
```

vadstep, 2016-09-06:
Looks like there is a mistake in your example:

`       bleDevice.establishConnection(this, false)
                       .flatMap(connection -> {
                         connection.discoverServices()
                             .flatMap(services -> services.getCharacteristic(uid))
                                  .flatMap(bluetoothGattCharacteristic ->                            connection.writeCharacteristic(bluetoothGattCharacteristic,
 getInputBytes155Increment()));
                })//HERE IS MISSING RETURN STATEMENT
                .observeOn(AndroidSchedulers.mainThread())
                .subscribe(
                        this::onWriteSuccess,
                        this::onWriteFailure
                );`

dariuszseweryn, 2016-09-06:
I haven't seen the whole of your code but I think you're trying to use a `bluetoothGattCharacteristic` that was obtained during some previous connection.
```
1. first connection starts
2. obtained bluetoothGattCharacteristic
3. first connection ends
4. first connection is cleaned up by the library
5. second connection starts
6. writing characteristic obtained on the first connection with the second connection
7. error
```
Lifecycle of items that are emitted from the `RxBleConnection` are bound to the lifecycle of it. They shouldn't be persisted and used across connections.

vadstep, 2016-09-06:
Ok thanks foa an answer. And GREAT thanks for a RxAndroidBle!!!

closed 2016-09-06 by dariuszseweryn