dariuszseweryn/RxAndroidBle #68: Sending data >20 bytes

nister, 2016-09-05:
### Summary
Ex: http://stackoverflow.com/questions/24135682/android-sending-data-20-bytes-by-ble

Need to send data > 20 bytes via BLE. It works normally if I using common Android BLE API. But I get only first part (20 bytes) with RxAndroidBle. Behavior is the same in case with .setupNotification, readCharacteristic, writeCharacteristic.

#### Preconditions

#### Steps to reproduce actual result
1. Send any data from BLE device > 20 bytes

#### Actual result
Only fiirst part receiving
<br />
#### Expected result
Get all data by parts
<br/>

Thanks!

zhuozhibin, 2016-09-22:
now I face the same problem, any solution?

dariuszseweryn, 2016-10-25:
Sorry for such a long wait. As for now you have to chunk the data yourself. I am currently chasing bugs mostly and unfortunately don't have enough time to research the problem and possible use cases.
Feel free to describe it to me in greater detail. :)

maoueh, 2016-11-03:
If it can help others, here the code I use to do the split. I release this to public domain without restrictions. Use (and adapt) at your own risk (note this is a slightly modified version of the class I use in my own project):

Usage:

```
bleClient.establishConnection().flatMap(connection -> {
  return new PacketWriter(connection).write(characteristic, bytes);
});
```

```
// PacketWriter.java
package co.acme.ble;

import com.google.common.annotations.VisibleForTesting;
import com.polidea.rxandroidble.RxBleConnection;

import java.util.ArrayList;
import java.util.Arrays;
import java.util.List;
import java.util.UUID;

import rx.Observable;

public class PacketWriter {

  private static final int MAX_BLE_PACKET_BYTE_COUNT = 20;

  private RxBleConnection mBleConnection;

  public PacketWriter(RxBleConnection bleConnection) {
    mBleConnection = bleConnection;
  }

  /**
   * FIXME: Make this more efficient by keeping a 20 bytes buffer that is re-used for full packet send instead of
   * copying to a new array over and over. We could even think about a specialized {@link Iterable} that would do the
   * split and caching "magically" and we would simply need to iterate over it.
   */
  public Observable<Void> write(UUID characteristic, byte[] bytes) {
    if (bytes == null || bytes.length <= 0) {
      return Observable.just(null);
    }

    return Observable.concat(computeWritePacketObservables(characteristic, bytes));
  }

  @VisibleForTesting
  int computePacketCount(byte[] bytes) {
    return (int) Math.ceil(bytes.length / (double) MAX_BLE_PACKET_BYTE_COUNT);
  }

  @VisibleForTesting
  List<Observable<Void>> computeWritePacketObservables(UUID characteristic, byte[] bytes) {
    int packetCount = computePacketCount(bytes);
    int lastPacketIndex = packetCount - 1;

    List<Observable<Void>> observables = new ArrayList<>();
    for (int i = 0; i < packetCount; ++i) {
      boolean isLastPacket = i == lastPacketIndex;

      int dataStartIndexInclusive = i * 20;
      int dataEndIndexInclusive = isLastPacket ? bytes.length : dataStartIndexInclusive + MAX_BLE_PACKET_BYTE_COUNT;

      observables.add(writeToCharacteristic(characteristic, bytes, dataStartIndexInclusive, dataEndIndexInclusive));
    }

    return observables;
  }

[... 152 lines cut ...]
    }

    public List<T> getEmittedElements() {
      return mEmittedElements;
    }

    public T getFirstElement() {
      return mEmittedElements.get(0);
    }

    public Throwable getOnErrorException() {
      return mOnErrorException;
    }

    public boolean isCompletedCalled() {
      return mIsCompletedCalled;
    }
  }
}
```

Good luck.

Regards,
Matt

pregno, 2016-12-15:
this could be managed changing the MTU

dariuszseweryn, 2016-12-15:
I have put somewhere what I do use for byte splitting but cannot find it for reference so I will put it here again:

    public class ByteSplitter extends Observable<byte[]> {
        
        protected ByteSplitter(@NonNull byte[] bytes) {
            super(new SyncOnSubscribe<ByteBuffer, byte[]>() {

                private static final int MAX_PACKET_LENGTH = 20;

                @Override
                protected ByteBuffer generateState() {
                    return ByteBuffer.wrap(bytes);
                }

                @Override
                protected ByteBuffer next(ByteBuffer state, Observer<? super byte[]> observer) {
                    final int remainingBytes = state.remaining();
                    final int bytePackageLength = Math.min(remainingBytes, MAX_PACKET_LENGTH);
                    if (bytePackageLength > 0) {
                        final byte[] bytesPacket = new byte[bytePackageLength];
                        state.get(bytesPacket);
                        observer.onNext(bytesPacket);
                    }
                    if (state.remaining() == 0) {
                        observer.onCompleted();
                    }
                    return state;
                }
            });
        }
    }

lowwor, 2016-12-16:
@dariuszseweryn #99 Here comes the reference.

dariuszseweryn, 2016-12-27:
@nister Could you elaborate more on how did you managed to send more than 20 bytes of data using the Android BLE API? The topic on stackoverflow is describing a custom BLE protocol in order to introduce some flow control in the communication. It describes nothing that the library is not supporting.

tahir9111, 2017-02-09:
Are we getting this support soon? Byte>20 transmit? How can we achieve it with current library, with bytesplitter

pregno, 2017-02-09:
If you are using Android 5.x or later you can set the MTU > 20 before sending the data

dariuszseweryn, 2017-02-09:
And if you use 4.3 / 4.4 or your peripheral does not support larger MTUs - you can just make several writes one after another with the next batches of what you want to transmit. The bytesplitter just splits your large byte array into batches that can be written at once - one after another.

I have finished a helper but it is still under a code review.

dariuszseweryn, 2017-02-10:
Functionality added with:
a2e85fc31cc2b4650e2240a1fbbb2e97d402ac12
and
b85fbf24552396bd724bb01dd15246a875322a68

closed 2017-02-10 by dariuszseweryn