dariuszseweryn/RxAndroidBle #69: Dificulty writing on characteristic

kikermo, 2016-09-10:
### Summary
When writing on a concrete characteristic within the same session, sometimes the writings are not effective. 
<br />
#### Preconditions
Create an activity passing a bluetooth device object
Let the lifecycle methods of my activity establish and release the connection with this device. In my case Start and Stop.
Create listeners that react when some input change, and then send messages to a known characteristic
<br />
#### Steps to reproduce actual result
1. Input change, triggering characteristic write using an observable connection previously established in onStart method.
2. Subscription, with logging
3. Read result with the ble device monitor.
#### Actual result
Every time input changes, it sends the message. No exception thrown. But normally the characteristic value is received in the other end, but there is a problem with the following ones. They are sent, but just some of them received. 
<br />
#### Expected result
Receive all characteristic without exception.
<br />

First of all, congratulations for this amazing library. I find it very useful and sometimes makes things like searching for devices or discovering services extremely easy. I am rising this as an issue, but it is very likely I am doing something wrong.

I also have to say that I tried with android vanilla ble api, and there was no problem writing on the same characteristic.

Thanks in advance.

dariuszseweryn, 2016-09-10:
Hello,
To be able to help you I would need to see the logs from the device (or both if possible) and the code you are using.
I am also thinking if this question should not go to www.stackoverflow.com with `rxandroidble` tag

kikermo, 2016-09-12:
Hi, 
You are absolutely right, I should have created a thread in stackoverflow. I didn't do it I guess because it was a question related to an specific library outside the android framework, but now I've seen there are some question targeting concrete third party libraries, so I will do that way next time. 
Anyway, I have solved the problem. I didn't know until I saw it in the sample app, that you needed to create a ConnectionSharingAdapter for multiple writings within a connection. Before that, I was opening and closing the connection every time I was writing on a characteristic, but I was having problems with that. And before that I was using a single connection as described when I started this thread, but it was crashing, logging that there was a connection already established.
I also have to say that the description of the issue is not accurate as in my mind I mixed up these two cases, having changed the code several times. 
In my humble opinion, there should be an example in the main page for multiple writings using ConnectionSharingAdapter, in order to make it a bit more obvious. 

Thanks a lot.

closed 2016-09-12 by kikermo

dariuszseweryn, 2016-09-13:
Glad to hear that you have resolved your issue. Thanks for the suggestion. I will probably also add a note that knowledge of `rxJava` is advised.