dariuszseweryn/RxAndroidBle #71: Can't read rssi from RxBleConecction

jkwiecien, 2016-09-16:
### Summary
I'd like to read rssi from ```RxBleConnection```
#### Preconditions
I connect to the device using ```establishConnection()```
#### Steps to reproduce actual result
The code for reading rssi:
```
                repository.readRssi(model.getConnection())
                        .delay(1, TimeUnit.SECONDS)
                        .subscribe(new Subscriber<Integer>() {
                            @Override
                            public void onCompleted() {

                            }

                            @Override
                            public void onError(Throwable e) {
                                e.printStackTrace();
                            }

                            @Override
                            public void onNext(Integer rssi) {
                                Log.w(LogTags.BLE, "device rssi: " + rssi);
                            }
                        });
```
#### Actual result
```onNext``` never happens but the rssi is visible in logs: ```D/RxBle#BluetoothGatt: onReadRemoteRssi rssi=-41 status=0```
#### Expected result
Expected ```onNex```t to return rssi
<br />

dariuszseweryn, 2016-09-27:
Hello there.
First of all, I think that this is a question for www.stackoverflow.com with the tag `rxandroidble`.
Secondly - it's hard to reason about the problem if we have no info about what does `repository.readRssi()` do or what comes from `model.getConnection()`
Best Regards

jkwiecien, 2016-09-27:
I updated the issue with the missing methods.

dariuszseweryn, 2016-09-27:
Could you also show `model.getConnection()` body?

jkwiecien, 2016-09-27:
It's just a getter returning RxBleConnection instance which was previously gotten from ```rxbledevice.connect();```

dariuszseweryn, 2016-09-27:
I have just checked the minimal code test with:
```
       rxBleClient.getBleDevice(macAddress)
                .establishConnection(this, false)
                .flatMap(RxBleConnection::readRssi)
                .delay(1, TimeUnit.SECONDS)
                .take(1)
                .subscribe(
                        rssi -> Log.d("XXX", "RSSI: " + rssi),
                        error -> Log.e("XXX", "Error", error)
                );
```
And got result:
```
09-27 16:15:06.757 17362-17362/com.polidea.rxandroidble.sample I/dalvikvm: Could not find method android.bluetooth.BluetoothGattCallback.onMtuChanged, referenced from method com.polidea.rxandroidble.internal.connection.RxBleGattCallback$1.onMtuChanged
09-27 16:15:06.757 17362-17362/com.polidea.rxandroidble.sample W/dalvikvm: VFY: unable to resolve virtual method 286: Landroid/bluetooth/BluetoothGattCallback;.onMtuChanged (Landroid/bluetooth/BluetoothGatt;II)V
09-27 16:15:06.757 17362-17362/com.polidea.rxandroidble.sample D/dalvikvm: VFY: replacing opcode 0x6f at 0x0016
09-27 16:15:06.774 17362-17362/com.polidea.rxandroidble.sample D/RxBle#Radio:   QUEUED RxBleRadioOperationConnect(1107995200)
09-27 16:15:06.777 17362-17376/com.polidea.rxandroidble.sample D/RxBle#Radio:  STARTED RxBleRadioOperationConnect(1107995200)
09-27 16:15:06.789 17362-17362/com.polidea.rxandroidble.sample D/BluetoothGatt: connect() - device: D0:5F:B8:3C:7C:7E, auto: false
09-27 16:15:06.789 17362-17362/com.polidea.rxandroidble.sample D/BluetoothGatt: registerApp()
09-27 16:15:06.848 17362-17362/com.polidea.rxandroidble.sample D/BluetoothGatt: registerApp() - UUID=4f63adcb-073d-4a7d-b407-fa0855cf56b3
09-27 16:15:06.851 17362-17375/com.polidea.rxandroidble.sample D/BluetoothGatt: onClientRegistered() - status=0 clientIf=6
09-27 16:15:34.650 17362-17373/com.polidea.rxandroidble.sample D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=6 device=D0:5F:B8:3C:7C:7E
09-27 16:15:34.651 17362-17373/com.polidea.rxandroidble.sample D/RxBle#BluetoothGatt: onConnectionStateChange newState=2 status=0
09-27 16:15:34.678 17362-18152/com.polidea.rxandroidble.sample D/RxBle#Radio:   QUEUED RxBleRadioOperationReadRssi(1108421000)
09-27 16:15:34.681 17362-18152/com.polidea.rxandroidble.sample D/RxBle#RadioOperationConnect$$Lambda: No subscribers, finishing operation
09-27 16:15:34.683 17362-17376/com.polidea.rxandroidble.sample D/RxBle#Radio: FINISHED RxBleRadioOperationConnect(1107995200)
09-27 16:15:34.683 17362-17376/com.polidea.rxandroidble.sample D/RxBle#Radio:  STARTED RxBleRadioOperationReadRssi(1108421000)
09-27 16:15:34.685 17362-17362/com.polidea.rxandroidble.sample D/BluetoothGatt: readRssi() - device: D0:5F:B8:3C:7C:7E
09-27 16:15:34.689 17362-17375/com.polidea.rxandroidble.sample D/BluetoothGatt: onReadRemoteRssi() - Device=D0:5F:B8:3C:7C:7E rssi=-77 status=0
09-27 16:15:34.690 17362-17375/com.polidea.rxandroidble.sample D/RxBle#BluetoothGatt: onReadRemoteRssi rssi=-77 status=0
09-27 16:15:34.692 17362-17376/com.polidea.rxandroidble.sample D/RxBle#Radio: FINISHED RxBleRadioOperationReadRssi(1108421000)
09-27 16:15:35.692 17362-18153/com.polidea.rxandroidble.sample D/XXX: RSSI: -77
09-27 16:15:35.697 17362-18153/com.polidea.rxandroidble.sample D/RxBle#Radio:   QUEUED RxBleRadioOperationDisconnect(1108010224)
09-27 16:15:35.700 17362-17376/com.polidea.rxandroidble.sample D/RxBle#Radio:  STARTED RxBleRadioOperationDisconnect(1108010224)
09-27 16:15:35.705 17362-17362/com.polidea.rxandroidble.sample D/BluetoothManager: getConnectionState()
09-27 16:15:35.706 17362-17362/com.polidea.rxandroidble.sample D/BluetoothManager: getConnectedDevices
09-27 16:15:35.721 17362-17362/com.polidea.rxandroidble.sample D/BluetoothGatt: cancelOpen() - device: D0:5F:B8:3C:7C:7E
09-27 16:15:35.723 17362-17373/com.polidea.rxandroidble.sample D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=6 device=D0:5F:B8:3C:7C:7E
09-27 16:15:35.724 17362-17373/com.polidea.rxandroidble.sample D/RxBle#BluetoothGatt: onConnectionStateChange newState=0 status=0
09-27 16:15:45.723 17362-17362/com.polidea.rxandroidble.sample D/BluetoothGatt: close()
09-27 16:15:45.725 17362-17362/com.polidea.rxandroidble.sample D/BluetoothGatt: unregisterApp() - mClientIf=6
09-27 16:15:45.737 17362-17376/com.polidea.rxandroidble.sample D/RxBle#Radio: FINISHED RxBleRadioOperationDisconnect(1108010224)
```
Which seems to be ok. I need more information about how you use the library. Are you familiar with rxJava? Maybe you're unsubscribing before the RSSI is read?

jkwiecien, 2016-09-27:
Started working after adding take(1)

dariuszseweryn, 2016-09-28:
`.take(1)` is there only for unsubscribing from the `.establishConnection()` observable effectively closing the connection. Without it the output looks like:
```
09-28 11:13:33.038 26924-26924/com.polidea.rxandroidble.sample D/RxBle#Radio:   QUEUED RxBleRadioOperationConnect(1107949800)
09-28 11:13:33.038 26924-26944/com.polidea.rxandroidble.sample D/RxBle#Radio:  STARTED RxBleRadioOperationConnect(1107949800)
09-28 11:13:33.054 26924-26924/com.polidea.rxandroidble.sample D/BluetoothGatt: connect() - device: D0:5F:B8:3C:7C:7E, auto: false
09-28 11:13:33.054 26924-26924/com.polidea.rxandroidble.sample D/BluetoothGatt: registerApp()
09-28 11:13:33.112 26924-26924/com.polidea.rxandroidble.sample D/BluetoothGatt: registerApp() - UUID=78705c47-7f83-46be-b7d7-639349c05806
09-28 11:13:33.115 26924-26935/com.polidea.rxandroidble.sample D/BluetoothGatt: onClientRegistered() - status=0 clientIf=6
09-28 11:13:35.022 26924-26936/com.polidea.rxandroidble.sample D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=6 device=D0:5F:B8:3C:7C:7E
09-28 11:13:35.023 26924-26936/com.polidea.rxandroidble.sample D/RxBle#BluetoothGatt: onConnectionStateChange newState=2 status=0
09-28 11:13:35.031 26924-27199/com.polidea.rxandroidble.sample D/RxBle#Radio:   QUEUED RxBleRadioOperationReadRssi(1108362048)
09-28 11:13:35.035 26924-27199/com.polidea.rxandroidble.sample D/RxBle#RadioOperationConnect$$Lambda: No subscribers, finishing operation
09-28 11:13:35.037 26924-26944/com.polidea.rxandroidble.sample D/RxBle#Radio: FINISHED RxBleRadioOperationConnect(1107949800)
09-28 11:13:35.039 26924-26944/com.polidea.rxandroidble.sample D/RxBle#Radio:  STARTED RxBleRadioOperationReadRssi(1108362048)
09-28 11:13:35.045 26924-26924/com.polidea.rxandroidble.sample D/BluetoothGatt: readRssi() - device: D0:5F:B8:3C:7C:7E
09-28 11:13:35.055 26924-26935/com.polidea.rxandroidble.sample D/BluetoothGatt: onReadRemoteRssi() - Device=D0:5F:B8:3C:7C:7E rssi=-79 status=0
09-28 11:13:35.056 26924-26935/com.polidea.rxandroidble.sample D/RxBle#BluetoothGatt: onReadRemoteRssi rssi=-79 status=0
09-28 11:13:35.058 26924-26944/com.polidea.rxandroidble.sample D/RxBle#Radio: FINISHED RxBleRadioOperationReadRssi(1108362048)
09-28 11:13:36.058 26924-27200/com.polidea.rxandroidble.sample D/XXX: RSSI: -79
```
The flow still works. There must be a different problem.

closed 2016-09-30 by dariuszseweryn