dariuszseweryn/RxAndroidBle #72: Any possibility for backing with RxJava 2?

pauldemarco, 2016-09-20:
### Summary
I have an entire project built with RxJava 2 and I'd like to now use RxAndroidBle.
<br />
#### Preconditions
Adding RxAndroidBle will also include RxJava 1 dependencies. Would this cause a larger apk?
<br />
#### Steps to reproduce actual result
1. Write an app using RxJava 2
2. Add RxAndroidBle functionality
#### Actual result
RxJava 1 is now added as a dependency
<br />
#### Expected result
RxAndroidBle should utilize RxJava2
<br />

uKL, 2016-09-21:
Hey!

We are internally investigating how to align RxAndroidBLE with RxJava 2. We will update you as soon as it is feasible.

trulsmag, 2016-11-22:
I think it is possible to create a wrapper around RxAndroidBLE using [RxJava2Interop](https://github.com/akarnokd/RxJava2Interop) until RxJava2 is supported by RxAndroidBLE.

dariuszseweryn, 2016-11-22:
Of course it is possible this way but it is the only way currently possible. I am thinking about a more flexible approach to the library to use a mechanism similar to retrofit2 adapters. But it would take a lot of work and I don't know yet when I will have time for that.

nateridderman, 2017-02-08:
It seems that one problem with an adapter approach is that you still need to use one or the other under the hood. Then if a developer uses the other one, the total method count for rxjava 1 and 2 and rxandroidble is ~16-17k.

dariuszseweryn, 2017-02-08:
Not necessarily - but we would need to switch the current RxJava1 implementation to some custom code.

HolySamosa, 2017-03-14:
With RxJava 1's upcoming feature freeze in June, and then end-of-life in March 2018, perhaps just move  to RxJava 2?

RobLewis, 2017-05-17:
I read that Java 9 has a new Reactive ["Flow" paradigm](https://community.oracle.com/docs/DOC-1006738), with classes Publisher, Subscriber, Processor, and Subscription. A Publisher seems to be equivalent to an Observable, and a Processor is an Operator (or perhaps a Subject). I wonder how this will impact the RxJava project.

dariuszseweryn, 2017-05-17:
There will be no impact yet for quite some time I am afraid. Remember that we are stuck with Java 7 still with no views for quick update.

PopTudor, 2017-05-19:
Kotlin is now a first class language for Android development as of yesterday's Google IO. They do port most of java 8 features but some will still be missing. As of Java 9, I think it won't come to Android anytime soon. Bear in mind that Java 8 has many more significant features than Java 9 and it took about 3 years to arrive and it's still not 100% backwards compatible. My guess is they will slowly drop support for it as the community will move on to Kotlin. But that's going to take some time, 2-5 years at most

vpratfr, 2017-07-20:
Quite interested in this RxJava2 support too. We are using Kotlin as well here. Java 9 will never see Android ever (Kotlin came in because Java 8 never really got supported).

RobLewis, 2017-09-27:
Is there any tentative schedule for RxJava 2 support? I don't have an active project right now, but for my next one I'd like use v. 2 if possible.

HolySamosa, 2017-09-27:
@RobLewis In the interim, I've been successfully using **RxJava2Interop** to integrate into an RxJava2 project.

dariuszseweryn, 2017-09-28:
Current plan is to introduce the new API in `1.5.0` which will be almost 1:1 in `2.0.0`. I would like to test the new API before bumping the major version as it is the only moment when we are able to cleanup the deprecated APIs.

I am a bit short of time now so I cannot really say when the new API will be ready.

For now you can use [RxJava2Interop](https://github.com/akarnokd/RxJava2Interop) as @HolySamosa mentioned.

patrykpoborca, 2017-10-04:
👍 to rx2

onivas, 2017-11-08:
any update here?

micHar, 2017-11-29:
Here's another soul craving for rxjava2 support

uKL, 2017-11-29:
Actually, a good news is that for a few days this is already in progress in my local branch! Please stay tuned :)

GuillaumeBo, 2018-01-05:
Any news about your progress @uKL ? Will be available with 1.5 ? :)

uKL, 2018-01-05:
We are targeting with Rx java 2 for a 2.0.0 as it is a breaking change. Unfortunately, I stopped the development for a few weeks but I plan to get back to it next week.

markotron, 2018-01-05:
In the meantime, if you're using RxJava2Intreop check this issue: https://github.com/akarnokd/RxJava2Interop/issues/21

Deemonser, 2018-01-16:
Here's another soul craving for rxjava2 support        +1

dariuszseweryn, 2018-02-16:
[image]
We are finishing the migration to the `RxJava 2`. There is a [snapshot](https://github.com/Polidea/RxAndroidBle#snapshot) release with the name `com.polidea.rxandroidble2:rxandroidble:1.5.0-SNAPSHOT` which you can try out!

RobLewis, 2018-02-17:
Sorry, I have never understood how to use these snapshots, and the
Sonatype site isn't much help IMO. Can you point me to a step-by-step?

Here's what I'm trying:

In the `repositories` block of the project build.gradle:
        maven { url
"https://oss.sonatype.org/content/repositories/snapshots" }

In the `dependencies` block of the module build.gradle:
    compile 'com.polidea.rxandroidble2:rxandroidble:1.5.0-SNAPSHOT'

The error is: "Failed to resolve:
com.polidea.rxandroidble2:rxandroidble:1.5.0-SNAPSHOT"

--Rob

> Dariusz Seweryn <mailto:notifications@github.com>
> February 16, 2018 at 09:37
> [image]
> We are finishing the migration to the `RxJava 2`. There is a
> [snapshot](https://github.com/Polidea/RxAndroidBle#snapshot) release
> with the name `com.polidea.rxandroidble2:rxandroidble:1.5.0-SNAPSHOT`
> which you can try out!
>

dariuszseweryn, 2018-02-17:
I have just tested it and seems to work for me. In the projects `build.gradle` I have:
```
allprojects {
    repositories {
        google()
        jcenter()
        maven { url "https://oss.sonatype.org/content/repositories/snapshots" }
    }
}
```
and in the module `build.gradle`:
```
dependencies {
    // ...
    implementation 'com.polidea.rxandroidble2:rxandroidble:1.5.0-SNAPSHOT'
    // ...
}
```

RobLewis, 2018-02-17:
Ah, I was putting the maven URL in the buildscript:repositories block,
not the allprojects:repositories one. Thanks.

Sadly, Gradle remains a mystery to me. I really must learn more about it.

--RL

P.S. I know you have plenty of work to do, but was wondering if you
might be planning a migration guide for moving RxAndroidBle to RxJava2.

> Dariusz Seweryn <mailto:notifications@github.com>
> February 16, 2018 at 23:58
> I have just tested it and seems to work for me. In the projects
> `build.gradle` I have:
> ```
> allprojects {
> repositories {
> google()
> jcenter()
> maven { url "https://oss.sonatype.org/content/repositories/snapshots" }
> }
> }
> ```
> and in the module `build.gradle`:
> ```
> dependencies {
> // ...
> implementation 'com.polidea.rxandroidble2:rxandroidble:1.5.0-SNAPSHOT'
> // ...
> }
> ```
>

dariuszseweryn, 2018-02-17:
Migration guide is not planned as for now. There is no functional difference in terms of the library API as far as I know and there are already plenty of [migration guides for RxJava 1 -> 2](https://www.google.pl/search?q=rxjava+1+to+2+migration) which can be checked out. There is no need to invent the wheel again

uKL, 2018-03-15:
1.5.0 is on maven with rxjava 2 support.

closed 2018-03-15 by uKL