dariuszseweryn/RxAndroidBle #75: Out-of-order emissions from RxBleGattCallback

trulsmag, 2016-09-27:
### Summary
Items emitted from RxBleGattCallback are received out-of-order.

E.g. if one uses notifications to transmit multiple BLE packets in sequence using a single characteristic (emulating a UART or similar) onCharacteristicChanged will schedule the call to changedCharacteristicPublishSubject::onNext on any thread from Schedulers.computation(), which often results in the ByteAssociation<UUID>'s emitted from changedCharacteristicPublishSubject being received out-of-order.

#### Steps to reproduce actual result
1. Send data divided into multiple packets.
2. Reassemble packets into data.
3. Compare with data sent.

#### Actual result
Often data is received out-of-order on packet boundaries resulting in invalid data.

##### Sometimes OK
`
RxBle#MyClient(4262): android_2
RxBle#Connection(4262): android_2
RxBle#Connection$$Lambda(4262): android_2
RxBle#Connection(4262): android_2
BluetoothGatt(4262): setCharacteristicNotification() - uuid: 6e400003-b5a3-f393-e0a9-e50e24dcca9e enable: true
RxBle#Radio$$Lambda(4262): onsubscribe: queue android_2
RxBle#Radio(4262):   QUEUED RxBleRadioOperationDescriptorWrite(615526424)
RxBle#Radio(4262):  STARTED RxBleRadioOperationDescriptorWrite(615526424)
RxBle#BluetoothGatt(4262): onDescriptorWrite descriptor=00002902-0000-1000-8000-00805f9b34fb status=0
MyClientImpl(4262): tx: blePacket =  [0, 1, 2, 3, 4, 5, 6]
RxBle#WriteWrapper(4262): rxble_3
RxBle#Connection(4262): rxble_3
RxBle#Connection(4262): rxble_3
RxBle#Radio$$Lambda(4262): onsubscribe: queue rxble_3
RxBle#Radio(4262):   QUEUED RxBleRadioOperationCharacteristicWrite(779151920)
RxBle#Radio(4262): FINISHED RxBleRadioOperationDescriptorWrite(615526424)
RxBle#Radio(4262):  STARTED RxBleRadioOperationCharacteristicWrite(779151920)
RxBle#BluetoothGatt(4262): onCharacteristicWrite characteristic=6e400002-b5a3-f393-e0a9-e50e24dcca9e status=0 Binder_1
RxBle#GattCallback(4262): just Binder_1 value = [0, 1, 2, 3, 4, 5, 6]
RxBle#GattCallback(4262): associateCharacteristicWithBytes Binder_1
RxBle#GattCallback$$Lambda(4262): justdef rxble_4 ByteAssociation[first=BluetoothGattCharacteristic(6e400002-b5a3-f393-e0a9-e50e24dcca9e), second=[0, 1, 2, 3, 4, 5, 6]]
RxBle#Radio(4262): FINISHED RxBleRadioOperationCharacteristicWrite(779151920)
RxBle#BluetoothGatt(4262): onCharacteristicChanged characteristic=6e400003-b5a3-f393-e0a9-e50e24dcca9e Binder_1
RxBle#GattCallback(4262): just Binder_1 value = [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19]
RxBle#GattCallback(4262): associateCharacteristicWithBytes Binder_1
RxBle#BluetoothGatt(4262): onCharacteristicChanged characteristic=6e400003-b5a3-f393-e0a9-e50e24dcca9e Binder_1
RxBle#GattCallback(4262): just Binder_1 value = [20, 21]
RxBle#GattCallback(4262): associateCharacteristicWithBytes Binder_1
RxBle#GattCallback$$Lambda(4262): justdef rxble_0 ByteAssociation[first=BluetoothGattCharacteristic(6e400003-b5a3-f393-e0a9-e50e24dcca9e), second=[0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19]]
RxBle#GattCallback$$Lambda(4262): justdef rxble_3 ByteAssociation[first=BluetoothGattCharacteristic(6e400003-b5a3-f393-e0a9-e50e24dcca9e), second=[20, 21]]
RxBle#GattCallback$1$$Lambda(4262): beforechanged: ByteAssociation[first=UUID(6e400003-b5a3-f393-e0a9-e50e24dcca9e), second=[0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19]] rxble_0
OperatorBites2Bite(4262):  : [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19] rxble_0
RxBle#GattCallback$1$$Lambda(4262): afterchanged: ByteAssociation[first=UUID(6e400003-b5a3-f393-e0a9-e50e24dcca9e), second=[0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19]] rxble_0
RxBle#GattCallback$1$$Lambda(4262): beforechanged: ByteAssociation[first=UUID(6e400003-b5a3-f393-e0a9-e50e24dcca9e), second=[20, 21]] rxble_3
OperatorBites2Bite(4262):  : [20, 21] rxble_3
BleConnectionManagerImpl(4262): Frodo => [@Observable#connection -> onUnsubscribe() :: @ObserveOn -> rxble_3]
BluetoothGatt(4262): setCharacteristicNotification() - uuid: 6e400003-b5a3-f393-e0a9-e50e24dcca9e enable: false
RxBle#Radio$$Lambda(4262): onsubscribe: queue rxble_3
RxBle#Radio(4262):   QUEUED RxBleRadioOperationDescriptorWrite(595407250)
RxBle#Radio(4262):  STARTED RxBleRadioOperationDescriptorWrite(595407250)
RxBle#GattCallback$1$$Lambda(4262): afterchanged: ByteAssociation[first=UUID(6e400003-b5a3-f393-e0a9-e50e24dcca9e), second=[20, 21]] rxble_3
`

##### Sometimes not OK
`
RxBle#MyClient(4262): android_1
RxBle#Connection(4262): android_1
RxBle#Connection$$Lambda(4262): android_1
RxBle#Connection(4262): android_1
BluetoothGatt(4262): setCharacteristicNotification() - uuid: 6e400003-b5a3-f393-e0a9-e50e24dcca9e enable: true
RxBle#Radio$$Lambda(4262): onsubscribe: queue android_1
RxBle#Radio(4262):   QUEUED RxBleRadioOperationDescriptorWrite(973242039)
RxBle#Radio(4262):  STARTED RxBleRadioOperationDescriptorWrite(973242039)
RxBle#BluetoothGatt(4262): onDescriptorWrite descriptor=00002902-0000-1000-8000-00805f9b34fb status=0
MyClientImpl(4262): tx: blePacket =  [0, 1, 2, 3, 4, 5, 6]
[... 87 lines cut ...]
RxBle#GattCallback(6230): associateCharacteristicWithBytes Binder_1
RxBle#GattCallback$$Lambda(6230): justdef rxble_4 ByteAssociation[first=BluetoothGattCharacteristic(6e400002-b5a3-f393-e0a9-e50e24dcca9e), second=[0, 1, 2, 3, 4, 5, 6]]
RxBle#Radio(6230): FINISHED RxBleRadioOperationCharacteristicWrite(540174070)
RxBle#BluetoothGatt(6230): onCharacteristicChanged characteristic=6e400003-b5a3-f393-e0a9-e50e24dcca9e Binder_2
RxBle#GattCallback(6230): just Binder_2 value = [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19]
RxBle#GattCallback(6230): associateCharacteristicWithBytes Binder_2
RxBle#GattCallback$$Lambda(6230): justdef Binder_2 ByteAssociation[first=BluetoothGattCharacteristic(6e400003-b5a3-f393-e0a9-e50e24dcca9e), second=[0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19]]
RxBle#GattCallback$1$$Lambda(6230): beforechanged: ByteAssociation[first=UUID(6e400003-b5a3-f393-e0a9-e50e24dcca9e), second=[0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19]] Binder_2
OperatorBites2Bite(6230):  : [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19] Binder_2
RxBle#GattCallback$1$$Lambda(6230): afterchanged: ByteAssociation[first=UUID(6e400003-b5a3-f393-e0a9-e50e24dcca9e), second=[0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19]] Binder_2
RxBle#BluetoothGatt(6230): onCharacteristicChanged characteristic=6e400003-b5a3-f393-e0a9-e50e24dcca9e Binder_2
RxBle#GattCallback(6230): just Binder_2 value = [20, 21]
RxBle#GattCallback(6230): associateCharacteristicWithBytes Binder_2
RxBle#GattCallback$$Lambda(6230): justdef Binder_2 ByteAssociation[first=BluetoothGattCharacteristic(6e400003-b5a3-f393-e0a9-e50e24dcca9e), second=[20, 21]]
RxBle#GattCallback$1$$Lambda(6230): beforechanged: ByteAssociation[first=UUID(6e400003-b5a3-f393-e0a9-e50e24dcca9e), second=[20, 21]] Binder_2
OperatorBites2Bite(6230):  : [20, 21] Binder_2
BleConnectionManagerImpl(6230): Frodo => [@Observable#connection -> onUnsubscribe() :: @ObserveOn -> Binder_2]
BluetoothGatt(6230): setCharacteristicNotification() - uuid: 6e400003-b5a3-f393-e0a9-e50e24dcca9e enable: false
RxBle#Radio$$Lambda(6230): onsubscribe: queue Binder_2
RxBle#Radio(6230):   QUEUED RxBleRadioOperationDescriptorWrite(1009668591)
RxBle#Radio(6230):  STARTED RxBleRadioOperationDescriptorWrite(1009668591)
RxBle#GattCallback$1$$Lambda(6230): afterchanged: ByteAssociation[first=UUID(6e400003-b5a3-f393-e0a9-e50e24dcca9e), second=[20, 21]] Binder_2
RxBle#BluetoothGatt(6230): onDescriptorWrite descriptor=00002902-0000-1000-8000-00805f9b34fb status=0
RxBle#Radio(6230): FINISHED RxBleRadioOperationDescriptorWrite(1009668591)
`

dariuszseweryn, 2016-10-25:
Just pushed a change to the master branch. Now all `BluetoothGattCallback` invocations will happen on a single specific thread. Should be available shortly in SNAPSHOT release.

zhangyud, 2016-12-08:
I have same situation,  and I use 1.1.0

dariuszseweryn, 2016-12-08:
@zhangyud show me the logs please and your code - it will be easier to reason about the issue

zhangyud, 2016-12-12:
basically, same with origin code:
my code:

connection.setupNotification(serial_char)
.subscribe(new Subscriber<Observable<byte[]>>() {
	onCompleted(){
	}
	onError(Throwable){
	}
	onNext(byte[] result){
		Log.d(TAG, "result:%d", result[0]);
	}
});

device serial_char notify seq:
time: char: content
0: serial_char, 0
20: serial_char, 1
40: serial_char, 2
60: serial_char, 3

mobile received serial_char notify seq:
char: content
serial_char, 1
serial_char, 0
serial_char, 2
serial_char, 3

when I change RxBleGattCallback
@NonNull
    private Observable<ByteAssociation<BluetoothGattCharacteristic>> just(BluetoothGattCharacteristic bluetoothGattCharacteristic) {
        final byte[] value = bluetoothGattCharacteristic.getValue();
        //this is my changed code
        return Observable.just(ByteAssociation.create(bluetoothGattCharacteristic, value));
//        this is the origin code
//        return Observable.defer(() -> Observable.just(ByteAssociation.create(bluetoothGattCharacteristic, value)));
    }

It work always fine.

zhangyud, 2016-12-12:
this is the real device log:

12-12 12:16:08.046 18733-18756/com.zmlock.zimalock D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=0000aaa5-0000-1000-8000-00805f9b34fb status=0
12-12 12:16:08.046 18733-18756/com.zmlock.zimalock D/RxBle#GattCallback: just uuid:0000aaa5-0000-1000-8000-00805f9b34fb content:[-115, -96, 30, 28, 125, 17, -54, 95, 63, -90, 22, -121, -60, 45, 24, -98]
12-12 12:16:08.051 18733-18761/com.zmlock.zimalock D/RxBle#Radio: FINISHED RxBleRadioOperationCharacteristicWrite(65237909)
12-12 12:16:08.106 18733-18806/com.zmlock.zimalock D/RxBle#BluetoothGatt: onCharacteristicChanged characteristic=0000aaa5-0000-1000-8000-00805f9b34fb
12-12 12:16:08.106 18733-18806/com.zmlock.zimalock D/RxBle#GattCallback: just uuid:0000aaa5-0000-1000-8000-00805f9b34fb content:[33, -17, -74, 65, -15, 7, 6, 83, 113, -70, 24, 83, -17, 5, 126, -4]
12-12 12:16:08.106 18733-18806/com.zmlock.zimalock D/RxBle#BluetoothGatt: onCharacteristicChanged characteristic=0000aaa5-0000-1000-8000-00805f9b34fb
12-12 12:16:08.111 18733-18806/com.zmlock.zimalock D/RxBle#GattCallback: just uuid:0000aaa5-0000-1000-8000-00805f9b34fb content:[95, -111, -103, -92, -36, 50, -37, 42, -35, -98, -7, -18, 15, 6, -74, -37]
12-12 12:16:08.111 18733-18794/com.zmlock.zimalock W/ProtocolQueue: content: 5F9199A4DC32DB2ADD9EF9EE0F06B6DB

zhangyud, 2016-12-12:
I Suggest use this style:

```.java
scheduler.createWorker().schedule(new Action0() {
            @Override
            public void call() {
                //some code work in specificed scheduler
                //and run it in order
            }
        })
```

dariuszseweryn, 2016-12-12:
Could you try changing this line:
```
private final PublishSubject<ByteAssociation<UUID>> changedCharacteristicPublishSubject = PublishSubject.create();
```
to this:
```
private final SerializedSubject<ByteAssociation<UUID>, ByteAssociation<UUID>> changedCharacteristicPublishSubject = PublishSubject.<ByteAssociation<UUID>>create().toSerialized();
```

zhangyud, 2016-12-12:
I think the problem is that defer is not call in order

zhangyud, 2016-12-13:
I test this code, but problem is still there
```
private final SerializedSubject<ByteAssociation<UUID>, ByteAssociation<UUID>> changedCharacteristicPublishSubject = PublishSubject.<ByteAssociation<UUID>>create().toSerialized();
```

log:
```
12-13 10:33:06.116 19188-19493/com.zmlock.zimalock D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicWrite(759691432)
12-13 10:33:06.116 19188-19235/com.zmlock.zimalock D/RxBle#Radio:  STARTED RxBleRadioOperationCharacteristicWrite(759691432)
12-13 10:33:06.126 19188-20104/com.zmlock.zimalock D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=0000aaa5-0000-1000-8000-00805f9b34fb status=0
12-13 10:33:06.126 19188-20104/com.zmlock.zimalock D/RxBle#GattCallback: serialized just uuid:0000aaa5-0000-1000-8000-00805f9b34fb content:[-104, 88, 108, 96, 41, 47, -113, -88, 124, -4, -11, 76, 90, -77, 100, -22]
12-13 10:33:06.126 19188-19235/com.zmlock.zimalock D/RxBle#Radio: FINISHED RxBleRadioOperationCharacteristicWrite(759691432)
12-13 10:33:06.201 19188-20104/com.zmlock.zimalock D/RxBle#BluetoothGatt: onCharacteristicChanged characteristic=0000aaa5-0000-1000-8000-00805f9b34fb
12-13 10:33:06.201 19188-20104/com.zmlock.zimalock D/RxBle#GattCallback: serialized just uuid:0000aaa5-0000-1000-8000-00805f9b34fb content:[-2, 51, 110, 68, -100, 52, -22, -82, -107, -98, 16, 41, -54, 80, 82, -24]
12-13 10:33:06.206 19188-20104/com.zmlock.zimalock D/RxBle#BluetoothGatt: onCharacteristicChanged characteristic=0000aaa5-0000-1000-8000-00805f9b34fb
12-13 10:33:06.206 19188-20104/com.zmlock.zimalock D/RxBle#GattCallback: serialized just uuid:0000aaa5-0000-1000-8000-00805f9b34fb content:[58, 109, 43, 77, -40, 14, -86, 93, -96, -51, 48, 15, -75, 90, 75, 105]
12-13 10:33:06.211 19188-19494/com.zmlock.zimalock I/ProtocolQueue: send completed package: 068508000138160000005B9BC9F022F3C68890A543EBF5CC2B06164AEA6E0000 cmd:0008
12-13 10:33:06.211 19188-20109/com.zmlock.zimalock D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicWrite(478862572)
12-13 10:33:06.211 19188-19235/com.zmlock.zimalock D/RxBle#Radio:  STARTED RxBleRadioOperationCharacteristicWrite(478862572)
12-13 10:33:06.221 19188-19199/com.zmlock.zimalock D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=0000aaa5-0000-1000-8000-00805f9b34fb status=0
12-13 10:33:06.226 19188-19199/com.zmlock.zimalock D/RxBle#GattCallback: serialized just uuid:0000aaa5-0000-1000-8000-00805f9b34fb content:[49, 118, 89, -117, 65, -33, -37, 19, 36, -94, -120, -44, -124, -22, -34, -116]
12-13 10:33:06.226 19188-19235/com.zmlock.zimalock D/RxBle#Radio: FINISHED RxBleRadioOperationCharacteristicWrite(478862572)
12-13 10:33:06.301 19188-19202/com.zmlock.zimalock D/RxBle#BluetoothGatt: onCharacteristicChanged characteristic=0000aaa5-0000-1000-8000-00805f9b34fb
12-13 10:33:06.301 19188-19202/com.zmlock.zimalock D/RxBle#GattCallback: serialized just uuid:0000aaa5-0000-1000-8000-00805f9b34fb content:[-43, 16, 62, 127, 111, -75, 23, -28, 126, 120, -10, -121, 69, -6, -26, 118]
12-13 10:33:06.301 19188-19202/com.zmlock.zimalock D/RxBle#BluetoothGatt: onCharacteristicChanged characteristic=0000aaa5-0000-1000-8000-00805f9b34fb
12-13 10:33:06.301 19188-19202/com.zmlock.zimalock D/RxBle#GattCallback: serialized just uuid:0000aaa5-0000-1000-8000-00805f9b34fb content:[-73, 46, -123, -48, 124, 98, 4, -70, -26, -67, -62, 0, -105, 0, -97, -107]
12-13 10:33:06.301 19188-19493/com.zmlock.zimalock W/ProtocolQueue: content: B72E85D07C6204BAE6BDC20097009F95
```

zhangyud, 2016-12-15:
execute on specified scheduler
```
Observable.just(1, 2, 3).map(new Func1<Integer, Integer>() {
            @Override
            public Integer call(Integer integer) {
                Thread cur = Thread.currentThread();
                System.out.printf("map:%d:%s:%d\n", cur.getId(), cur.getName(), integer);
                return integer;
            }
        }).subscribeOn(Schedulers.io()).observeOn(Schedulers.newThread()).subscribe();
```

dariuszseweryn, 2017-01-30:
This issue should be now fixed - or so I hope. `SerializedSubject` and no deferring. Please check how does it work for you - if not good enough - feel free to reopen. Cheers!

closed 2017-01-30 by dariuszseweryn