dariuszseweryn/RxAndroidBle #76: Library/RxBleClient enters faulty state / stops working

Stjerndal, 2016-10-03:
### Summary
My application connects to fixed bluetooth addresses and performs a few operations every ~5 minutes, indefinitely. After X* amounts of days, the app is no longer able to connect to devices. When in this state, it can't even scan for devices, not using the RxBleClient. After encountering this problem I created a test page in the app which would utilize the old-fashioned way of manually handling Android BLE (based on code from the official samples https://developer.android.com/samples/index.html which I used before finding RxAndroidBle). Now this problem happened again and I was surprised to find this old method was still able to scan/connect to devices, when the RxBleClienct could not. So the faulty state is in RxAndroidBle and not Androids BLE stack itself it seems.

Disabling/enabling wifi/bluetooth on the device does nothing. Restarting the app does solve the problem however.

Strangely enough, I don't get the impression that the RxBleClient is even trying to to perform bluetooth operations when in this state, as the log shows no messages from the native Android Ble stack at all.

*The amount of days varies, most recently it was about 2.5, but it has occurred in shorter time, or not at all with the app running for a week.
<br />
#### Preconditions
This problem has occurred on two different units, Samsung Galaxy Tab A and Samsung Galaxy Tab S2.
<br />
#### Steps to reproduce actual result
1.  Perform regular connects/disconnects and operations every 5 minutes.
2. Wait X* amount of time
3. ?
#### Actual result
The RxBleClient enters a faulty state. Unclear how or why.
Log for an attempt to connect to a device:
```
10-03 09:41:59.741 22196-9037/com.test.myapplication V/com.test.debug: Queueing Ble Device checks
10-03 09:41:59.741 22196-9037/com.test.myapplication V/com.test.debug: RxBlePolling Ble Device D5:8F:12:3F:2F:17
10-03 09:41:59.741 22196-9037/com.test.myapplication V/RxBleOperation: D5:8F:12:3F:2F:17: Connecting
10-03 09:41:59.746 22196-9037/com.test.myapplication E/com.test.debug: D5:8F:12:3F:2F:17: Refreshing delayed disconnect
10-03 09:41:59.996 3386-4221/? V/AlarmManager: Expired Alarm result :8
10-03 09:42:00.001 4322-4322/? D/KeyguardUpdateMonitor: received broadcast android.intent.action.TIME_TICK
10-03 09:42:00.001 4322-4322/? D/KeyguardUpdateMonitor: handleTimeUpdate
10-03 09:42:00.006 4322-4322/? D/Clock: received broadcast android.intent.action.TIME_TICK
10-03 09:42:00.036 4322-4322/? D/DateView: received broadcast android.intent.action.TIME_TICK
10-03 09:42:00.056 3386-3559/? V/BroadcastQueue: [background] Process cur broadcast BroadcastRecord{cc42270 u0 com.microsoft.office.outlook.action.TIME_CHANGED qIdx=4}, state= (APP_RECEIVE) DELIVERED for app ProcessRecord{3ea1350 5307:com.microsoft.office.outlook/u0a34}
10-03 09:42:00.056 4322-4322/? D/NotificationStackScrollLayout:  scroll range should be extended : 964
10-03 09:42:00.061 3386-5884/? V/BroadcastQueue: [background] Process cur broadcast BroadcastRecord{cc42270 u0 com.microsoft.office.outlook.action.TIME_CHANGED qIdx=4}, state= (APP_RECEIVE) DELIVERED for app ProcessRecord{3ea1350 5307:com.microsoft.office.outlook/u0a34}
10-03 09:42:00.066 3386-7256/? V/BroadcastQueue: [background] Process cur broadcast BroadcastRecord{4a0f9e9 u0 com.microsoft.office.outlook.action.AGENDA_APPWIDGET_UPDATE qIdx=4}, state= (APP_RECEIVE) DELIVERED for app ProcessRecord{3ea1350 5307:com.microsoft.office.outlook/u0a34}
10-03 09:42:00.121 3386-4221/? V/AlarmManager: Expired Alarm result :8
10-03 09:42:00.516 3386-4197/? D/SensorService: [SO] -0.509 0.193 9.652
10-03 09:42:01.131 3386-4221/? V/AlarmManager: Expired Alarm result :8
10-03 09:42:02.156 3386-4221/? V/AlarmManager: Expired Alarm result :8
10-03 09:42:03.186 3386-4221/? V/AlarmManager: Expired Alarm result :8
10-03 09:42:03.311 5307-5425/? D/ClClient: Not sending keepalive.  Current connection state=STOPPED
10-03 09:42:03.566 3386-6155/? D/SSRM:s: SIOP:: AP = 340, PST = 323 (W:18), LCD = 82
10-03 09:42:03.566 3386-6155/? D/ConnectivityService: returning getNetworkInfo for network type 1 : [type: WIFI[] - WIFI, state: CONNECTED/CONNECTED, reason: (unspecified), extra: "Service_Not_Available", roaming: false, failover: false, isAvailable: true]
10-03 09:42:03.761 3386-5884/? D/BatteryService: !@BatteryListener : batteryPropertiesChanged!
10-03 09:42:03.761 3386-5884/? D/BatteryService: level:100, scale:100, status:3, health:2, present:true, voltage: 4402, temperature: 301, technology: Li-ion, AC powered:false, USB powered:true, POGO powered:false, Wireless powered:false, icon:17303401, invalid charger:0, maxChargingCurrent:0
10-03 09:42:03.761 3386-5884/? D/BatteryService: online:6, current avg:222, charge type:2, power sharing:false, high voltage charger:false, capacity:280000, batterySWSelfDischarging:false, current_now:222
10-03 09:42:03.761 3386-3386/? D/BatteryService: Sending ACTION_BATTERY_CHANGED.
10-03 09:42:03.766 4322-4322/? D/KeyguardUpdateMonitor: received broadcast android.intent.action.BATTERY_CHANGED
10-03 09:42:03.766 4322-4322/? D/PowerUI: priorPlugType = 2 mPlugType =  2
10-03 09:42:03.766 4322-4322/? D/KeyguardUpdateMonitor: handleBatteryUpdate
10-03 09:42:03.776 1072-1072/? V/HeadsetService: HeadsetService - Received Intent.ACTION_BATTERY_CHANGED
10-03 09:42:03.776 1072-8657/? D/HeadsetStateMachine: Disconnected process message: 10, size: 0
10-03 09:42:03.791 4322-4322/? D/BatteryMeterView: ACTION_BATTERY_CHANGED : level:100 status:3 health:2
10-03 09:42:03.791 4322-4322/? D/BatteryMeterView: ACTION_BATTERY_CHANGED : level:100 status:3 health:2
10-03 09:42:04.196 3386-4221/? V/AlarmManager: Expired Alarm result :8
10-03 09:42:05.221 3386-4221/? V/AlarmManager: Expired Alarm result :8
10-03 09:42:06.251 3386-4221/? V/AlarmManager: Expired Alarm result :8
10-03 09:42:07.276 3386-4221/? V/AlarmManager: Expired Alarm result :8
10-03 09:42:08.306 3386-4221/? V/AlarmManager: Expired Alarm result :8
10-03 09:42:09.321 3386-4221/? V/AlarmManager: Expired Alarm result :8
10-03 09:42:09.751 22196-9102/com.test.myapplication E/com.test.debug: D5:8F:12:3F:2F:17: Timeout operation! Disconnecting!
10-03 09:42:09.751 22196-9102/com.test.myapplication V/RxBleOperation: D5:8F:12:3F:2F:17: Operation timed out.
10-03 09:42:09.751 22196-9102/com.test.myapplication V/RxBleOperation: Clearing subscription D5:8F:12:3F:2F:17
10-03 09:42:10.346 3386-4221/? V/AlarmManager: Expired Alarm result :8
10-03 09:42:10.521 3386-4197/? D/SensorService: [SO] -0.486 0.188 9.661
10-03 09:42:11.246 3386-4223/? D/InputReader: Input event(1): value=1 when=1329153094447000
10-03 09:42:11.246 3386-4223/? I/InputReader: Touch event's action is 0x0 (deviceType=0) [pCnt=1, s=0.6841 ] when=1329153094447000
10-03 09:42:11.251 3386-4223/? D/InputReader: lastThreadEndTime = 1329139545923152, currentThreadStartTime = 1329139545928235, diff = 0
10-03 09:42:11.251 3386-4222/? I/InputDispatcher: Delivering touch to (4322): action: 0x4, toolType: 1
10-03 09:42:11.251 3386-4222/? I/InputDispatcher: Delivering touch to (22196): action: 0x0, toolType: 1
10-03 09:42:11.251 22196-22196/com.test.myapplication D/ViewRootImpl: ViewPostImeInputStage processPointer 0
10-03 09:42:11.376 3386-4221/? V/AlarmManager: Expired Alarm result :8
10-03 09:42:11.381 3386-4223/? D/InputReader: Input event(1): value=0 when=1329153229128000
10-03 09:42:11.381 3386-4223/? I/InputReader: Touch event's action is 0x1 (deviceType=0) [pCnt=1, s=] when=1329153229128000
10-03 09:42:11.381 3386-4222/? I/InputDispatcher: Delivering touch to (22196): action: 0x1, toolType: 1
10-03 09:42:11.386 22196-22196/com.test.myapplication D/ViewRootImpl: ViewPostImeInputStage processPointer 1
```
<br />

Stjerndal, 2016-10-03:
Some code for connecting etc.

```
protected void prepareConnection(ConnectionSharingAdapter connectionSharingAdapter,
                                     GattStatusListener gattStatusListener,
                                     FinishedListener finishedListener) {
        this.mListener = gattStatusListener;
        this.mFinishedListener = finishedListener;
        mConnectionObservable = mBleDevice
                .establishConnection(mRxAppCompatActivity, false)
                .takeUntil(disconnectTriggerSubject)
                .compose(mRxAppCompatActivity.bindUntilEvent(PAUSE))
                .doOnUnsubscribe(this::clearSubscription)
                .compose(connectionSharingAdapter);

        mConnectionStateSubscription = mBleDevice.observeConnectionStateChanges()
                .subscribe(this::onConnectionChange);
}

public void connect() {
    if(mConnectionObservable == null) {
        return;
    }
    mConnectionSubscription = mConnectionObservable.subscribe(rxBleConnection -> {
                onConnected();
            }, this::onConnectionFailure);
            refreshDelayedDisconnect(DEFAULT_CONNECT_TIMEOUT);
}

public void triggerDisconnect() {
        disconnectTriggerSubject.onNext(null);
}

private Runnable disconnectRunnable = () -> {
        LocalDebug.logE(getDeviceAddress() + ": Timeout operation! Disconnecting!");
        updateStatusListener("Operation timed out.");
        triggerDisconnect();
    };

public void refreshDelayedDisconnect(int delay) {
    LocalDebug.logE(getDeviceAddress() + ": Refreshing delayed disconnect");
    if(mExecutor == null|| mExecutor.isShutdown())
        return;
    clearPendingDisconnects();
    pendingTimeouts.add(mExecutor.schedule(disconnectRunnable, delay, TimeUnit.MILLISECONDS));
}
```

dariuszseweryn, 2016-10-03:
It would be best to have the logs from the library - alternate the `RxBleLog` to log the data into a file and paste it here for reference. Would that be possible?

Stjerndal, 2016-10-04:
Yes, I'll set it up.

dariuszseweryn, 2016-10-20:
The above commit fixes an issue that may be the root cause of this problem as well. Could you re-test it with the newest snapshot release?

Stjerndal, 2016-10-21:
The problem had not occurred since my last post, but interestingly enough this new update causes it (or something with similar symptoms)  to happen regularly after ~15 minutes of action (always after 26 successful connects or 52 characteristic writes etc).

Here is a log excerpt: http://pasted.co/6064d801
In the program, basically, every couple of minutes ble device interactions occur. (1. ReadChara, 2. WriteChara, 3. ReadChara and depending on this value WriteChara, 4.  WriteChara and wait for notifications 5. When a certain indication is received, disconnect).

At about 15:08:45 in the log, things start to not work. ("No subscribers, finishing operation").
Before mentioned the commit, the exact same code worked fine (at least for several days).

dariuszseweryn, 2016-10-21:
The `"No subscribers, finishing operation"` is for cancelling connection in case of user unsubscribing from connection before it was established. It shouldn't interfere with your problem.
`status=257` indicates `GATT_FAILURE` according to https://developer.android.com/reference/android/bluetooth/BluetoothGatt.html
For me this is a different issue than the original one as operations are correctly processed - it is the Android OS BLE stack which is returning `257`
Could you try rebooting your phone and check if the issue will reoccur?

dariuszseweryn, 2016-10-21:
I have reverted the previous changes. I have introduced a bug in which a `BluetoothGatt` was not closed. I have made additional tests for that and created a proper fix but it will wait till Monday for someone else's eyes.

dariuszseweryn, 2016-10-25:
Fix merged into the `master`

closed 2016-10-29 by dariuszseweryn