dariuszseweryn/RxAndroidBle #77: Scan Results are not replayed if multiple subscribers subscribe.

shashanktomar, 2016-10-06:
### Summary
As per the documentation, RxBleClient.scanBleDevices() should replay the scan results if a new subscriber subscribe. But this is not the case. The reason behind this is following:
- the RxBleRadioOperationScan returns a **ReplaySubject** when operation is queued on **Radio**
- the method **createScanOperation()** in class **RxBleClientImpl**  **share()** the given observable
- the default RXJava behaviour is that if **share()** is used on **ReplaySubject**, it won't replay the values but will behave like a regular **PublishSubject**

#### Preconditions
Sensors are in vicinity and scan is initiated
#### Steps to reproduce actual result
1. Call **scanBleDevices** and get observable of **Observable<RxBleScanResult>**
2. Subscribe to the observable to start the scan
3. Assume it return results "ScanResult1" and "ScanResult2"
4. Again subscribe the observable using a new subscriber. Don't unsubscribe the first subscriber as it will stop the scan.
5. The observable never return "ScanResult1" and "ScanResult2" to the second subscriber.
#### Actual result
The scan results are not replayed for the second subscription.
#### Expected result
The scan results should be replayed for the second subscription.
<br />

uKL, 2016-10-25:
Hey!

Sorry for the delay, we are between projects now and trying to keep up with the library.

First of all, could you paste the documentation where it stands that results are replayed? That may be a mistake.

Our idea was to share scan operations, with respect of a passed filter. However we didn't plan to replay past results to any future subscriber. You can expect that when you'll subscribe with a new subscriber you will get an update, once there is a new scan result.

Do you think such feature might be useful? Could you provide some scenarios?

dariuszseweryn, 2016-11-10:
@shashanktomar Could you provide more information? It would be valuable to know where we can clarify the documentation or think about possible use cases.

closed 2016-11-11 by uKL