dariuszseweryn/RxAndroidBle #79: BleGattException when trying to do a particular write using .writeCharacteristic()

Orbyt, 2016-10-13:
### Summary
Writing to a specific characteristic crashes throws the following exception:

```
 Caused by: BleGattException{status=8, bleGattOperation=BleGattOperation{description='CHARACTERISTIC_WRITE'}}
      at com.polidea.rxandroidble.internal.connection.RxBleGattCallback.propagateStatusErrorIfGattErrorOccurred(RxBleGattCallback.java:245)
      at com.polidea.rxandroidble.internal.connection.RxBleGattCallback.access$100(RxBleGattCallback.java:26)
      at com.polidea.rxandroidble.internal.connection.RxBleGattCallback$1.onCharacteristicWrite(RxBleGattCallback.java:110)
      at android.bluetooth.BluetoothGatt$1.onCharacteristicWrite(BluetoothGatt.java:407)
      at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:279)
```
#### Preconditions
Connection is established, other methods both read and write seem to work fine

#### Steps to reproduce actual result
Code being used:
 ```
.writeCharacteristic(UUID, bytes)
.observeOn(AndroidSchedulers.mainThread());
```
My first thought was that perhaps the characteristic does not have a write permission enabled,
but the following log statement for `characteristic.getProperties()` returns 8, indicating it does in fact have write permissions:
```
.getCharacteristic(CharacteristicUUID)
                    .subscribe(new Action1<BluetoothGattCharacteristic>() {
                        @Override
                        public void call(BluetoothGattCharacteristic characteristic) {
                            Log.d(TAG, "characteristic permissions: " + characteristic.getPermissions());
                            Log.d(TAG, "characteristic properties: " + characteristic.getProperties());
                        }
                    });
```

So what might the issue be?

dariuszseweryn, 2016-10-13:
This question should go to the www.stackoverflow.com with tag `rxandroidble`

Orbyt, 2016-10-13:
Will do.

Orbyt, 2016-10-13:
Link: http://stackoverflow.com/questions/40031985/blegattexception-when-writing-to-a-particular-characteristic

closed 2016-10-14 by dariuszseweryn