dariuszseweryn/RxAndroidBle #81: RxBleRadioImpl is getting stuck after resetting Bluetooth

kanat, 2016-10-14:
RxBleRadioImpl is getting stuck after resetting Bluetooth
<br />
Need connected app
<br />
1) connect
2) switch off bluetooth
3) switch on bluetooth
4) connect
#### Actual result
nothing happens, No connection
#### Expected result
Connection established

I can't see FINISHED after disabling bluetooth and also it's trying to write after disconnection:
10-14 17:44:02.136 2072-2931/ru.starline.key D/RxBle#Radio:   QUEUED RxBleRadioOperationDisconnect(945209293)
10-14 17:44:02.136 2072-2133/ru.starline.key D/RxBle#Radio:  STARTED RxBleRadioOperationDisconnect(945209293)
10-14 17:44:02.156 2072-2931/ru.starline.key D/RxBle#Radio:   QUEUED RxBleRadioOperationDescriptorWrite(411833235)
10-14 17:44:02.166 2072-2133/ru.starline.key D/RxBle#Radio: FINISHED RxBleRadioOperationDisconnect(945209293)
10-14 17:44:02.166 2072-2133/ru.starline.key D/RxBle#Radio:  STARTED RxBleRadioOperationDescriptorWrite(411833235)
10-14 17:44:02.166 2072-2931/ru.starline.key D/RxBle#Radio:   QUEUED RxBleRadioOperationDescriptorWrite(840731593)

aldoborrero, 2016-10-14:
I've noticed the same exact input. The operations are queued even if you specified the operation retryWhen().

aldoborrero, 2016-10-17:
To have more insight on this particular issue. I'll copy here the steps I'm doing to reproduce it:

I have a class which is called BleDeviceManager:

```java

  @Override public void connect(@NonNull final String address) {
      rxBleDevice = rxBleClient.getBleDevice(address);
      
      // Setup observation of connectivity changes
      connectionStateSubscription = rxBleDevice.observeConnectionStateChanges().observeOn(AndroidSchedulers.mainThread()).subscribe(this::onConnectionStateChanged);

      // Start properly the connection to the device
      connectionObservable = rxBleDevice.establishConnection(context, false)
          .takeUntil(disconnectionTrigger)
          .retryWhen(this::onShouldRetryConnectionOperation)
          .compose(new ConnectionSharingAdapter());

      // Subscribe to the observable
      connectionObservable.subscribe(this::onConnectionSuccess, this::onConnectionFailure);
  }

  private Observable<?> onShouldRetryConnectionOperation(Observable<? extends Throwable> observable) {
    return observable.delay(RECONNECTION_DELAY_TIME, TimeUnit.SECONDS).filter(throwable -> {
      if (throwable instanceof BleDisconnectedException) {
        return true;
      }

      if (throwable instanceof BleGattException) {
        BleGattException bleGattException = (BleGattException) throwable;

        switch (bleGattException.getStatus()) {
          case BLUEDROID_GATT_ERROR:
          case BLUEDROID_GATT_CONNECTION_TIMEOUT:
            return true;
        }
      }

      return false;
    });
  }

  private void onConnectionSuccess(RxBleConnection rxBleConnection) {
        deviceSubscription = connectionObservable.observeOn(Schedulers.computation())
            .flatMap(connection -> connection.discoverServices()
                .flatMap(services -> services.getService(SERVICE_ID).map(BluetoothGattService::getCharacteristics)), Pair::with)
            .flatMap(pair -> {
              final RxBleConnection connection = pair.getValue0();
              final List<BluetoothGattCharacteristic> characteristics = pair.getValue1();
              return readInitialValues(connection, characteristics).concatWith(setupNotifications(connection, characteristics));
            })
            .subscribe(ignored -> {
            }, throwable -> {
              logger.e(TAG, "Error: " + throwable);
            });
  }

  private Observable<Triplet<UUID, ValueAdapter<?>, byte[]>> readInitialValues(final RxBleConnection connection,
      final List<BluetoothGattCharacteristic> characteristics) {
    return Observable.from(characteristics)
        .filter(characteristic -> BleUtils.hasReadProperty(characteristic.getProperties()))
        .flatMap(connection::readCharacteristic, Pair::with)
        .observeOn(Schedulers2.singleThread())
        .compose(valueAdapterTransformer)
        .doOnNext(triplet -> {
              // Do things with the values received
        });
  }

  private Observable<Triplet<UUID, ValueAdapter<?>, byte[]>> setupNotifications(final RxBleConnection connection,
[... 507 lines cut ...]
10-17 09:14:48.822 26709-26709/com.test.bluetooth D/RxBle#Radio:   QUEUED RxBleRadioOperationDescriptorWrite(95024266)
10-17 09:14:48.822 26709-26709/com.test.bluetooth D/BluetoothGatt: setCharacteristicNotification() - uuid: 6e4a0020-a5a3-f393-e0a9-e50e24dcaa9a enable: false
10-17 09:14:48.824 26709-26709/com.test.bluetooth D/RxBle#Radio:   QUEUED RxBleRadioOperationDescriptorWrite(117466904)
10-17 09:14:48.824 26709-26709/com.test.bluetooth D/BluetoothGatt: setCharacteristicNotification() - uuid: 6e4a001d-a5a3-f393-e0a9-e50e24dcaa9a enable: false
10-17 09:14:48.826 26709-26709/com.test.bluetooth D/RxBle#Radio:   QUEUED RxBleRadioOperationDescriptorWrite(238735190)
10-17 09:14:48.826 26709-26709/com.test.bluetooth D/BluetoothGatt: setCharacteristicNotification() - uuid: 6e4a0019-a5a3-f393-e0a9-e50e24dcaa9a enable: false
10-17 09:14:48.828 26709-26709/com.test.bluetooth D/RxBle#Radio:   QUEUED RxBleRadioOperationDescriptorWrite(186703044)
10-17 09:14:48.828 26709-26709/com.test.bluetooth D/BluetoothGatt: setCharacteristicNotification() - uuid: 6e4a001a-a5a3-f393-e0a9-e50e24dcaa9a enable: false
10-17 09:14:48.830 26709-26709/com.test.bluetooth D/RxBle#Radio:   QUEUED RxBleRadioOperationDescriptorWrite(86805218)
10-17 09:14:48.831 26709-26709/com.test.bluetooth D/BluetoothGatt: setCharacteristicNotification() - uuid: 6e4a0015-a5a3-f393-e0a9-e50e24dcaa9a enable: false
10-17 09:14:48.832 26709-26709/com.test.bluetooth D/RxBle#Radio:   QUEUED RxBleRadioOperationDescriptorWrite(95486256)
10-17 09:14:48.833 26709-26709/com.test.bluetooth D/BluetoothGatt: setCharacteristicNotification() - uuid: 6e4a0018-a5a3-f393-e0a9-e50e24dcaa9a enable: false
10-17 09:14:48.834 26709-26709/com.test.bluetooth D/RxBle#Radio:   QUEUED RxBleRadioOperationDescriptorWrite(76313390)
10-17 09:14:48.835 26709-26709/com.test.bluetooth D/BleDeviceManager: Device has been disconnected!
10-17 09:14:53.783 26709-26709/com.test.bluetooth D/DeviceManager: Connection state changed: RxBleConnectionState{CONNECTING}
10-17 09:14:53.787 26709-27395/com.test.bluetooth D/RxBle#Radio:   QUEUED RxBleRadioOperationConnect(6119269)
```

This is the current state of the `RxBleRadio` queue (entries in the queue ranging from 2 to 15 are descriptor writes):

[image]

Is there a way to clear the queue? Or maybe to add a timeout to the operations?

Thanks

dariuszseweryn, 2016-10-17:
This should be easy to fix. In my opinion this scenario should emit `BleDisconnectedException`. To achieve that we will move lines 55-58 of `RxBleConnectionConnectorImpl` below the `.first()` clause. Then the observing disconnection will be active as long as someone is using the connection.

dariuszseweryn, 2016-10-18:
Hello. I was thinking about the problem. Actually it is not possible to cancel the current operation when it is executing because only establishing connection can be effectively cancelled. 
I was always thinking that rxJava is supposed to be used in a single flow describing all actions (from establishing connection, discovering services if needed, performing read / writes / notifies, etc.) with a single `.subscribe()`. Then the subscriber would be always informed about any error that happens during the connection.
But like in this situation when a disconnection would happen when an operation is in the middle of execution it won't be notified about anything and will effectively be frozen without any chance to resume. This may be actually the cause of a different issue #76 .
I will definitely look into it as soon as possible.

aldoborrero, 2016-10-18:
Hi @dariuszseweryn thanks for your comment!

Is there any possibility that we can have a "hacky" solution meanwhile? Even if is not the most elegant one?

(I would like to help you to sort this out).

dariuszseweryn, 2016-10-18:
I am starting with writing tests for `RxBleGattCallback`. I didn't checked it yet though inserting
```
observeDisconnect(),
```
at line 256 of `RxBleGattCallback` could do the trick.

aldoborrero, 2016-10-18:
Bad luck:

[image]

```
10-18 20:44:53.248 20817-20832/com.test.bluetooth D/RxBle#Radio: FINISHED RxBleRadioOperationCharacteristicRead(122289203)
10-18 20:44:53.249 20817-20832/com.test.bluetooth D/RxBle#Radio:  STARTED RxBleRadioOperationCharacteristicRead(20002153)
10-18 20:44:53.329 20817-20829/com.test.bluetooth D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=6 device=D8:7A:09:F1:66:8E
10-18 20:44:53.330 20817-20829/com.test.bluetooth D/RxBle#BluetoothGatt: onConnectionStateChange newState=0 status=0
10-18 20:44:53.333 20817-20817/com.test.bluetooth D/DeviceManager: Connection state changed: DISCONNECTED
10-18 20:44:53.333 20817-20916/com.test.bluetooth D/RxBle#Radio:   QUEUED RxBleRadioOperationDisconnect(237186712)
10-18 20:44:53.335 20817-20817/com.test.bluetooth D/RxBle#Radio:  REMOVED RxBleRadioOperationCharacteristicRead(43958053)
10-18 20:44:53.335 20817-20832/com.test.bluetooth D/RxBle#Radio: FINISHED RxBleRadioOperationCharacteristicRead(20002153)
10-18 20:44:53.335 20817-20832/com.test.bluetooth D/RxBle#Radio:  STARTED RxBleRadioOperationCharacteristicRead(90095275)
10-18 20:44:53.337 20817-20817/com.test.bluetooth D/RxBle#Radio:  REMOVED RxBleRadioOperationCharacteristicRead(161360015)
10-18 20:44:53.340 20817-20817/com.test.bluetooth D/RxBle#Radio:  REMOVED RxBleRadioOperationCharacteristicRead(84564003)
10-18 20:44:53.342 20817-20817/com.test.bluetooth D/RxBle#Radio:  REMOVED RxBleRadioOperationCharacteristicRead(77670561)
10-18 20:44:53.344 20817-20817/com.test.bluetooth D/RxBle#Radio:  REMOVED RxBleRadioOperationCharacteristicRead(71867015)
10-18 20:44:53.345 20817-20817/com.test.bluetooth D/RxBle#Radio:  REMOVED RxBleRadioOperationCharacteristicRead(168209885)
10-18 20:44:53.345 20817-20817/com.test.bluetooth D/BleDeviceManager: Device has been disconnected!
10-18 20:44:58.336 20817-20817/com.test.bluetooth D/DeviceManager: Connection state changed: CONNECTING
10-18 20:44:58.344 20817-20916/com.test.bluetooth D/RxBle#Radio:   QUEUED RxBleRadioOperationConnect(108345520)
```

dariuszseweryn, 2016-10-18:
Another shot which theoretically should work:
```
    private <T> Observable<T> withHandlingStatusError(Observable<T> observable) {
        //noinspection unchecked
        return Observable.merge(
                statusErrorSubject.asObservable(), // statusErrorSubject emits only errors
                disconnectedErrorObservable,
                observable
        );
    }
```
where
```
    private final Observable disconnectedErrorObservable = connectionStatePublishSubject
            .filter(this::isDisconnectedOrDisconnecting)
            .doOnNext(rxBleConnectionState -> bluetoothGattBehaviorSubject.onCompleted())
            .cache()
            .flatMap(rxBleConnectionState -> Observable.error(new BleDisconnectedException()));
```
I have quickly designed two test cases:
```
    def ".observeDisconnect() should emit error when .onConnectionStateChange() receives DISCONNECTED state"() {

        given:
        objectUnderTest.observeDisconnect().subscribe(testSubscriber)

        when:
        objectUnderTest.getBluetoothGattCallback().onConnectionStateChange(null, 0, 0)

        then:
        testSubscriber.assertError(BleDisconnectedException)
    }

    def ".observeDisconnect() should emit error if .onConnectionStateChange() received DISCONNECTED state"() {
        given:
        objectUnderTest.getBluetoothGattCallback().onConnectionStateChange(null, 0, 0)

        when:
        objectUnderTest.observeDisconnect().subscribe(testSubscriber)

        then:
        testSubscriber.assertError(Throwable)
    }
```
The first one passes and the second one fails with somewhat funny message:
```
java.lang.AssertionError: No errors (0 completions)

	at rx.observers.TestSubscriber.assertionError(TestSubscriber.java:663)
	at rx.observers.TestSubscriber.assertError(TestSubscriber.java:528)
	at com.polidea.rxandroidble.internal.connection.RxBleGattCallbackTest..observeDisconnect() should emit error if .onConnectionStateChange() received DISCONNECTED state(RxBleGattCallbackTest.groovy:80)
Caused by: com.polidea.rxandroidble.exceptions.BleDisconnectedException
	at com.polidea.rxandroidble.internal.connection.RxBleGattCallback.lambda$new$1(RxBleGattCallback.java:49)
	at rx.internal.operators.OnSubscribeMap$MapSubscriber.onNext(OnSubscribeMap.java:69)
	at rx.internal.operators.NotificationLite.accept(NotificationLite.java:152)
	at rx.internal.operators.CachedObservable$ReplayProducer.replay(CachedObservable.java:407)
	at rx.internal.operators.CachedObservable$CacheState.dispatch(CachedObservable.java:223)
	at rx.internal.operators.CachedObservable$CacheState.onNext(CachedObservable.java:194)
	at rx.internal.operators.CachedObservable$CacheState$1.onNext(CachedObservable.java:174)
	at rx.internal.operators.OperatorDoOnEach$1.onNext(OperatorDoOnEach.java:86)
	at rx.internal.operators.OnSubscribeFilter$FilterSubscriber.onNext(OnSubscribeFilter.java:76)
	at rx.subjects.SubjectSubscriptionManager$SubjectObserver.onNext(SubjectSubscriptionManager.java:225)
	at rx.subjects.PublishSubject.onNext(PublishSubject.java:113)
	at rx.internal.util.ActionSubscriber.onNext(ActionSubscriber.java:39)
	at rx.observers.SafeSubscriber.onNext(SafeSubscriber.java:134)
	at rx.internal.operators.OperatorObserveOn$ObserveOnSubscriber.call(OperatorObserveOn.java:227)
	at rx.internal.schedulers.EventLoopsScheduler$EventLoopWorker$1.call(EventLoopsScheduler.java:172)
	at rx.internal.schedulers.ScheduledAction.run(ScheduledAction.java:55)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$201(ScheduledThreadPoolExecutor.java:180)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:293)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)
```
You can give it a shot - I do not have any Android device at the moment and can only test with unit tests.

aldoborrero, 2016-10-19:
With this change, it looks very promising! For now, it does the reconnection properly!!!

I'll tell you if I detect something wrong, testing more :)

dariuszseweryn, 2016-10-20:
The above commit should fix the issue for good. In case of anything wrong - feel free to reopen the issue. Best Regards.

aldoborrero, 2016-10-20:
Thanks @dariuszseweryn for your great work! Worth mentioning how fast you solved this thing! :)

dariuszseweryn, 2016-10-21:
Reopening the issue as the fix introduced a different bug and was reverted. The fix is ready though it will wait till Monday for a code review.

aldoborrero, 2016-10-24:
Can you share the new changes on the code? Or is already commited? I can test it on my device @dariuszseweryn Thanks!! :)

dariuszseweryn, 2016-10-24:
I have just pushed branch `fix/issue_81` which you can checkout. It's waiting for a code review.

aldoborrero, 2016-10-24:
For now, working fine with my real device! 👍

dariuszseweryn, 2016-10-25:
Fix merged. Should be available in the SNAPSHOT.

closed 2016-10-25 by dariuszseweryn