dariuszseweryn/RxAndroidBle #83: Can't establish connection on Android 7.1.1

klemzy, 2016-10-20:
### Summary
Establish connection with `autoconnect=true` connection crashes with reflection exception.

#### Preconditions
Android 7.1.1
#### Steps to reproduce actual result
1. Try to establish connection with `autoconnect=true`
#### Actual result
java.lang.IllegalArgumentException: method android.bluetooth.BluetoothGatt.<init> argument 1 has type android.bluetooth.IBluetoothGatt, got context
	at java.lang.reflect.Constructor.newInstance0(Native Method)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:430)
	at com.polidea.rxandroidble.internal.util.BleConnectionCompat.createBluetoothGatt(BleConnectionCompat.java:108)
	at com.polidea.rxandroidble.internal.util.BleConnectionCompat.connectGatt(BleConnectionCompat.java:53)
	at com.polidea.rxandroidble.internal.operations.RxBleRadioOperationConnect.lambda$getConnectedBluetoothGatt$40(RxBleRadioOperationConnect.java:77)
	at com.polidea.rxandroidble.internal.operations.RxBleRadioOperationConnect$$Lambda$8.call(Unknown Source)

<br />
#### Expected result
Connection got established
<br />

dariuszseweryn, 2016-10-21:
Using reflection for `autoConnect = true` is no longer needed past Android 7.0.0. Fix is on it's way.

dariuszseweryn, 2016-10-25:
Just pushed to the master. Should be available in the SNAPSHOT release shortly.

closed 2016-10-25 by dariuszseweryn

klemzy, 2016-10-25:
Great. Thanks!