dariuszseweryn/RxAndroidBle #87: Turn off Bluetooth in the scan,Will lead to the application of forced exit

lvzhongyi, 2016-10-28:
```
Subscription scanSubscription = rxBleClient.scanBleDevices()
    .subscribe(rxBleScanResult -> {
        // Process scan result here.
    });

// When done, just unsubscribe.
scanSubscription.unsubscribe();
```
throw OnErrorNotImplementedException

dariuszseweryn, 2016-10-28:
Hello @lvzhongyi 

I am afraid that you forgot to handle an erroneous situation when Bluetooth adapter switches off and the scanning must be stopped.
The code should look like this:

```
Subscription scanSubscription = rxBleClient.scanBleDevices()
    .subscribe(
        rxBleScanResult -> {
            // Process scan result here.
        },
        throwable -> {
            // Handle error here.
        }
    );

// When done, just unsubscribe.
scanSubscription.unsubscribe();
```

You can find more on error handling with rxJava here: https://github.com/ReactiveX/RxJava/wiki/Error-Handling

Best Regards.

closed 2016-10-28 by dariuszseweryn

lvzhongyi, 2016-11-04:
sorry,forgot to get back to you,you may need to update your Home 'README.md'

dariuszseweryn, 2016-11-04:
Thanks for the suggestion. Will do that. :) Cheers!