dariuszseweryn/RxAndroidBle #90: Not receiving state change for DISCONNECTED when turning connected device off

markschmid, 2016-11-02:
We're observing the connection state with observeConnectionStateChanges() and expect to receive a state change DISCONNECTED if a connected device is turned off (or gets out of range etc.), but we do not get notified in those cases. The state changes for CONNECTING and CONNECTED are received correctly.

See our code below. Are we doing something wrong?

```
mBleDevice.observeConnectionStateChanges()
  .observeOn(AndroidSchedulers.mainThread())
  .subscribe(this::onConnectionStateChange);

mConnectionSubscription = mBleDevice.establishConnection(MyApplication.getAppContext(), true)
.observeOn(AndroidSchedulers.mainThread())
.subscribe(this::onConnectionReceived, this::onConnectionFailure);
```

dariuszseweryn, 2016-11-09:
Hello,

I have just tested against the recent SNAPSHOT this code:

    connectedSubscription = bleDevice.observeConnectionStateChanges()
            .subscribe(state -> Log.e("STATE", state.toString()));
    scanSubscription =
            bleDevice
                    .establishConnection(this, false)
                    .flatMap(rxBleConnection -> Observable
                            .interval(1, TimeUnit.SECONDS)
                            .flatMap(t -> rxBleConnection.readCharacteristic(philRxAvailableUuid))
                            .take(1)
                    )
                    .subscribe(
                            bytes -> Log.e("TEST", "bytes: " + Arrays.toString(bytes)),
                            throwable -> Log.e("TEST", "ERROR", throwable)
                    );

And then put the device into a Faraday's Cage. I got a log:

```
11-09 17:37:02.370 30125-30145/com.polidea.rxandroidble.sample D/RxBle#Radio:  STARTED RxBleRadioOperationConnect(1108013008)
11-09 17:37:02.441 30125-30125/com.polidea.rxandroidble.sample D/BluetoothGatt: connect() - device: D0:5F:B8:3C:7C:7E, auto: false
11-09 17:37:02.441 30125-30125/com.polidea.rxandroidble.sample D/BluetoothGatt: registerApp()
11-09 17:37:02.477 30125-30125/com.polidea.rxandroidble.sample D/BluetoothGatt: registerApp() - UUID=f6ff4013-2b52-4b1a-a48b-6a06905c14ca
11-09 17:37:02.481 30125-30136/com.polidea.rxandroidble.sample D/BluetoothGatt: onClientRegistered() - status=0 clientIf=5
11-09 17:37:02.528 30125-30125/com.polidea.rxandroidble.sample I/Adreno-EGL: <qeglDrvAPI_eglInitialize:381>: EGL 1.4 QUALCOMM build: AU_LINUX_ANDROID_KK_2.7_RB1.04.04.04.007.066_msm8960_KK_2.7_RB1__release_AU ()
                                                                             OpenGL ES Shader Compiler Version: 17.01.12.SPL
                                                                             Build Date: 09/11/14 Thu
                                                                             Local Branch: mybranch4451390
                                                                             Remote Branch: quic/kk_2.7_rb1.43
                                                                             Local Patches: NONE
                                                                             Reconstruct Branch: AU_LINUX_ANDROID_KK_2.7_RB1.04.04.04.007.066 + 73685fa + 87c0666 + c4bc670 + 6154422 + 594ea9d +  NOTHING
11-09 17:37:02.573 30125-30125/com.polidea.rxandroidble.sample D/OpenGLRenderer: Enabling debug mode 0
11-09 17:37:02.633 30125-30137/com.polidea.rxandroidble.sample D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=5 device=D0:5F:B8:3C:7C:7E
11-09 17:37:02.633 30125-30137/com.polidea.rxandroidble.sample D/RxBle#BluetoothGatt: onConnectionStateChange newState=2 status=0
11-09 17:37:02.644 30125-30175/com.polidea.rxandroidble.sample E/STATE: RxBleConnectionState{CONNECTED}
11-09 17:37:02.690 30125-30175/com.polidea.rxandroidble.sample D/RxBle#RadioOperationConnect$$Lambda: No subscribers, finishing operation
11-09 17:37:02.691 30125-30145/com.polidea.rxandroidble.sample D/RxBle#Radio: FINISHED RxBleRadioOperationConnect(1108013008)
11-09 17:37:03.705 30125-30180/com.polidea.rxandroidble.sample D/RxBle#Radio:   QUEUED RxBleRadioOperationServicesDiscover(1108595112)
11-09 17:37:03.722 30125-30145/com.polidea.rxandroidble.sample D/RxBle#Radio:  STARTED RxBleRadioOperationServicesDiscover(1108595112)
11-09 17:37:03.730 30125-30125/com.polidea.rxandroidble.sample D/BluetoothGatt: discoverServices() - device: D0:5F:B8:3C:7C:7E
(... services, characteristics, descriptors ...)
11-09 17:37:03.832 30125-30137/com.polidea.rxandroidble.sample D/BluetoothGatt: onSearchComplete() = Device=D0:5F:B8:3C:7C:7E Status=0
11-09 17:37:03.834 30125-30137/com.polidea.rxandroidble.sample D/RxBle#BluetoothGatt: onServicesDiscovered status=0
11-09 17:37:03.839 30125-30175/com.polidea.rxandroidble.sample D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(1108717920)
11-09 17:37:03.840 30125-30145/com.polidea.rxandroidble.sample D/RxBle#Radio: FINISHED RxBleRadioOperationServicesDiscover(1108595112)
11-09 17:37:03.841 30125-30145/com.polidea.rxandroidble.sample D/RxBle#Radio:  STARTED RxBleRadioOperationCharacteristicRead(1108717920)
11-09 17:37:03.862 30125-30125/com.polidea.rxandroidble.sample D/BluetoothGatt: readCharacteristic() - uuid: a59bfff2-6343-4053-a67c-357cc7f8f1a9
11-09 17:37:03.947 30125-30136/com.polidea.rxandroidble.sample D/BluetoothGatt: onCharacteristicRead() - Device=D0:5F:B8:3C:7C:7E UUID=a59bfff2-6343-4053-a67c-357cc7f8f1a9 Status=0
11-09 17:37:03.948 30125-30136/com.polidea.rxandroidble.sample D/RxBle#BluetoothGatt: onCharacteristicRead characteristic=a59bfff2-6343-4053-a67c-357cc7f8f1a9 status=0
11-09 17:37:03.950 30125-30175/com.polidea.rxandroidble.sample E/TEST: bytes: [1]
11-09 17:37:03.951 30125-30145/com.polidea.rxandroidble.sample D/RxBle#Radio: FINISHED RxBleRadioOperationCharacteristicRead(1108717920)
11-09 17:37:20.871 30125-30137/com.polidea.rxandroidble.sample D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=5 device=D0:5F:B8:3C:7C:7E
11-09 17:37:20.875 30125-30137/com.polidea.rxandroidble.sample D/RxBle#BluetoothGatt: onConnectionStateChange newState=0 status=0
11-09 17:37:20.886 30125-30175/com.polidea.rxandroidble.sample E/TEST: ERROR
                                                                       com.polidea.rxandroidble.exceptions.BleDisconnectedException
[... cut ...]
11-09 17:37:20.900 30125-30125/com.polidea.rxandroidble.sample D/BluetoothManager: getConnectedDevices
11-09 17:37:20.902 30125-30145/com.polidea.rxandroidble.sample D/RxBle#Radio: FINISHED RxBleRadioOperationDisconnect(1108027760)
11-09 17:37:20.903 30125-30125/com.polidea.rxandroidble.sample D/BluetoothGatt: close()
11-09 17:37:20.903 30125-30125/com.polidea.rxandroidble.sample D/BluetoothGatt: unregisterApp() - mClientIf=5
```

And this looks correct. What version of the library are you using?

dariuszseweryn, 2016-11-16:
Since I cannot recreate the issue and the author doesn't respond - closing. Feel free to reopen and add comments when available.
Best Regards.

closed 2016-11-16 by dariuszseweryn

sw-tt-rutulshah, 2018-03-22:
please give the state of scanSubscription and also about the philRxAvailableUuid from where I have to take it?