dariuszseweryn/RxAndroidBle #92: Confusion with setting a characteristic changed notification

Orbyt, 2016-11-08:
The documentation gives the following snippet for setting a notification for characteristic changes: 

```
device.establishConnection(context, false)
    .flatMap(rxBleConnection -> rxBleConnection.setupNotification(characteristicUuid))
    .doOnNext(notificationObservable -> {
        // Notification has been set up
    })
    .flatMap(notificationObservable -> notificationObservable) // <-- Notification has been set up, now observe value changes.
    .subscribe(
        bytes -> {
            // Given characteristic has been changes, here is the value.
        },
        throwable -> {
            // Handle an error here.
        }
    );
```

A few questions:

1. What is the purpose of the second `flatMap()`? It just returns the same observable.
2. Previously, i've had a `connect()` method that runs the following to setup a basic connection:

```
mEstablishedConnection = device.establishConnection(mContext, false)
                .observeOn(AndroidSchedulers.mainThread())
                .doOnError(new Action1<Throwable>() {
                    @Override
                    public void call(Throwable throwable) {
                        throwable.printStackTrace();
                        connectionListener.onFailed();
                    }
                })
                .subscribe(rxBleConnection -> {
                    mConnection = rxBleConnection;
                    connectionListener.onConnected();
                }, throwable -> {
                    if (throwable != null) {
                        throwable.printStackTrace();
                    }
                });
```

My question here is, to setup characteristic changed notifications, should I add the call in the flow of my exisiting `device.estaablishConnection()` call? Or do I need to do another `device.establishConnection()`?

With the former, subscribing will emit a `byte[]` instead of a `rxBleConnection`, but the `rxBleConnection` is needed for future BLE interaction. Which is it?

dariuszseweryn, 2016-11-09:
Hello @Orbyt 

Ad. 1. Setting up the notifications / indications is an asynchronous action and generally will finish at some point in the future. When the `Observable<byte[]>` is emitted then the setting up of the notification is completed. You can think of it as performing a read operation on a connection but instead of one value it will return an `Observable` that will emit actual values.

Ad. 2. You cannot have two concurrent connections opened for the same device. You can have two concurrent connections for two different devices.
To interact with the device you need to have a reference to the `RxBleConnection` wether it will be stored as `mConnection` or will be passed in a more "reactive" way.

Best Regards

dariuszseweryn, 2016-11-11:
@Orbyt With lack of any comments in this topic I assume my response did answered you question. In case it did not - feel free to reopen.

Best Regards

closed 2016-11-11 by dariuszseweryn