dariuszseweryn/RxAndroidBle #93: Can't setup notification in some device

lowwor, 2016-11-10:
### Summary

For some device, we can't not setup ble notification using current implementation of ble notification setup in the library.

However, if I change 
`bluetoothGatt.writeDescriptor(bluetoothGattDescriptor);`
to
`bluetoothGatt.setWriteType(BluetoothGattCharacteristic.WRITE_TYPE_DEFAULT);
 bluetoothGatt.writeDescriptor(bluetoothGattDescriptor);`

the notification is set up correctly.

I think it's a known bug as also stated and solved in the below repo.
[`Android-nRF-Toolbox`](https://github.com/NordicSemiconductor/Android-nRF-Toolbox/blob/master/app/src/main/java/no/nordicsemi/android/nrftoolbox/profile/BleManager.java#L543)

dariuszseweryn, 2016-11-10:
Will be fixed in the next build of SNAPSHOT release (around 10 minutes or so).
Thanks for the info!

closed 2016-11-10 by dariuszseweryn

lowwor, 2016-11-11:
Great. But it seems the snapshot build failed for some checkstyle problems [(travis)](https://travis-ci.org/Polidea/RxAndroidBle/builds/174813424), please check with that. Thanks!

dariuszseweryn, 2016-11-11:
I find the checkstyle slightly annoying... I will add it to my local build script.
Now it works. Thanks for the info!